socallinuxexpo/scale-network #6: Investigate network interface MAC offset

sarcasticadmin, 2018-01-03:
With the configuration from last year: https://github.com/socallinuxexpo/SCALE_network_public/tree/master/SCALE15x/wireless/openwrt
https://github.com/socallinuxexpo/SCALE_network_private/blob/master/scale15x/makeAPconfig.sh

The interfaces receive a hardware MAC address which were offset `xx |0x02`. What causes this to occur? If possible can we assign port `#4` on the AP switch to be assigned the expected MAC (i.e. the MAC on the underside of the case).

sarcasticadmin, 2018-02-11:
According to @davidelang this offset is due to the WAN port being the MAC thats posted on the back of the case. And the offset MAC is the MAC of the CPUs eth port.

davidelang, 2018-03-04:
the flashing script will give us the correct mac address in any case

sarcasticadmin, 2018-03-05:
Thanks @davidelang thats true. This MAC offset should not longer be an issue.

closed 2018-03-05 by sarcasticadmin