socallinuxexpo/scale-network #16: Pre 16x Shopping List

sarcasticadmin, 2018-02-11:
What does the Tech team need to order before the week of the conference?

**Current List**
EMPTY

kidicarus2, 2018-02-11:
Was the gaffer tape ordered?

To Your Success
- Steven

kylerisse, 2018-02-11:
We need more than 5 or more 6' HDMI cables. Last year we had to borrow from the AV team for a few of the facility supplied televisions. Can't guarantee their availability.

owendelong, 2018-02-11:
I was told that it was, but I will confirm. If not, I will order it along with HDMI and DVI cables.

* 8 DVI to DVI monitor cables

sarcasticadmin, 2018-02-11:
Thanks everyone! I updated the list

owendelong, 2018-02-14:
OK.. I'm closing out the shopping list and placing the order now. Anything added below here likely won't get ordered for the show.

sbibayoff, 2018-02-14:
If it's now too late:
Threshold hold Gaffers tape (yellow, with black stripes), - 2 roles

owendelong, 2018-02-14:
It should be, but I happened to glance at this just before submitting the order.

I’ll add it.

Owen

owendelong, 2018-02-14:
Added 2 rolls of caution striped yellow/black gaffer tape (3") to the list per Steve Bibayoff's request.

owendelong, 2018-02-14:
Order has been placed for all items above this marker.

closed 2018-02-14 by owendelong