socallinuxexpo/scale-network #25: IPv4 subnet sizes for wifi

kylerisse, 2018-02-15:
@owendelong question about the IPv4 subnets

Currently the IP space allocated for VLANS 100, 101, 500, and 501 are all /24. Looking at last years notes it seems we allocated a /21 for each of the public WIFI networks in each building. Do you think /24s will be sufficient replacements? My concern is that during keynotes and high expo hall usage times we might be cutting it close as people pack into a single building at a higher Expo to Conference ratio. I don't know what last years numbers look like to compare though. 

And yes, I'll pre-agree that IPv6 "should" be sufficient.

owendelong, 2018-02-15:
Good question.

Actually, we allocated /21s to each of the segments across the entire show. We didn’t have separation between
the buildings.

If we could assume an even split between buildings (we can’t, not even close, I suspect), we could use as /22
for each VLAN this year.

Pretty confident in the /64s. Will revisit the WiFi numbers. That’s probably going to blow my easy numbering plan
out of the water. :(

Owen

kylerisse, 2018-02-15:
I thought that also, but when going back to the documentation it showed:

```
| SCALE-public 2g  | 10.40.0.0/21 | 2001:470:f3fc:40::/64 | vlan=40 | Guest network conf building | 10.40.0.20 - 10.40.7.250 |
| SCALE public 5g | 10.41.0.0/21 | 2001:470:f3fc:41::/64 | vlan=41 | Guest network conf building | 10.41.0.20 - 10.41.7.250 |
| SCALE-public 2g  | 10.42.0.0/21 | 2001:470:f3fc:42::/64 | vlan=42 | Guest network expo building | 10.42.0.20 - 10.42.7.250 |
| SCALE public 5g | 10.43.0.0/21 | 2001:470:f3fc:43::/64 | vlan=43 | Guest network expo building | 10.43.0.20 - 10.43.7.250 |
| SCALE Staff  | 10.60.0.0/23 | 2001:470:f3fc:60::/64 | vlan=60 | Connected to the staff wireless network | 10.60.0.20 - 10.60.1.250 |
```

We spanned the Staff Wifi across both, but according to this had public similar to this year's plan. I think that might be what caused the confusion for both of us. I don't know if we have anything to reference for actual usage vs allocation. 

Thanks Owen.

owendelong, 2018-02-15:
I’m 99% positive that we didn’t route them separately (or at least they were bridged all over both buildings) last year.

Anyway, it doesn’t matter. Since we can’t assume an even split, we need to go with a /21 in the Expo center anyway. Once
we do that, we might as well go to a /21 in the Conference center as well.

It’s all RFC-1918, so it really doesn’t matter. I might move the WIFI into the 192.168 or 172.16 realms just to avoid messing up the existing plan with the 10.x networks.

Fortunately, nothing is set in stone yet, so this isn’t fatal. Good catch!

I just caught another issue that I’ll be emailing shortly. Unfortunately, the mistake I caught is a little more serious and has a greater ($$) impact. Namely that we might need optical modules for more of the switches than I expected.

Owen

kylerisse, 2018-02-16:
Thanks Owen, looks like this is going into PR#27.

closed 2018-02-16 by kylerisse