socallinuxexpo/scale-network #33: Enable and configure lldpd on LEDE

sarcasticadmin, 2018-02-21:
Enable and configure lldpd on LEDE from previous years on lede image build process

sarcasticadmin, 2019-03-06:
https://github.com/vincentbernat/lldpd/issues/24#issuecomment-11254453

When trying out the `lldpd` daemon with the juniper switches I was not able to get it to show neigbors via `lldpctl`:

```
~# lldpd -I eth0,eth1 -d
2019-03-06T19:03:47 [INFO/main] protocol LLDP enabled
2019-03-06T19:03:47 [INFO/main] protocol CDPv1 disabled
2019-03-06T19:03:47 [INFO/main] protocol CDPv2 disabled
2019-03-06T19:03:47 [INFO/main] protocol SONMP disabled
2019-03-06T19:03:47 [INFO/main] protocol EDP disabled
2019-03-06T19:03:47 [INFO/main] protocol FDP disabled
2019-03-06T19:03:47 [INFO/event] libevent 2.1.8-stable initialized with epoll method
2019-03-06T19:03:47 [INFO/privsep] unable to get system name
2019-03-06T19:03:47 [INFO/privsep] unable to get system name
2019-03-06T19:03:47 [INFO/privsep] unable to get system name
2019-03-06T19:03:47 [INFO/lldpctl] iface-pattern set to new value lo
2019-03-06T19:03:47 [INFO/privsep] unable to get system name
2019-03-06T19:03:47 [INFO/privsep] unable to get system name
2019-03-06T19:03:47 [INFO/lldpctl] description set to new value OpenWrt SNAPSHOT @ openwrt
2019-03-06T19:03:47 [INFO/lldpctl] lldpd should resume operations
```

```
~# lldpctl
-------------------------------------------------------------------------------
LLDP neighbors:
-------------------------------------------------------------------------------
```

davidelang, 2019-03-10:
need to change the interfaces in /etc/config/lldpd and figure out how to to get it out an interface

/etc/config/lldpd

config lldpd config
	option enable_cdp 1
	option enable_fdp 1
	option enable_sonmp 1
	option enable_edp 1

	
	option lldp_class 4
	option lldp_location "2:FR:6:Commercial Rd:3:Roseville:19:4"

	# if empty, the distribution description is sent
	#option lldp_description "OpenWrt System"

	# interfaces to listen on
	list interface "eth1"

PR #289 (2020-02-24): [READY] issue-33 - Adding last configs for scale 18x
## Description of PR

Fixes: #33 

lldpd was broken before since the default configuration was referencing
interfaces that didnt exist in our builds.

## Previous Behavior
* `lldpd` installed but not working

## New Behavior
* `lldpd` configured and working
* Added network.1 and wireless.1 for simple network config
* Adding new davids ssh key

## Tests
* lldpd works from junipers for openwrt:

```
root> show lldp neighbors
Local Interface    Parent Interface    Chassis Id          Port info          System Name
ge-0/0/6.0         -           [... cut]
closed 2020-02-24 by owendelong