socallinuxexpo/scale-network #43: Enable dhcpv6 options for lede APs

sarcasticadmin, 2018-03-01:
We should be getting dhcpv6 options from the dhcp server and configuring some of the lede config dynamically.

owendelong, 2019-02-28:
Moving to SCaLE 18x milestone.

owendelong, 2024-12-05:
If we can document the options desired here, I can probably produce the necessary JSON to get them into kea-dhcp6.

sarcasticadmin, 2024-12-09:
DHCP options: `224-226`. Its easy enough to move the options for keav6. Theyre defined here: https://github.com/socallinuxexpo/scale-network/blob/9aa9bc76ea80dd81e5a8de55bd5732f04b4f9ea1/facts/inventory.py#L474-L502

And well have to update the dhcp client config in openwrt. Probably enabling the dhcpv6 client since I think they are separate. As an example: the dhcpv4 options are parsed by openwrt here: https://github.com/socallinuxexpo/scale-network/blob/9aa9bc76ea80dd81e5a8de55bd5732f04b4f9ea1/openwrt/files/etc/udhcpc.user