socallinuxexpo/scale-network #45: Investigate incorporating additional wifi metrics in LEDE

sarcasticadmin, 2018-03-01:
To quote the `Make Wifi Fast" team:
```
Hmm, things that might be interesting:

- Distribution of client capabilities (5/2.4Ghz, MIMO mode, n/ac,
   achieved rates, etc. - however much data you can reasonably gather)

- Lasthop latency experienced by the clients (you could get this by
   capturing TCP handshakes and measuring the ACK-SYNACK delay, as shown
   in Figure 1 of this paper:
   https://dl.acm.org/citation.cfm?doid=2999572.2999603) - this could
   give an indication of bloat in the client drivers.

- QoS usage (how much traffic is sent on each of the VO/VI/BE/BK
   queues).

- The number of stations that have packets queued at the AP over time.

- Distribution of traffic between protocols (v4/v6, TCP/UDP/others).
```

Investigate what it would take to send these metrics to `zabbix` and/or `rsyslog`

davidelang, 2018-03-02:
On Thu, 1 Mar 2018, Robert James Hernandez wrote:

> - Distribution of client capabilities (5/2.4Ghz, MIMO mode, n/ac,
>   achieved rates, etc. - however much data you can reasonably gather)

we gather this indirectly in that we get the band a client connects to and the 
rates they use (the 2016 scripts)

> - Lasthop latency experienced by the clients (you could get this by
>   capturing TCP handshakes and measuring the ACK-SYNACK delay, as shown
>   in Figure 1 of this paper:
>   https://dl.acm.org/citation.cfm?doid=2999572.2999603) - this could
>   give an indication of bloat in the client drivers.

This involves a patch to the kernel, so I think we should skip that this year. 
Next year we can work with folks like this to have them provide us patches to 
run rather than trying to figure it out on our own.

> - QoS usage (how much traffic is sent on each of the VO/VI/BE/BK
>   queues).

this is probably in the mac80211 module /sys directories

> - The number of stations that have packets queued at the AP over time.

I have no idea wwhere to find this

> - Distribution of traffic between protocols (v4/v6, TCP/UDP/others).

In the past, we gathered this information from the firewall at the internet 
perimeter rather than on each device. This was done with iptables and snmp 
queries. I think a lot of this info is available in /proc or /sys, if so we 
could gather it with a script and forward it via syslog for post-show analysis.

owendelong, 2018-03-02:
> we gather this indirectly in that we get the band a client connects to and the
> rates they use (the 2016 scripts)
>
>
> This involves a patch to the kernel, so I think we should skip that this year. [...]

Unless someone wants to figure out how to pull it out of the MIBs on the Junipers, this isn’t likely to happen that way this year.

So if it can be pulled from /proc or /sys on the APs, I suggest that having more granular data might be useful in other ways as well.

Owen

sarcasticadmin, 2018-03-03:
Partially completed in #42

owendelong, 2019-02-28:
Moved to preconference prep for SCaLE 18x milestone.

sarcasticadmin, 2024-01-29:
This will be related to whats going on as part of #567

nixinator, 2024-02-01:
https://www.comparitech.com/net-admin/snmpwalk-examples-windows-linux/

you can probably pull the raw data with this, even without the MIB it will look 'raw'. Which maybe enough to gleam the data for later processing.

I've used similar stuff to https://github.com/mteg/braa

for doing 'interesting' things with snmp ...  :-)

nixinator, 2024-02-01:
A script on each AP to disassociate even or odd MAC Hardware addresses might be a crude way of getting clients to become sticky to certain AP's per room.

If there are 2 AP's per room, the law of averages that roughly half of the wireless clients will become sticky to 1 AP. 

It's crude, but it's simple.

sarcasticadmin, 2024-02-01:
> You can probably pull the raw data with this, even without the MIB it will look 'raw'. Which maybe enough to gleam the data for later processing.

At this time I think we already have enough data coming from the prometheus endpoint thats in the openwrt image. Plus the APs no longer have snmp #32 

> I've used similar stuff to https://github.com/mteg/braa

Thats pretty cool Ill keep that in mind should we need snmp things :)

> A script on each AP to disassociate even or odd MAC Hardware addresses might be a crude way of getting clients to become sticky to certain AP's per room.

Dissociation seems to be outside the scope of this issue. If you want to capture this in a separate issue thats fine, but lets keep this discussion to metrics.

davidelang, 2024-02-01:
polling a lot of data via snmp is not a great idea. It's a lot of overhead on 
both ends compared to haivng an agent that can send a bunch of data at once.

We still have the scripts that send data via syslog correct?

David Lang

On Thu, 1 Feb 2024, Robert James Hernandez wrote:

> At this time I think we already have enough data coming from the prometheus endpoint thats in the openwrt image. Plus the APs no longer have snmp #32
>
>
> Thats pretty cool Ill keep that in mind should [...]

sarcasticadmin, 2024-02-01:
yes those are on the image