socallinuxexpo/scale-network #108: printer with photocopier functionality

davidelang, 2018-03-10:
upgrade the printer to one with photocopier/scanner functionality. There are a lot of cases where we have hand-written notes (or notations on maps) and need to copy them for multiple people.

owendelong, 2019-02-28:
This isn't going to happen this year, but I will see what I can do about getting one next year.

Do team members have any strong opinions regarding ink-jet vs. laser?

From my perspective, I prefer ink-jet for the following reasons:

- Much less expensive, especially for color
- I find the output quality to be superior, but that's a personal preference thing.
- Large Format (11x17) and double-sided scanning can be achieved economically
- Lower cost per page for sparse prints, especially sparse color prints

The advantages to laser are:
- Ink gets old sitting in the printer unused and we might need to buy new ink tanks each year. This is less of an issue with toner in laser printers. (about $120 for a set of ink tanks)
- Sometimes have lower print speeds due to raster printing nature. Less of an issue in modern printers
- For dense print pages, can be a higher cost per page for printing. We don't tend to do a lot of dense prints.
- Some people prefer the output of laser printers.
- Ink Jet prefers papers with high clay content to reduce bleeding, can be more expensive than random plain paper

owendelong, 2019-02-28:
I'm personally very happy with my HP Officejet 7740 which offers A4/letter/legal duplex scanning from the document feeder and 11x17 scanning from the flatbed.

https://smile.amazon.com/HP-OfficeJet-Wireless-Printing-G5J38A/dp/B01JUCLLGK/ref=sr_1_3?_encoding=UTF8&hvadid=178481533861&hvdev=c&hvlocphy=1021148&hvnetw=g&hvpos=1t1&hvqmt=e&hvrand=16984261974441163053&hvtargid=kwd-204464428358&keywords=officejet+7740&qid=1551324689&s=gateway&sr=8-3&tag=googhydr-20

owendelong, 2019-09-06:
No significant feedback from team. I'll check the current offerings competitive to 7740 and then submit the request to Ilan/Bala.

owendelong, 2020-02-15:
Postponed to SCaLE 19x due to our large SCaLE 18x budget for carts and other more urgent things.

davidelang, 2023-03-30:
see #602

owendelong, 2023-09-13:
This was resolved with the donation of an OfficeJet 7740 by me this year. If that proves inadequate, we can open a new ticket and purchase a laser printer next year. HP has excellent options available now that would meet our needs in the $500 range.

closed 2023-09-13 by owendelong