socallinuxexpo/scale-network #112: Scratchpad for Postmortem

owendelong, 2018-03-11:
Use this issue to record any notes of things that went well or could be improved for next year to be reviewed during the postmortem later this month.

owendelong, 2018-03-11:
APs deployment location needs to be better coordinated/tracked so we know what got deployed where.

owendelong, 2018-03-11:
Room setup needs to use correct cable lengths rather than working down from longest cables.

owendelong, 2018-03-11:
Need to pick up additional patch cords in the 5, 7.5, 10, 15, 100, 120, 150 foot lengths. Need MANY of the shorter lengths and a few of each of the longer lengths.

owendelong, 2018-03-11:
Need faster/smoother/easier workflow for updating automation and deploying to switch(es).

owendelong, 2018-03-11:
Need to add security policies to routers.

owendelong, 2018-03-11:
PVLAN approach has a number of drawbacks. Return to VLAN per Vendor with VLAN creation on demand.

owendelong, 2018-03-11:
Clear communication to Vendors about Wire on request. Ideally requests submitted by one month prior to show are guaranteed. Requests submitted later are "best effort".

owendelong, 2018-03-11:
IPv6 Payment processor

davidelang, 2018-03-11:
We _really_ need a place for a testbed where we have all the pieces of the 
network connected together for a while before the show.

a single 1-2 day weekend does not cut it (even if it were in a place with enough 
power to run everything), we need things online for a couple of weeks with 
remote access for at least a skeleton network followed by a couple days of 
everything in one place so that everything can be updated prior to the week of 
the show.

davidelang, 2018-03-11:
fix sign memory leak

davidelang, 2018-03-11:
replace pi computers??
replace piB+ (512K) models with something with more ram

consider looking for something that supports 4K since the screens the venue is 
providing are 4K screens.

davidelang, 2018-03-11:
tune rsyslog parameters on APs to make sure it doesn't use too much ram (main 
queue size)

sbibayoff, 2018-03-15:
Tuning parameters for Zabbix in github.

sbibayoff, 2018-03-15:
Better network device naming scheme. Would prefer Building-Room-type_of_device_with_number , This way it will be easier to locate the device in a list.
Switches
- Expo-booth416-sw1 
- Conf-Rm211-sw1
- Ball-RoomF-sw1
AP's
- Conf-Rm103-AP1
- Ball-RoomA-AP3
- Expo-booth328-AP2
PI's
- Reg-Self-pi3
- Reg-Printer-pi4
- Reg-Desk-pi2
- Hall-West-pi3
- Hall-Expo-pi1
- Conf-upper-pi2

This way, just from the name, it would be easier tell someone where to check for a device.

sbibayoff, 2018-03-15:
Paper checklist to give out to physical layout crew. Includes what physical ports to use on switch for what. What needs to be done first. other?

davidelang, 2018-03-15:
On Thu, 15 Mar 2018, sbibayoff wrote:

> Better network device naming scheme. Would prefer Building-Room-type_of_device_with_number , This way it will be easier to locate the device in a list.

just the room id is enough to identify things, building is just noise (with the 
possible exception of the lobby locations)
ap102-2
apball-a-2
apreg-1

I agree that putting the instance number in front of the location was a horrible 
idea, but I didn't think it was worth going back and re-labeling everything 
after the issue was discovered.

I would also put the type of device first, we tend to care far more about the 
type of device than the location.

we never did try to name the raspberry pi locations.

sbibayoff, 2018-03-15:
Decide on OS distro for workstation and servers. Same? different types (desktop vs server?)? Deb vs rpm based?

kidicarus2, 2018-03-15:
Get the actual room layouts and room transition dates/times as soon as it
is decided.

Utalizing the desired AP layout plan from this year and have power dropped
to each ap location.

Put a moratorium on altering images for any hardware after 5pm on Monday.
There is no reason all hardware should not be on the floor and operational
by day end Tuesday.

Create specific set of ethernet cables for the expo floor drops so we don't
waste the long lengths.

I would like to take full control of tech shirt distribution and only give
them to the core team (not just any volunteer) To many people with tech
shirts that where unreliable when needed this year.

Due to complaints of noise from the switches, plan to place all switches
under the stage and/or as far away from audience as possible.

Have cable tester on site so if we do have to create new cables on the fly
we can test and label them accordingly.

Get the placement plan of AV hardware from AV team prior to setup for wire
placement planning.

Upgrade/build new carts for switches. (I know this is already planned, but
for safety of the hardware while moving cart around this is a must)

Have a prepacked set of wires needed for both capture the flag and the
install fest. (And any other events that might happen requiring a boat load
of shortish ethernet cables)

Ditto on what was said in regaurd to having a paper with what port numbers
are for what for each room.

Evolve the AP layout plan to include the name of each AP so they are
planned and placed right where they belong.

Develop a breakdown scheduale based on this years experience so i can
better cordinate with volunteers and not have people standing idle for
periods of time.

Organize all hardware needed for registration on to one cart.

Come up with something I can give to volunteers in appreciation for helping
(especially for the little kids that helped)

kidicarus2, 2018-03-15:
Test and label the dual ethernet cable runs. Maybe even make a few more.
Utalizing those in certain areas would have also saved us from wasting a
lot of the longer cables.

davidelang, 2018-03-17:
On Thu, 15 Mar 2018, kidicarus2 wrote:

> Get the actual room layouts and room transition dates/times as soon as it
> is decided.

part of the problem was that things weren't sent to us on the basis that they 
weren't "final" yet.

We need them as soon as they are proposed, we can work with small changes 
happening to the layouts FAR better than we can work based on "it's the same as 
last year, except where it isn't"

> Utalizing the desired AP layout plan from this year and have power dropped
> to each ap location.

it costs a LOT to drop power at a location ($85 per location based on what we 
were told our first year in PCC)

There may be a few cases where we really want them to drop power, but it's far 
easier to run an extension cord ourselves.

> Put a moratorium on altering images for any hardware after 5pm on Monday.
> There is no reason all hardware should not be on the floor and operational
> by day end Tuesday.

well, if the image doesn't work, it doesn't work.

> Create specific set of ethernet cables for the expo floor drops so we don't
> waste the long lengths.

what length do we need? we have lots of cables of different lenths, which ones 
should we not 'waste'

> Due to complaints of noise from the switches, plan to place all switches
> under the stage and/or as far away from audience as possible.

some are much noisier than others, we need to know which ones are noisy and use 
them in places where the noise doesn't matter.

not all rooms have places that are far from the audience, and in many cases it 
will take a LOT of extra cable to do so.

If we have too many that are noisy, we should check if the noise is in the fan 
tray or power supply and see if we can buy some extras of those and get quieter 
ones.

> Have cable tester on site so if we do have to create new cables on the fly
> we can test and label them accordingly.

yes

> Get the placement plan of AV hardware from AV team prior to setup for wire
> placement planning.

yes

> Upgrade/build new carts for switches. (I know this is already planned, but
> for safety of the hardware while moving cart around this is a must)
>
> Have a prepacked set of wires needed for both capture [...]

the number of cables for these varies from year to year (based on the layout 
details), how many cables of each length do you need for each of these?

and is it work having two extra bins for these special mixed length kits as 
opposed to just the bins for the different lengths (think cleanup, it's one 
thing to put things in the right bins for their lengths, it something more to 
put the right number of specific lengths in one bin and keep count.

> Organize all hardware needed for registration on to one cart.

we are close to that with the monitor cart, what would you remove from the 
monitor cart to make room for whatever else you want to put there?

davidelang, 2018-03-17:
On Thu, 15 Mar 2018, kidicarus2 wrote:

> Test and label the dual ethernet cable runs. Maybe even make a few more.
> Utalizing those in certain areas would have also saved us from wasting a
> lot of the longer cables.

I think we did test them, just not put labels on them.

We should also consider daisy-chaining through the APs. That would have been a 
problem this year due to an issue we never did track down, but with good 
monitoring (and actually knowing what's where) and the auto-shutdown of wifi, it 
could save a lot of long cable.

kidicarus2, 2018-03-17:
> Get the actual room layouts and room transition dates/times as soon as it
> is decided.

part of the problem was that things weren't sent to us on the basis that
they
weren't "final" yet.

We need them as soon as they are proposed, we can work with small changes
happening to the layouts FAR better than we can work based on "it's the
same as
last year, except where it isn't"

Super, just getting the most amount of info before we show up monday.

> Utalizing the desired AP layout plan from this year and have power dropped
> to each ap location.

it costs a LOT to drop power at a location ($85 per location based on what
we
were told our first year in PCC)

There may be a few cases where we really want them to drop power, but it's
far
easier to run an extension cord ourselves.

I heard this excuse already except the fact of the matter is there where so
many wire drops in everyroom for just the random laptop users it is just
that. An excuse. I would say 80% of the aps desired locations where already
close enough to power drops. It would behove us yo not take ilan up on his
ask for where we want power dropped like he asked us this year. (I didnt
have ap layouts in time or i would have made the request. )

> Put a moratorium on altering images for any hardware after 5pm on Monday.
> There is no reason all hardware should not be on the floor and operational
> by day end Tuesday.

well, if the image doesn't work, it doesn't work.

Rumor has it we had a working image prior to showing up but it was decided
to be rebuilt come sunday/monday.

> Create specific set of ethernet cables for the expo floor drops so we
don't
> waste the long lengths.

what length do we need? we have lots of cables of different lenths, which
ones
should we not 'waste'

This was a responce to owens comment on utalizing long lengths when shorter
ones will due. The expo floor had 10 cables that where at least 20 feet to
long each. Just pointing out if we created dedicated wires for that it
would save those for use elsewhere.

> Due to complaints of noise from the switches, plan to place all switches
> under the stage and/or as far away from audience as possible.

some are much noisier than others, we need to know which ones are noisy and
use
them in places where the noise doesn't matter.

not all rooms have places that are far from the audience, and in many cases
it
will take a LOT of extra cable to do so.

If we have too many that are noisy, we should check if the noise is in the
fan
tray or power supply and see if we can buy some extras of those and get
quieter
ones.

At this point in the game, cable length is not relavent nor will it take
"alot" of extra. We can make more if needed and we put the thing in the
corner of the room anywhere and there will be eally long cables and really
short cables needed. As far as the noise, i didnt find one i felt was
quite. If a day is planned where we can have a "how noisy" test then i can
plan what goes where accordingly.

> Have cable tester on site so if we do have to create new cables on the fly
> we can test and label them accordingly.

yes

> Get the placement plan of AV hardware from AV team prior to setup for wire
> placement planning.

yes

> Upgrade/build new carts for switches. (I know this is already planned, but
> for safety of the hardware while moving cart around this is a must)
>
> Have a prepacked set of wires needed for both capture the flag and the
> install fest. (And any other events that might happen requiring a boat
load
> of shortish ethernet cables)

the number of cables for these varies from year to year (based on the
layout
details), how many cables of each length do you need for each of these?

and is it work having two extra bins for these special mixed length kits as
opposed to just the bins for the different lengths (think cleanup, it's one
thing to put things in the right bins for their lengths, it something more
to
put the right number of specific lengths in one bin and keep count.

This would be for help with deployment not clean up. I assume we would know
the info of whats needed prior to monday.

> Organize all hardware needed for registration on to one cart.

we are close to that with the monitor cart, what would you remove from the
monitor cart to make room for whatever else you want to put there?

Thats what i figured, i still plan to hit the storage some time in april to
put brain power into how i want the carts organized.

davidelang, 2018-03-17:
On Sat, 17 Mar 2018, kidicarus2 wrote:

> < it costs a LOT to drop power at a location ($85 per location based on what we
>
>
>
> I heard this excuse already except the fact of the matter is there where so
> many wire drops in everyroom for just [...]

two years ago we spent a lot of time planning the power drops (we were told 
that we would not be allowed to plug in to any wall outlets, so if we needed to 
plug anything in, we needed to plan an outlet), and then we got there and the 
order for the drops was lost and we had to scramble on our second day there to 
go buy a lot of extension cables to get things plugged in.

They charge for the circuit, it costs just as much for a single outlet for the 
12w AP to plug into as for an entire string of outlets under the chairs, both 
are a single circuit.

No it doesn't make sense, but it's standard practice for PSAV.

> Rumor has it we had a working image prior to showing up but it was decided
> to be rebuilt come sunday/monday.

Rumor was incorrect. We thought we had a working image, but found problems (both 
things not implemented in the image, and a wierd packet loss on pings that only 
showed up when connecting the AP switch to the Juniper switch) and so spent two 
plus days scrambling to get something working.

> This was a responce to owens comment on utalizing long lengths when shorter
> ones will due. The expo floor had 10 cables that where at least 20 feet to
> long each. Just pointing out if we created [...]

I don't disagree with the idea of using the right length for the job, I'm just 
cautious about creating special kits for rooms.

> At this point in the game, cable length is not relavent nor will it take
> "alot" of extra. We can make more if needed and we put the thing in the
> corner of the room anywhere and there will be eally [...]

> This would be for help with deployment not clean up. I assume we would know
> the info of whats needed prior to monday.

ok, with the new switch carts, we may have some space for pre-planned bins for a 
few locations.

We will need to figure this out well before monday to pack the bins.

Two years ago, Mike spent a lot of time in CAD measureing cable runs and 
planning just the right lengths. This made the cabling in the rooms look good, 
but then when we didn't have power where we planned, it meant that we couldn't 
adjust. I'm not sure it made things go any faster (anyone from two years ago 
want to comment on this)

If it helps, go for it. I'm just concerned about it being a lot of effort for 
low reward

> Thats what i figured, i still plan to hit the storage some time in april to
> put brain power into how i want the carts organized.

sounds good. better to plan things out _before_ we build the new carts :-)

sbibayoff, 2018-03-17:
Hello,

On Fri, Mar 16, 2018 at 5:06 PM, David Lang <notifications@github.com>
wrote:

> On Thu, 15 Mar 2018, kidicarus2 wrote:
>
> > Test and label the dual ethernet cable runs. Maybe even make a few more.
> > Utalizing those in certain areas would have also saved us from wasting a
> > lot of the longer cables.
>
> I think we did test them, just not put labels on them.
>

Those double cables where tested, but none of them really "passed". If I
tested each cable separately, they passed. But when I hooked up the loop
back device on one cable, and tested on the other cable, I got some
measurement(cable length & pass through), indicating some type of cross
signaling. It's either at the junction box (which I suspect), or the wires
being too close to each other. Or, it could just be the cheap tester.
Anyway, this needs to be closer looked at.

Steve

>

owendelong, 2018-03-17:
> two years ago we spent a lot of time planning the power drops (we were told
> that we would not be allowed to plug in to any wall outlets, so if we needed to
> plug anything in, we needed to plan an [...]

Technically, this being electrical, it is handled by Edlen, not PSAV.

> Rumor was incorrect. We thought we had a working image, but found problems (both
> things not implemented in the image, and a wierd packet loss on pings that only
> showed up when connecting the AP [...]

Let’s not spend a lot of time trying to poke holes in the decisions that were made
about how to build things. Instead, if there are lessons to be learned, let’s
identify those and move forward.

Let us focus on fixing problems, not fixing blame.

> I don't disagree with the idea of using the right length for the job, I'm just
> cautious about creating special kits for rooms.

My comment was focused more on the use of unnecessarily long cables used in some
of the conference rooms (which I admittedly based on second-hand reports and a
small number of personal observations) rather than the (admittedly over-long)
cables I deployed on the show floor.

By the time we were wiring the show floor, we actually had no further need for
significantly long cables (other than a couple of cases on the show floor which
arose due to incomplete switch configuration loads).

>

I made a point of visiting most of the rooms while they were idle during various
parts of the show.

While none of the switches are silent (you can’t have a 48 Port POE switch
which moves gigabit packets non-blocking at wire speed and not have to provide
some cooling), most of them are not actually all that loud.

The only rooms I found to be significantly noisy were BallroomG, Room 107,
and the room where Stu gave his talk (211 IIRC).

In the others, the switch noise was barely noticeable in an empty room until
you were very close to the switch (which in most cases required being present
at the mixer).

> ok, with the new switch carts, we may have some space for pre-planned bins for a
> few locations.

My thinking is that we can put cable loads for certain special purposes in large
ziplock bags during a preparation party. This wouldn’t involve extra bins, but
simply separating the cable loads out ahead of time.

Cleanup would restore them to their original bins and we’d simply pack the ziplock
bags for re-use during next year’s prep.

> We will need to figure this out well before monday to pack the bins.
>
> Two years ago, Mike spent a lot of time in CAD measureing cable runs and
> planning just the right lengths. This made the cabling [...]

It did make things go somewhat faster, but now we also know what we’re up against
in terms of electrical and we should be able to plan that right along side the
network cabling requirements. As such, I think it’s worth while to try again.

> If it helps, go for it. I'm just concerned about it being a lot of effort for 
> low reward

Fair point and it might be. Let’s try it and see what lessons we learn.

> sounds good. better to plan things out _before_ we build the new carts :-)

Amen!

Owen

owendelong, 2018-03-17:
> On Mar 16, 2018, at 18:16 , sbibayoff <notifications@github.com> wrote:
> 
> Hello,
> 
> On Fri, Mar 16, 2018 at 5:06 PM, David Lang <notifications@github.com>
> wrote:
> 
> > On Thu, 15 Mar 2018, kidicarus2 wrote:
> >
> > > Test and label the dual ethernet cable runs. Maybe even make a few more.
> > > Utalizing those in certain areas would have also saved us from wasting a
> > > lot of the longer cables.
> >
> > I think we did test them, just not put labels on them.
> >
> 
> Those double cables where tested, but none of them really "passed". If I
> tested each cable separately, they passed. But when I hooked up the loop
> back device on one cable, and tested on the other cable, I got some
> measurement(cable length & pass through), indicating some type of cross
> signaling. It's either at the junction box (which I suspect), or the wires
> being too close to each other. Or, it could just be the cheap tester.
> Anyway, this needs to be closer looked at.
> 
> Steve
> 

I’ve never really liked those assemblies and I’m not surprised they had
questionable test results. Turns out in a couple of cases, they saved
our bacon this year, but having some duplex couplers would have worked
just as well as those did and would have offered additional flexibility.

I’m fine with testing them if people want to, but unless we get really
good results, my vote would be to retire them and buy a batch of duplex
couplers to replace them.

Owen

owendelong, 2019-02-26:
I'm going to go through and add emojis to items in this issue and then close it.

The thumbs up emoji will indicate "issue addressed".

The thumbs down emoji will indicate "issue deferred to next year" and should serve as a flag for copying the issue into the new scratchpad issue ticket I will open and or to a new separate issue.

The rocket ship emoji will be a marker that I need to turn this item into a separate issue and ask someone to work on it urgently before SCALE 17x.

If I encounter other corner cases, I may apply other emojis and document them in a reply to this item.

owendelong, 2019-02-26:
> I'm going to go through and add emojis to items in this issue and then close it.
>
> The thumbs up emoji will indicate "issue addressed".
>
> The thumbs down emoji will indicate "issue deferred to next [...]

The eyeball emoji will represent an item where the status is currently unknown. If you have knowledge, please apply the proper emoji from the selection above.
> 
> If I encounter other corner cases, I may apply other emojis and document them in a reply to this item.

owendelong, 2019-02-26:
> Get the actual room layouts and room transition dates/times as soon as it is decided. Utalizing the desired AP layout plan from this year and have power dropped to each ap location. Put a moratorium [...]

Steven, could you please break any of these items that still need to be addressed out into individual issues. Trying to deal with this as a monolith in one comment is too difficult.

Thanks!

owendelong, 2019-02-26:
> The eyeball emoji will represent an item where the status is currently unknown. If you have knowledge, please apply the proper emoji from the selection above.

The sad face emoji is used to denote comments which simply are no longer relevant and/or fall into the "won't fix" category.

> 
> > If I encounter other corner cases, I may apply other emojis and document them in a reply to this item.

closed 2019-02-26 by owendelong