socallinuxexpo/scale-network #136: Scratchpad for SCaLE 17x Postmortem issues

owendelong, 2019-02-26:
This is a new scratchpad ticket. It's intended to allow us to quickly capture items we wish to discuss during the post mortem after the show.

RULES for this issue:

1. ONE ITEM and ONLY ONE ITEM per comment. PLEASE start a new comment for each topic.
2. NO discussion -- If a topic(comment in this issue) warrants an in-issue discussion, start by creating an  issue, then move the discussion into that issue. If you want to provide a pointer to the new issue as a hyperlink within the comment text, that is allowed, even encouraged. 
3. Identify and fix problems, not blame. History can be useful for learning what happened and identifying improvements. It is not about looking back for who was at fault.

davidelang, 2019-03-06:
see if we can get a few radios for tech earlier, cell/wifi doesn't work in the basement area

owendelong, 2019-03-06:
For next year buy at least one of these:
https://www.amazon.com/GaffTech-G20NN0-Applicator-Complete-Extension/dp/B0171JYP8K/ref=sr_1_1?keywords=gaffgun&qid=1551896019&s=gateway&sr=8-1
(Test first one and consider buying second one)

kylerisse, 2019-03-07:
wire maps deployed with switches to avoid having to check with NOC for port assignments

kylerisse, 2019-03-07:
request monitors be deployed earlier than Wednesday evening for pi/sign validation

kylerisse, 2019-03-07:
monitoring should be up as soon as possible with switches already defined, switch installations should be validated against monitoring status when deployed

kylerisse, 2019-03-07:
need way to track people who have requested T-shirts, one person didn't' remember if they filled out the form or not plus no audit trail leaves room for abuse

sbibayoff, 2019-03-08:
If we don't have the switches/AP's ready by Monday, have paper taped to where they would be placed. So physical network team could place cables(with correct length) where they are needed. This way there is a lesser chance of cable being placed in a wrong place.

sbibayoff, 2019-03-08:
Have a print out instruction set (1 page) for the physical layout crew which includes port assignments on switches.
example (needs modification):
1) places devices(switches & APs) in correct places
2) connect power & appropriate network cables between devices and switches (on correct ports, list on sheet)
3) Verify, with noc, that the device is seen is good
4) Tape down, after all devices are verified.

Move on

sbibayoff, 2019-03-08:
An idea, 

Have radios available on Monday, tech is in charge of them till AV shows up on Tuesday, then hand them over tracking and everything else to AV on Tuesday.

sbibayoff, 2019-03-27:
Maybe third time I'm bringing this up would be the charm,

Label the AP's with their mac address on top. In a large enough size so it could be easily read without having to pick up the AP.

Or, maybe better yet, have a label with a Scale produced s/n , where we would have a list of s/n to mac address.

sbibayoff, 2019-03-27:
Have power "bars" in the noc. Something like:
https://www.frys.com/product/5960984?site=sr:SEARCH:MAIN_RSLT_PG

Instead of power strips daisy changed and cords all tied up in knots, have 2, 3, or 4 bars on the main table (each plugged into a different outlet/circuit).

Hopefully it would make the cords more manageable.

sbibayoff, 2019-03-28:
Allocate more physical ports on the switches for AP's. We where running out of AP ports in the IDF's for the AP't that where plugged into the walls, hallways, and such.

I think it would be easier to add a few more ports than configure 3 unique switches.

owendelong, 2019-03-28:
That’s because we plugged way more APs into the hallways than were supposed to be. e.g. Registration, there were 3 APs that should have been in the Reg switch that went to wall ports for some reason.

Owen

hriday, 2019-03-28:
I don't think they were plugged into the walls either. I picked up 3 AP's
at reg that were connected to power but had no ethernet lines run.

It was probably missed, but we can get better monitoring next year. Kate
has some amazing diagrams now that Kyle and Steven have enhanced.

owendelong, 2019-03-28:
Use a single AP Naming convention for Maps, APs, host files, etc.

owendelong, 2019-03-28:
NOC Power -- Bring in power box and multiple 20A circuits

davidelang, 2019-03-28:
big TV for the NOC for monitoring

owendelong, 2019-03-28:
Set up and test flashing stations prior to show. (tested server OS and software)

owendelong, 2020-01-08:
> Have a print out instruction set (1 page) for the physical layout crew which includes port assignments on switches.
> example (needs modification):
>
> 1. places devices(switches & APs) in correct places [...]

This is harder to automate, so it is not addressed by #265 However, I will have printed portmaps ready to tape to the top of each switch this year. PostScript files are auto-generated by the switch configuration code at the same time as the switch configuration files as part of the code in #265.

sarcasticadmin, 2021-03-24:
Closing as this scratchpad has served its purpose

closed 2021-03-24 by sarcasticadmin