socallinuxexpo/scale-network #138: Renumber and relabel switches

owendelong, 2019-02-28:
Currently the switch types file is in no logical order by anything rational because the switch numbering doesn't correspond in any way to loudness groups.

I'd like to renumber and relabel the switches into an order that places them in 3 groups (loud, normal, quiet), then reorder the switch types file accordingly.

It does _NOT_ make sense to add this to our workload for the conference, however.

sarcasticadmin, 2022-11-22:
@owendelong during the last team call you mentioned this was achieved already but I dont remember it being associated with any given PR.

Any context would be greatly appreciated, leaving this open for now.

sarcasticadmin, 2023-03-23:
Closing as I believe this has already been done or is no longer an issue.

closed 2023-03-23 by sarcasticadmin