socallinuxexpo/scale-network #143: Enable perl validations in circleci process

sarcasticadmin, 2019-03-01:
Enable perl validations in `circleci` test process:

The simplest thing we can tell is ``is it valid?'' and enable warnings. For example:
```
perl -Mstrict -Mdiagnostics -cw switch_template.pl
```

If we want we can also run this across all `*.pl` files in the switch-configuration dir:
```
find ./switch-configuration/ -iname "*.pl" -exec perl -Mstrict -Mdiagnostics -cw {} \;
```

We get a few warnings but syntax checks out (which is what we care about)
```
\1 better written as $1 at
        ./switch-configuration/config/scripts/switch_template.pl line 301 (#1)
    (W syntax) Outside of patterns, backreferences live on as variables.
    The use of backslashes is grandfathered on the right-hand side of a
    substitution, but stylistically it's better to use the variable form
    because other Perl programmers will expect it, and it works better if
    there are more than 9 backreferences.
    
"my" variable $net masks earlier declaration in same scope at
        ./switch-configuration/config/scripts/switch_template.pl line 478 (#2)
    (W shadow) A "my", "our" or "state" variable has been redeclared in the
    current scope or statement, effectively eliminating all access to the
    previous instance.  This is almost always a typographical error.  Note
    that the earlier variable will still exist until the end of the scope
    or until all closure references to it are destroyed.
    
./switch-configuration/config/scripts/switch_template.pl syntax OK
./switch-configuration/config/scripts/build_switch_configs.pl syntax OK
```

Well need to make sure we have a container that has `perl5` installed.

This isnt something we need immediately but I think it would really be nice to have

kylerisse, 2019-03-01:
We could also run the script(s) to make sure there’s exits 0. That would potentially validate issues with parsing the tsv files. A reason I run inventory.py in the ansible tests at the end is to catch breaking changes in one of those tsvs. Happened quite a few times last year during the show.

sarcasticadmin, 2019-03-01:
@kylerisse thats a good suggestion, i think we should definitely include it in this

owendelong, 2019-03-01:
+1, but you'll need to find new lint examples as the ones above have been rectified. 😛

sarcasticadmin, 2019-03-02:
@owendelong dont worry with the work we did on #155 and #156 circleci will be sure to tell you! :shipit:

closed 2019-03-02 by sarcasticadmin