socallinuxexpo/scale-network #144: Make sure Changes from branch od.17.01 don't break Ansible integration

owendelong, 2019-03-01:
Kyle,

In checking in od.17.01 (or was it OD.17.01?), I changed the content of some of the tsv files. I don't think I did anything that will break ansible and you can look at the diffs for the README file that describes them to see what changed (I did update all the documentation too).

Please make sure, however, that I didn't break any of your scripts with my changes and/or update your scripts accordingly.

kylerisse, 2019-03-01:
I did check those and looks good. The columns in switch types are the same, just with an added column. Ansible wasn't using anything in the booth_list so we're good there. Also verified the json being spit out looks sane. https://circleci.com/gh/socallinuxexpo/scale-network/90 

The DHCP for the vendors was handled by the router last time. I wont need to do anything if we follow that model again.

owendelong, 2019-03-01:
> On Feb 28, 2019, at 20:19 , kylerisse <notifications@github.com> wrote:
> 
> I did check those and looks good. The columns in switch types are the same, just with an added column. Ansible wasn't using anything in the booth_list so we're good there. Also verified the json being spit out looks sane. https://circleci.com/gh/socallinuxexpo/scale-network/90 <https://circleci.com/gh/socallinuxexpo/scale-network/90>
> The DHCP for the vendors was handled by the router last time. I wont need to do anything if we follow that model again.
> 
> 
Yeah, we didn’t follow that model this time.

You’ll need to generate DHCP for the Vendor VLANs built by the combination of VVRNG and VVLAN directives.

It’s a bit of a complex process (sorry) and I’m happy to talk it over with you on the hangout if you have questions.

Owen

kylerisse, 2019-03-01:
Seems strait forward enough. Is there already a working implementation of this in the perl scripts? When I run build_switch_configs.pl from master I don't see any vendor VLANs defined in Expo switches. My current assumption is that I will not until booth_list.tsv is populated and that functionality is added. Is that something #150 addresses?

owendelong, 2019-03-01:
Yes.. There is… I find it interesting that you don’t see them… When I run it, there definitely is.

If you look at output/Expo1A.conf, I see the following matching lines:

kiev:owen (360) ~/SCALE/networking/scale17x/scale-network/switch-configuration/config % grep -n -i vendor output/ExpoA1.conf                         2019/03/01 11:39:24
85:        description "TRUNK Port (exSCALE-SLOW exSCALE-FAST exSpeaker exInfra exAVLAN exSigns exStaff exDWCC vendor_backbone)";
89:                vlan members [ exSCALE-SLOW exSCALE-FAST exSpeaker exInfra exAVLAN exSigns exStaff exDWCC vendor_backbone ];
94:        description "TRUNK Port (exSCALE-SLOW exSCALE-FAST exSpeaker exInfra exAVLAN exSigns exStaff exDWCC vendor_backbone)";
98:                vlan members [ exSCALE-SLOW exSCALE-FAST exSpeaker exInfra exAVLAN exSigns exStaff exDWCC vendor_backbone ];
224:            description "Vendor VLAN 200"
228:                    members vendor-vlan-200;
235:            description "Vendor VLAN 201"
239:                    members vendor-vlan-201;
246:            description "Vendor VLAN 202"
250:                    members vendor-vlan-202;
257:            description "Vendor VLAN 203"
261:                    members vendor-vlan-203;
268:            description "Vendor VLAN 204"
272:                    members vendor-vlan-204;
279:            description "Vendor VLAN 205"
283:                    members vendor-vlan-205;
290:            description "Vendor VLAN 206"
294:                    members vendor-vlan-206;
301:            description "Vendor VLAN 207"
305:                    members vendor-vlan-207;
312:            description "Vendor VLAN 208"
316:                    members vendor-vlan-208;
323:            description "Vendor VLAN 209"
327:                    members vendor-vlan-209;
334:            description "Vendor VLAN 210"
338:                    members vendor-vlan-210;
345:            description "Vendor VLAN 211"
349:                    members vendor-vlan-211;
356:            description "Vendor VLAN 212"
360:                    members vendor-vlan-212;
367:            description "Vendor VLAN 213"
371:                    members vendor-vlan-213;
378:            description "Vendor VLAN 214"
382:                    members vendor-vlan-214;
389:            description "Vendor VLAN 215"
393:                    members vendor-vlan-215;
609:            group vendors {
653:        group vendors {
981:    vendor_backbone {
982:        description "Vendor Backbone";
986:    vendor-vlan-200 {
990:    vendor-vlan-201 {
994:    vendor-vlan-202 {
998:    vendor-vlan-203 {
1002:    vendor-vlan-204 {
1006:    vendor-vlan-205 {
1010:    vendor-vlan-206 {
1014:    vendor-vlan-207 {
1018:    vendor-vlan-208 {
1022:    vendor-vlan-209 {
1026:    vendor-vlan-210 {
1030:    vendor-vlan-211 {
1034:    vendor-vlan-212 {
1038:    vendor-vlan-213 {
1042:    vendor-vlan-214 {
1046:    vendor-vlan-215 {

Lines 85, 89, 94, 98 are trunk definitions, so don’t really count.
Lines 224-393, OTOH, are the physical interfaces being mapped onto Vendor VLANs.
Not matching the GREP, but also present in the file starting at line 406 are the Vendor VLAN Layer 3 interface definitions.
Lines 609 an 653 are the DHCPv6 and DHCPv4 group definitions for the vendor VLANs
Lines 986 are the actual VLAN definitions.

Hope that helps.

Owen

kylerisse, 2019-03-01:
I see now, thanks for clarifying. All I care about is the layer 3 interfaces anyways so this is good info. They'll also need DHCP helper or whatever juniper calls it. 

So what it boils down to is you're looking for the `VV_get_prefix4` and `VV_get_prefix6` sub routines to be re-implemented in `inventory.py` so that it derives the same layer 3 info as the perl essentially?

Is there any chance of convincing you to run DHCP services on the switch and generate the pools on the device that's hosting that L3 gateway at the same time the interfaces are generated? The same data would build both so they'd be guaranteed consistent. Pardon my ignorance as maybe that's not possible with this gear, but seems a bit more reliable to lean on a single dynamic source of truth, a bit less brittle to change.

owendelong, 2019-03-01:
> On Mar 1, 2019, at 12:29 , kylerisse <notifications@github.com> wrote:
> 
> I see now, thanks for clarifying. All I care about is the layer 3 interfaces anyways so this is good info. They'll also need DHCP helper or whatever juniper calls it.
> 
> 

That’s the dhcp-relay portion of the configuration.
> So what it boils down to is you're looking for the VV_get_prefix4 and VV_get_prefix6 sub routines to be re-implemented in inventory.py so that it derives the same layer 3 info as the perl essentially?
> 
Yes… I guess that’s what you’ll need to do to get the correct DHCP ranges.

Alternatively, I don’t think you care about associating the addresses with VLANs per se, so you could just build configurations based on splitting the specified IPv6 and IPv4 prefixes into /64s and /24s, respectively and then treating those identically to the static specifications on the VLAN directives for normal VLANs.

> Is there any chance of convincing you to run DHCP services on the switch and generate the pools on the device that's hosting that L3 gateway at the same time the interfaces are generated? The same [...]
Yeah, I could do that, but I thought we might want the visibility afforded by the single DHCP server in each building.

Alternatively, the dynamic algorithm used in VV_get_* is very simplistic and easy to reliably replicate…

VLID = $VLAN_BASE + $instance
VLIPv4 = We convert the IPv4 Base address to a 32-bit value and then add the ($instance*256) to that
			(in order to generate sequential /24s) and then convert that back to a dotted quad.

VLIPv6 = Similar to IPv4, we use the base address specified and add $instance to the last quartet
		in the /64 to achieve a prefix.

So in all cases that you care about (the exception is the host address assigned on the vendor_backbone
VLAN, you’re essentially offsetting the supplied base value by the instantiation number.

The reason the configuration file has ranges is to allow sanity-checking that I’m not overrunning into
resources reserved for another purpose. (Which I just noticed the IPv4 and IPv6 checks are failing and I actually
need to expand the prefix to a /15 and a /54 after I figure out why I wasn’t getting errors from that).

Considering this statement:
VVRNG	ex_v_			200-498	2001:470:f325:200::/55	10.1.0.0/16	Dynamically allocated and named booth VLANs

The second field is currently ignored (there’s already an issue for this).

SO VLAN_BASE is (in this case) 200, but I limit the max VLID to 498 (499 is the vendor_backbone).

2001:470:f325:200::/55 goes from 2001:470:f235:200::/64 to 2001:470:f325:3ff (theoretically), but we’re BCD-encoding the
instance number offset from 200 (so it ends up being identical to the VLID, but software should assume that will always be the case).

10.1.0.0/16 goes from 10.1.0.0/24 to 10.1.255.0/24. In practice, we end up with 10.1.<VLID-200>.0/24 for each VLAN.

kylerisse, 2019-03-01:
Alright thanks for the feedback @owendelong . Very clear.

My concern isn't the complexity of the VV_get_ subroutines but the potential for them to change, for cases where we need to resize the vendor VLANs and/or their supernet(s). 

Understood on the dual DHCP servers, but if their gateway is down they wont get an address either way. Totally agree though that the centralized logging with not VVRNG VLANs is a more ideal situation, makes sense.

I'll probably lean towards generically pre-populating 10.1.<VLID-199>.0/24 and 2001:470:f325:<VLID>::/64 or whatever the equivalent calculation becomes. I'm going to hold off on this issue until #153 has more info and the new VVRNG ideas are fleshed out a bit.

owendelong, 2019-03-02:
> On Mar 1, 2019, at 14:23 , kylerisse <notifications@github.com> wrote:
> 
> Alright thanks for the feedback @owendelong <https://github.com/owendelong> . Very clear.
> 
> My concern isn't the complexity of the VV_get_ subroutines but the potential for them to change, for cases where we need to resize the vendor VLANs and/or their supernet(s).
> 
> 
If we resize the supernets, that would be in the config file.

The v6 subnet size won’t change… I’m pretty sure we won’t move away from /64s for any IPv6 subnet.

The v4 subnets could change, but since it’s private address space, it shouldn’t need to. Further, if I had my way, we’d simply eliminate v4 from these networks. ;-)
> Understood on the dual DHCP servers, but if their gateway is down they wont get an address either way. Totally agree though that the centralized logging with not VVRNG VLANs is a more ideal situation, makes sense.
> 
Sure, but that’s true of ALL of our LAN segments. It’s not unique to these.
> I'll probably lean towards generically pre-populating 10.1..0/24 and 2001:470:f325:::/64 or whatever the equivalent calculation becomes. I'm going to hold off on this issue until #153 <https://github.com/socallinuxexpo/scale-network/issues/153> has more info and the new VVRNG ideas are fleshed out a bit.
> 
That's done with the PR for od.17.04 which should be merged shortly if it hasn’t been already. I agree that the most logical thing to do on the DHCP side is just pre-populate all the possible subnets.

Note: When enumerating the IPv6 prefix, we won’t be using any subnets that don’t fit within BCD in the bits between the specified supernet prefix length and /64. In other words, you can do 200, 201, … 209, 210, 211, … 219, 220, etc., but you don’t need to do 20a, 20b, …, 20f or 21a, …, 21f, etc.

This is done so that the IPv6 prefix still lines up with the VLAN number even though the IPv4 prefix cannot.

Owen

kylerisse, 2019-03-02:
Yea we're on the same page here with all this. The only thing I'm confused about is given a potential range of 200-498 specified on the VVRNG line in Expo when I review the output from the last push to master why don't I see anything past 455? For example would I expect to see a 456 with 10.3.0.1/24 and 2001:470:f325:456::1/64?

this is the last interface I can find in the 400 range:

`unit 455 {
            family inet {
                10.2.255.1/24;
                filter input only_to_internet;
                filter output only_from_internet;
            }
            family inet6 {
                2001:470:f325:455::1/64;
                filter input only_to_internet6;
                filter output only_from_internet6;
            }
        }`

I guess the simplified question is, were I to just omit parsing VVRNG this year and populate 200-455 using very basic rules, would that harm us? Are we expecting to see VLANs 456-498 show up? Is this the expected output from the command? 

I'm basing these question on what circle spit out at https://circleci.com/api/v1.1/project/github/socallinuxexpo/scale-network/131/output/104/0?file=true and unsure if that's what you're seeing locally. 

I'm going to proceed with the assumption that we might see vlans past 455 and IPs into the 10.3.x.x range, supporting 200-498. Just confused why I don't see them already. Could be something basic I'm missing, apologies if that's the case. Either way I think the incoming fix will be valid.

owendelong, 2019-03-02:
Because I only generate enough to satisfy the VVLAN directives in types.d/* based on the type field contents in switchtypes.

kylerisse, 2019-03-02:
haha lucky to hit 256 exactly. We can close this once #158 is merged.

owendelong, 2019-03-03:
Looking at the code in #158 (sorry it took me a while to get to that), it looks like you're parsing the VLAN directive instead of the VVLAN directive for Vendor VLANs.

Please note the difference between (e.g. config/types/cfRoom (VLAN directive for conventional VLANs) and config/types/Booth (VVLAN directive for vendor VLANs)).

In the case of the VLAN directive, it specifies a particular VLAN and a number of ports as arguments.

The VVLAN directive specifies only a number of ports and each one is placed into a separate Vendor VLAN.

The Vendor VLANs are unique across all switches, so the order in which the switches are parsed matters. The PERL script sorts them (see get_switchlist in switch_template.pl) into alphabetical order if it's doing the entire set. If the user supplies the list of switches, they are done in the order the user specified. (This probably should be logged as a bug. It is noted in the README file for now).

kylerisse, 2019-03-03:
Please refer back to the code and testing. This is all based on parsing `vlans.d` files which are either `VLAN` or `VVRNG`. Feel free to refer to the output of inventory.py at https://circleci.com/gh/socallinuxexpo/scale-network/139 or the documented testing steps in the PR if you think something was missed and file any relevant bugs. If you are looking for fundamental re-design of what is parsed then that can be tracked in #149 for post 17x.

Here is a sample of a generated VLAN:
`        
          {
          "name": "ex_v_379",
          "id": 379,
          "ipv6prefix": "2001:470:f325:379::",
          "ipv6bitmask": 64,
          "ipv4prefix": "10.2.179.0",
          "ipv4bitmask": 24,
          "building": "Expo",
          "description": "Dyanmic vlan 379",
          "ipv6dhcp1a": "2001:470:f325:379:1::1",
          "ipv6dhcp1b": "2001:470:f325:379:0fff::fffe",
          "ipv6dhcp2a": "2001:470:f325:379:f000::1",
          "ipv6dhcp2b": "2001:470:f325:379:ffff::fffe",
          "ipv4dhcp1a": "10.2.179.10",
          "ipv4dhcp1b": "10.2.179.128",
          "ipv4dhcp2a": "10.2.179.129",
          "ipv4dhcp2b": "10.2.179.254",
          "ipv4router": "10.2.179.1",
          "ipv4netmask": "255.255.255.0",
          "ipv6dns1": "2001:470:f325:103::5",
          "ipv6dns2": "2001:470:f325:503::5",
          "ipv4dns1": "10.0.3.5",
          "ipv4dns2": "10.128.3.5"
        },
`

closed 2019-03-03 by kylerisse

owendelong, 2019-03-03:
Ah... Got it... Yeah, if you're not parsing the types/* files, then that makes sense. My bad.