socallinuxexpo/scale-network #149: ansible/inventory.py ignores directives and makes invalid assumptions

owendelong, 2019-03-01:
In pr#148, the change brings to light the fact that the VLANs files are not being parsed correctly by ansible/inventory.py.

The intent is to parse the "vlans" file only. If that file contains "#include" directives, then only files identified by those directives should be parsed. The switch configuration generator will not process files from "vlans.d" which are not referenced in such a manner and neither should inventory.py.

Further, the directives in VLAN configuration files (whether the master "vlans" file or one of the included files from "vlans.d/" (currently) fall into two categories with different syntax.

Currently, inventory.py simply assumes that every directive is "VLAN" which is invalid. The syntax for a "VVRNG" specification is different and serves a different purpose. This line should not be commented out as it affects the config generation for switches affected by a file in "types/" which contains a "VVLAN" directive (currently only "types/Booth").

The correct approach is to compare the first field of each line (directive) in the file to a known directive understood by the parser. If the directive in question is not understood, the safest action for the parser to take is to emit a warning and ignore the line.

owendelong, 2019-03-01:
See https://github.com/socallinuxexpo/scale-network/blob/master/switch-configuration/README.md for complete documentation of the configuration files and how they should be parsed and interpreted as well as documentation of all valid directives.

kylerisse, 2019-03-02:
Good call, we should process the file as intended. This should come after #144 and #153 .

kylerisse, 2019-03-02:
Suggestions for added error handling https://github.com/socallinuxexpo/scale-network/pull/158#discussion_r261841739

owendelong, 2019-09-06:
I believe this is resolved... Kyle, can you verify and if so, close this issue?

PR #272 (2020-01-22): [READY] Issues 149 and 258
## Description of PR
Fixes #149 
Fixes #258 

Fixes for inventory.py related issues.

## Previous Behavior
- inventory.py didn't parse the vlans source files as they were intended, didn't support the #include directive
- csv files were not git beautified
- pylint was set to a less strict mode due to inefficiencies in inventory.py

## New Behavior
- inventory.py now properly parses the vlans as intended and supports #include per spec
- csv files are git beautified
- pylint is now set to default strict mode

## Tests
- examining the inventory.py file output using jq, c [... cut]
closed 2020-01-22 by sarcasticadmin