socallinuxexpo/scale-network #153: VVRNG IPv6 Prefix sanity check isn't working

owendelong, 2019-03-01:
A /55 is too small for our current utilization. Expanding to a /54 won't be an issue and the configurations that are resulting will work for now, but this bug should be fixed.

owendelong, 2019-03-01:
Turns out the v4 checks aren't working either and we blew past the /16. This actually is a problem and will require some renumbering, some software assumption changes, and some creativity to resolve.

owendelong, 2019-03-02:
Resolved and tested in branch od.17.04 -- Pushing now.

owendelong, 2019-03-02:
This whole vendor VLAN process is still horribly hacky and should be replaced.

closed 2019-03-02 by owendelong

davidelang, 2019-03-02:
what we did in the past was to allocate the booth number + 100 as the VLAN 
number and used the booth number as the 3rd octet in the IPv4 subnet

do we have a map of the show floor so that we can map what booth numbers are on 
what switch?

David Lang

owendelong, 2019-03-02:
There’s a text file which actually does this (sort of) in the repository.

Look at the switch-configuration/config/booth_list.tsv file.

The switch names now specify the booth group. Starting from the group closest to the entry lobby,
the three groups in each row are numbered {A, B, C} respectively. Starting from the side closest
to the Game Night area, the rows of groups are numbered 1, 2, 3, 4, 5, and W.

Row 1 includes the 100s (all odd numbers) and even 200s.

Row 2 includes the even 200s and odd 300s.

Row W is the series of booths up against the wall, treated as a single group and allocated a single switch.
(Last year we used two switches for this area). Booth numbers are in the 700 range and are all odd.

The reason for the dichotomy between group row numbers and booth numbers is because the booth
numbers are based on the aisle (which makes sense for humans traversing the aisle and results in
an opposite-side-of-the-street-odd-even mix similar to street addresses. However, for switch deployment,
we care about the way the booths are clustered and the aisles are irrelevant.

The VLAN numbers aren’t booth specific and because of the way we’re generating things currently,
they cannot have a rational relationship to booth numbers. We can look at improving that for next
year, but for this year at this stage of the process, running code trumps elegance.

Owen