socallinuxexpo/scale-network #222: AP image inconsistancy

davidelang, 2019-03-09:
a few APs ended up with disabled=1 in the radio0 and radio1 stanzas in /etc/config/wireless (the rest didn't have anything)

I have fixed the ones I found, which were:
104-ap3
103-ap2
101-ap5

this could be auto-generated, or it could be a flashing problem
@sarcasticadmin please take a look

sarcasticadmin, 2019-03-28:
To note `golden` files show that its set: https://github.com/socallinuxexpo/scale-network/blob/master/tests/unit/openwrt/golden/ar71xx/etc/config/wireless

davidelang, 2019-03-28:
it's possible that this was APs running last year's image (failed flash), or 
that the first startup of the wifi subsystem in openwrt modified these radio 
stanzas.

Since we were flashing the same image to all of them, it's not the case that the 
different images had different file contents in them

There were not a lot of these, but when I went through and did this to every AP, 
there were at least a dozen throughout the show that were like this.

David Lang

owendelong, 2019-09-06:
I'll open a separate issue to add better validation to the AP flashing process.

owendelong, 2019-09-06:
Believed to have been images not getting flashed.

closed 2019-09-06 by owendelong