socallinuxexpo/scale-network #239: setup a printer to be available via wifi

davidelang, 2019-03-10:
we need to be able to keep registration operating later than we need the registration stations (and the registration switch)

Owen's suggestion was to setup a PI to act as a bridge to wifi

David's suggestion is to setup an AP to act as a switch and connect a reg desk workstation, and a printer and a wall jack

owendelong, 2019-03-13:
My thinking is as follows:

1. We don't really need an additional AP at the reg desk.
2. If we're getting PI 3B+ for registration next year, then we have built in ethernet + wifi on the 3B+.
3. It's really easy to set up our standard image so that the PI bridges the Ethernet and WiFi ports.
4. We could make all of the Reg PIs work over WiFi by default and not have to cable them (if we chose).
5. We could plug the printers into PIs in all cases if we did this, which might obviate the need for the Registration switch.
6. It wouldn't create additional AP image dependencies and/or limitations.
7. There are more useful bridge configurations we might wish to consider (e.g. signs VLAN) for AP deployment. Using an AP for this would either require a custom AP configuration fo this one unit or creative port limitations on the APs.

davidelang, 2019-03-14:
I prefer for registration to be wired so we can better protect it.

this would be a custom build for either a Pi or an AP

David Lang

On Tue, 12 Mar 2019, owendelong wrote:

> My thinking is as follows:
>
> 1. We don't really need an additional AP at the reg desk.
> 2. If we're getting PI 3B+ for registration next year, then we have built in ethernet + wifi on the 3B+.
> 3. It's [...]

davidelang, 2019-03-14:
On Thu, 14 Mar 2019, owendelong wrote:

> Pretty minimal customization on the PI.

not much more on an AP :-)

> Not sure what you perceive as the additional protections via wired, can you elaborate?

Well, this is assuming that we implement network security next year, but there 
is a much smaller attack surface for wired than wireless (wireless can be 
attacked by anyone in the area, wired requires that someone plug something in on 
the right network before they can start attacking)

David Lang

owendelong, 2019-03-14:
> On Mar 14, 2019, at 15:19 , David Lang <notifications@github.com> wrote:
> 
> On Thu, 14 Mar 2019, owendelong wrote:
> 
> > Pretty minimal customization on the PI.
> 
> not much more on an AP :-)
> 
> > Not sure what you perceive as the additional protections via wired, can you elaborate?
> 
> Well, this is assuming that we implement network security next year, but there 
> is a much smaller attack surface for wired than wireless (wireless can be 
> attacked by anyone in the area, wired requires that someone plug something in on 
> the right network before they can start attacking)

While that’s true, there are enough wired ports available to plug into easy enough that I’m not sure I’d call it “much” smaller.

If we go wired, we retain the dependency on the registration switch, or, we have to cable that AP to the wall, printer, and PI.

OTOH, if we go wifi, then the PI can be cabled directly to the printer and we’re done. No risk of accidental unplugging from the switch, etc. Much less complicated cabling.

davidelang, 2019-03-28:
another options would be to have one printer plugged in to a wall jack

owendelong, 2019-09-06:
Let's discuss this in the meeting tonight and get a team consensus. The more I think about it, the more I'm willing to accept the wireless attack surface and I don't think it has to be a custom PI image. I think we can make WiFi/Wireless Bridged on the same IP network a standard configuration.

kylerisse, 2020-01-21:
Lets attempt to capture this requirement in #256