socallinuxexpo/scale-network #240: zabbix tweaks

davidelang, 2019-03-10:
things we needed to do for performance

increase postgesql max connections and shared buffers as per https://stackoverflow.com/questions/30778015/how-to-increase-the-max-connections-in-postgres

adjust poller numbers per https://blog.zabbix.com/monitoring-how-busy-zabbix-processes-are/457/

lots of template tweaks

set auto-registration actions

Linux (including OpenWRT) systems can auto-register themselves, routers and switches need to be manually added. the zabbix_agentd.conf should have a metadata tag that says what type of system it is (AP, server, sign, self-reg, etc

we installed the lldp module from 
https://share.zabbix.com/network_devices/l2-discovery-module-for-lldp there does not seem to be a practical difference between the two templates (we need to look at that more)

in 17x  we did not setup users or security.

davidelang, 2019-03-10:
https://blog.zabbix.com/scalable-zabbix-lessons-on-hitting-9400-nvps/2615/

sarcasticadmin, 2020-02-13:
`openwrt` agent config: https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/files/etc/zabbix_agentd.conf

sarcasticadmin, 2020-02-13:
`ansible` agent config mirrors the `openwrt` and correctly sets `HostMetadata`: https://github.com/socallinuxexpo/scale-network/blob/master/ansible/roles/zabbixagent/templates/zabbix_agentd.conf.j2

owendelong, 2024-01-28:
Closing this since we are moving away from Zabbix.

closed 2024-01-28 by owendelong