socallinuxexpo/scale-network #241: problems accessing expo srx from noc/zabbix

davidelang, 2019-03-10:
could ping via ipv4, but snmp did not work

owendelong, 2019-09-06:
I have some suspicions and I have the SRXs here... Need to set them up in a lab-environment and duplicate the issue, then test further.

owendelong, 2020-02-15:
Yeah, IPv4 SNMP isn't expected to work... IPv6 is configured...

CF-MDF: 
set snmp community <redacted> clients 2001:470:f325:103::/64
set snmp community <redacted> clients 2001:470:f325:503::/64

EX-MDF:
set snmp community <redacted> clients 2001:470:f325:103::/64
set snmp community <redacted> clients 2001:470:f325:503::/64

BR-MDF:
set snmp community <redacted> clients 2001:470:f325:103::/64
set snmp community <redacted> clients 2001:470:f325:503::/64

Is there some reason we need IPv4 SNMP?

sbibayoff, 2020-02-24:
Zabbix could/does snmp and monitoring through zabbix agents through IPv6. Will initially set it up to ONLY do IPv6 to see how that behaves.

I'm keeping this issue open till we verify.

owendelong, 2020-02-25:
Closing this one out as "works as intended".

closed 2020-02-25 by owendelong