socallinuxexpo/scale-network #248: Buy gaffer tape

owendelong, 2019-03-11:


owendelong, 2019-09-06:
https://smile.amazon.com/ProTapes-Pro-Premium-Gaffers-Adhesive/dp/B00DYB2TTC/ref=sr_1_2_sspa?keywords=gaffer+tape+box&qid=1567731889&s=gateway&sr=8-2-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExSlBFTEg0TE9STUEmZW5jcnlwdGVkSWQ9QTAzNjkwMDgyMDBLUURJVDE2RFJJJmVuY3J5cHRlZEFkSWQ9QTAwMDE2MzJPVTVZTDdVOVAxREQmd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

Will send to Ilan/Bala for purchase

hriday, 2019-09-06:
I thought we had a ton of gaffer tape?

owendelong, 2019-09-06:
Two years ago, we bought a ton of gaffer tape (a box, as described below).

One year ago, we had half a ton of gaffer tape left.

This year, it is mostly gone.  We use roughly half a box of gaffer tape per year. This has been communicated more than once.

It’s been two years since we bought gaffer tape, thus, it’s time to order again.

Owen

owendelong, 2020-02-11:
Steven, Please confirm this was received. If so, please close this issue.

kidicarus2, 2020-02-11:
Confirmed it is received,

unknown how to close issues.

thanks

To Your Success!
Steven

closed 2020-02-11 by irabinovitch