socallinuxexpo/scale-network #249: integration between google docs and ansible?

davidelang, 2019-03-11:
We have a number of source-of-truth docs that are current .tsv files

google docs has versioning, branching, and user permissions, and in addition will allow simultaneous edits (avoiding merging issues)

Can we find some way to pull the source-of-truth files from google docs (ideally spreadsheets) and use them locally? In git this may be a trigger that happens on a pull to get the docs, or we may want to pull them into git and version them daily (in case someone doesn't tag a version)

kylerisse, 2019-03-11:
related to #232

kylerisse, 2019-05-25:
Doing this as a 2 part process. The first portion is to have the Ansible inventory script consume a CSV  instead of a TSV which is included in PR #252 . The second portion will be a future effort to implement a new sync script from Google Sheets to populate the CSV that Ansible consumes.

owendelong, 2019-05-26:
I'm not sure I'm wild about converting everything to csv vs. tsv. Having said that, if we're really determined to move forward with it, then I think the Google sheet integration is a critical first step rather than the other way around.

Editing CSVs is a huge PITA and for the most part, I've been the one maintaining these files and using vi to do it.

If we're dead set on doing this, then there's a python script as part of this project (https://www.sparkfun.com/sparkx/blog/2591)  which extracts a CSV from a google sheet automatically that you might find useful.

kylerisse, 2019-05-26:
@owendelong this has nothing to do with the files you maintain and they have been omitted precisely because of this objection. We discussed this at SCaLE per @davidelang's request. It is captured in #232. This only applies to APs, Pis, servers, and routers which are all files you don't touch. Your switch configuration scheme remains unchanged and the reason I requested @davidelang and @sarcasticadmin review as they are the ones impacted.

kylerisse, 2020-01-21:
Punting this for now. Let's clarify what the solution looks like in practice.

sarcasticadmin, 2023-03-23:
Closing as we have continued to leverage aplist.csv and upcoming improvements via: https://github.com/socallinuxexpo/scale-network/pull/585

closed 2023-03-23 by sarcasticadmin