socallinuxexpo/scale-network #250: implement security

davidelang, 2019-03-28:
attendees should not have the ability to send packets to infrastructure devices.

owendelong, 2020-02-21:
Well... Some modifications to this policy...

Attendees need to be able to send DNS requests to the DNS servers.
Attendees need to be able to send DHCP requests to the DHCP servers.
Looks like we don't provide local time servers (Should we?).
Any other local services that need to be accessible?
Do we need to permit access to the AV LAN from the attendee WiFi networks?

owendelong, 2020-02-21:
First stab at enabling this:

owen@cf-mdf-01# show | compare 
[edit]
+  policy-options {
+      prefix-list protected-local-networks {
+          10.0.0.0/8;
+          172.16.0.0/12;
+          192.168.0.0/16;
+          2001:470:f325::/48;
+      }
+      prefix-list dns_servers {
+          10.0.3.5/32;
+          10.128.3.5/32;
+          2001:470:f325:103::5/128;
+          2001:470:f325:503::5/128;
+      }
+  }
[edit firewall family inet]
      filter ctf-from-internet { ... }
+     filter from-attendee {
+         term permit_dns {
+             from {
+                 destination-prefix-list {
+                     dns_servers;
+                 }
+                 destination-port domain;
+             }
+             then accept;
+         }
+         term permit_dhcp {
+             from {
+                 destination-port [ bootps dhcp ];
+             }
+             then accept;
+         }
+         term block_local {
+             from {
+                 destination-prefix-list {
+                     protected-local-networks;
+                 }
+             }
+             then {
+                 discard;
+             }
+         }
+         term allow_everything {
+             then accept;
+         }
+     }
[edit firewall family inet6]
      filter ctf-from-internet6 { ... }
+     filter from-attendee {
+         term permit_dns {
+             from {
+                 destination-prefix-list {
+                     dns_servers;
+                 }
+                 destination-port domain;
+             }
+             then accept;
+         }
+         term permit_dhcp {
+             from {
+                 destination-port [ bootps dhcp ];
+             }
+             then accept;
+         }
+         term block_local {
+             from {
+                 destination-prefix-list {
+                     protected-local-networks;
+                 }
+             }
+             then discard;
+         }
+         term allow_everything {
+             then accept;
+         }
+     }

owendelong, 2020-02-25:
Will look at implementing this in situ at SCaLE (hopefully Tuesday or Wednesday) with testing.

owendelong, 2023-09-13:
Given the deprecation of the "Staff" VLAN I think this has become moot.

closed 2023-09-13 by owendelong