socallinuxexpo/scale-network #256: Create build procedure for Pi 4B

kylerisse, 2019-09-11:
Create and document build procedure for Pi 4B images

sarcasticadmin, 2020-01-09:
Interesting project I came across the other day: https://buildroot.org/

I dont think we have time to address the build process with this particular approach but I think this might be something to consider for post-18x

owendelong, 2020-01-10:
For things like this, please create a new issue and tag the new issue with post-18 or something like that.

Owen

sarcasticadmin, 2020-01-13:
`buildroot` issue can be found here: https://github.com/socallinuxexpo/scale-network/issues/271

kylerisse, 2020-02-01:
The acceptance criteria for this task is to product 3 distinct image files for pis with use case profiles:
- Registration / Signs
- Flashing
- Helpdesk

kylerisse, 2020-02-01:
Further acceptance criteria includes:
- imaging PIs have no network or gui and aren't part of the further requirements in this list.
- all pis should have zabbix agent
- test 3 different monitor types (4k, 1080p, and [scale monitors])
- LLDP enabled, validated by juniper switch
- signs / registration using existing python browser setup (detect mouse)
- priority order is flashing station > everything else
- pis should get their names from dhcp

owendelong, 2020-02-11:
Registration/Sign PI image must meet the requirements in #239.

Ideally:

1. PI detects whether wired connection has a DHCP server.
- 1. If not, Connect wirelessly to Registration VLAN and bridge Wired port
- 2. If so, disable WiFi Interface and operate normally.
2. If PI is operating in bridged mode, WiFi is the upstream interface. Printer will be attached to wired interface.
- 1. PI should ascertain printer's IP address (snoop the bridged DHCP transaction between printer and DHCP server)
- 2. PI should send all badges to the detected attached printer.

The purpose of this is to have a self-contained registration station which does not depend on the registration switch or any of the wired infrastructure other than having a working AP within range. This will allow us do most of the registration tear-down earlier on Sunday.

davidelang, 2020-02-11:
I think it's simpler to just wire to the AP, it only adds one more wire

we already have the multi-network config support in the APs.

We have an AP in the area of the help desk, so we should be able to hook up one 
printer and pi to the AP rather than the reg switch.

David Lang

owendelong, 2020-02-13:
OK... Let's go with that for this year.

PR #290 (2020-02-25): [READY] add new signs/reg image instructions for pi4
## Description of PR
Fixes #256 

## Previous Behavior
Only had pi4 images for "imaging" and "general" pis, none for registration / signs.

## New Behavior
Have pi4 images for registration / signs. Instructions added.

## Tests
Building pis at work party. Having David and Peter try to do nefarious things. Creating compressed image.
closed 2020-02-25 by sarcasticadmin