socallinuxexpo/scale-network #268: Create ansible configuration for new server hardware

sarcasticadmin, 2020-01-09:
## Description

We have new hardware that was donated very generously by ixsystems.com . This will replace the majority of standalone desktop machines that were previous fullfilling roles as "servers" from previous conferences.

Were going to be running the following configuration on the hardware:

* Base OS: FreeBSD
* Hypervisor: Bhyve
* Individual VMs for each server role

## Acceptance Criteria
* Servers are up and running with consistent builds via ansible
* VMs are stable and provisioning has been tested for all existing roles
* Network interfaces and bridges are correctly working with both IPv4 and IPv6

kylerisse, 2020-02-01:
Additional acceptance criteria:
- all vms running

sarcasticadmin, 2020-03-03:
Completed as of #302

closed 2020-03-03 by sarcasticadmin