socallinuxexpo/scale-network #271: Investigate buildroot for pi images

sarcasticadmin, 2020-01-13:
## Description

Interesting project I came across the other day: https://buildroot.org/

We could do something very similar to what were doing in https://github.com/socallinuxexpo/scale-network/tree/master/openwrt but for the pis. This might be the way to go in the future so we can iterate on the image vs just letting it decay.

## Acceptance Critera
* See how feasible it is to build pi images with buildroot
* Spin out follow issue as needed

sarcasticadmin, 2022-11-22:
Build procedure for pi's will be going in a different direction via netboot, closing this for now

davidelang, 2022-11-22:
we still need to create the image that we netboot from, so I think we should 
keep this open.

David Lang

On Mon, 21 Nov 2022, Robert James Hernandez wrote:

> Build procedure for pi's will be going in a different direction via netboot, closing this for now
>
>

owendelong, 2022-11-28:
Yes, the original point of this was to investigate it for building the net-bootable images.

Owen

sarcasticadmin, 2025-02-23:
The Kiosk Pi image being built with nix in https://github.com/socallinuxexpo/scale-kiosk we no longer need to investigate buildroot for pis images. Any changes will be done as part of the nix build process

closed 2025-02-23 by sarcasticadmin