socallinuxexpo/scale-network #281: Investigate ath10k-ct driver solutions

sarcasticadmin, 2020-02-02:
## Description

Related to the finding during #264 and currently seeing issue for the TPLINK C2600

Dumping ground for notes around this issue:
* https://github.com/openwrt/openwrt/blob/db992e7b53f5a3b614c27f9f2d589128c570a821/package/kernel/ath10k-ct/Makefile 
* http://lists.infradead.org/pipermail/openwrt-devel/2020-January/021482.html
* https://github.com/openwrt/openwrt/pull/1077
* https://bugs.openwrt.org/index.php?do=details&task_id=2664

## Acceptance Criteria
* Resolve the issues with the ath10k-ct driver

owendelong, 2020-02-13:
We're going to skip this for SCaLE 18x and only support the Netgear Routers this SCaLE.

sarcasticadmin, 2020-04-09:
Progress on 10k, courtesy of @davidelang:

1. https://forum.openwrt.org/t/aql-and-the-ath10k-is-lovely/59002
2. http://lists.infradead.org/pipermail/openwrt-devel/2020-March/022412.html
3. http://lists.infradead.org/pipermail/openwrt-devel/2020-March/022411.html

sarcasticadmin, 2020-07-12:
The saga continues for these small devices. This could effect our our tplinks too:

- https://bugs.openwrt.org/index.php?do=details&task_id=2651#comment8515

sarcasticadmin, 2020-10-16:
most recently, looks like there some instability with 5.4 kernel on the  c2600. It seems like you need to disable cpufreq at build to keep it from crashing: https://bugs.openwrt.org/index.php?do=details&task_id=3099#comment8901

davidelang, 2020-10-16:
locking the devices to full speed rather than trying to save power doesn't seem 
like a big problem to me :-)

On Fri, 16 Oct 2020, Robert James Hernandez wrote:

> Date: Fri, 16 Oct 2020 08:18:51 -0700
> From: Robert James Hernandez <notifications@github.com>
> Reply-To: socallinuxexpo/scale-network
>     <reply@reply.github.com>
> To: socallinuxexpo/scale-network <scale-network@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Mention <mention@noreply.github.com>
> Subject: Re: [socallinuxexpo/scale-network] Investigate ath10k-ct driver
>     solutions (#281)
> 
> most recently, looks like there some instability with 5.4 kernel on the  c2600. It seems like you need to disable cpufreq at build to keep it from crashing: https://bugs.openwrt.org/index.php?do=details&task_id=3099#comment8901
>
>

sarcasticadmin, 2022-11-22:
Were most likely going to forego adopting the C2600 due to its track record of instability, going to close this since well be focusing on the newer hardware as part of #410

closed 2022-11-22 by sarcasticadmin