socallinuxexpo/scale-network #283: As much monitoring as possible should be done via IPv6

owendelong, 2020-02-15:
## Description
Reference issue #241 -- Works as expected
SNMP permits are only from the infrastructure IPv6 addresses in both buildings.
set snmp community <redacted> clients 2001:470:f325:103::/64
set snmp community <redacted> clients 2001:470:f325:503::/64

## Acceptance Criteria
Zabbix configuration should use IPv6 for all devices that have an IPv6 address (which should be (very nearly, if not) everything at this point).

sbibayoff, 2020-02-24:
Same comment as from issue #241 
Zabbix could/does snmp and monitoring through zabbix agents through IPv6. Will initially set it up to ONLY do IPv6 to see how that behaves.

I'm keeping this issue open till we verify.

owendelong, 2020-03-07:
This was completed.

closed 2020-03-07 by owendelong