socallinuxexpo/scale-network #295: Current Raspbian Ping doesn't report malformed Echo Replies

owendelong, 2020-02-24:
## Description
We don't see echo replies coming back even though they are passed from the kernel to the ping binary.

This is because the uboot firmware on the netgears is sending malformed echo replies.

## Acceptance Criteria
We need a version of ping that still reports these malformed echos.

owendelong, 2020-02-24:
Strace of current ping ignoring response packets (note the recvmsg system calls:

root@raspberrypi:/home/pi# ping 192.168.1.1 &
[1] 1118
root@raspberrypi:/home/pi# PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

root@raspberrypi:/home/pi# strace -fp 1118
strace: Process 1118 attached
gettimeofday({tv_sec=1580619490, tv_usec=379680}, NULL) = 0
poll([{fd=3, events=POLLIN}], 1, 235)   = 0 (Timeout)
gettimeofday({tv_sec=1580619490, tv_usec=615477}, NULL) = 0
gettimeofday({tv_sec=1580619490, tv_usec=615643}, NULL) = 0
sendto(3, "\10\0\vz\4^\0\v\342V6^\333d\t\0\10\t\n\v\f\r\16\17\20\21\22\23\24\25\26\27"..., 64, 0, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, 16) = 64
recvmsg(3, {msg_name={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, msg_namelen=128->16, msg_iov=[{iov_base="E\0\0<\0\31\0\0@\1\367R\300\250\1\1\300\250\1\4\0\0P\363\4^\0\vabcd"..., iov_len=192}], msg_iovlen=1, msg_control=[{cmsg_len=20, cmsg_level=SOL_SOCKET, cmsg_type=SCM_TIMESTAMP, cmsg_data={tv_sec=1580619490, tv_usec=615974}}], msg_controllen=20, msg_flags=0}, 0) = 60
recvmsg(3, {msg_namelen=128}, MSG_DONTWAIT) = -1 EAGAIN (Resource temporarily unavailable)
gettimeofday({tv_sec=1580619490, tv_usec=616574}, NULL) = 0
poll([{fd=3, events=POLLIN}], 1, 999)   = 0 (Timeout)
gettimeofday({tv_sec=1580619491, tv_usec=616942}, NULL) = 0
gettimeofday({tv_sec=1580619491, tv_usec=617099}, NULL) = 0
sendto(3, "\10\0Zs\4^\0\f\343V6^\213j\t\0\10\t\n\v\f\r\16\17\20\21\22\23\24\25\26\27"..., 64, 0, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, 16) = 64
recvmsg(3, {msg_name={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, msg_namelen=128->16, msg_iov=[{iov_base="E\0\0<\0\32\0\0@\1\367Q\300\250\1\1\300\250\1\4\0\0P\362\4^\0\fabcd"..., iov_len=192}], msg_iovlen=1, msg_control=[{cmsg_len=20, cmsg_level=SOL_SOCKET, cmsg_type=SCM_TIMESTAMP, cmsg_data={tv_sec=1580619491, tv_usec=617451}}], msg_controllen=20, msg_flags=0}, 0) = 60
recvmsg(3, {msg_namelen=128}, MSG_DONTWAIT) = -1 EAGAIN (Resource temporarily unavailable)
gettimeofday({tv_sec=1580619491, tv_usec=618021}, NULL) = 0
poll([{fd=3, events=POLLIN}], 1, 999)   = 0 (Timeout)
gettimeofday({tv_sec=1580619492, tv_usec=618384}, NULL) = 0
gettimeofday({tv_sec=1580619492, tv_usec=618539}, NULL) = 0
sendto(3, "\10\0\271l\4^\0\r\344V6^+p\t\0\10\t\n\v\f\r\16\17\20\21\22\23\24\25\26\27"..., 64, 0, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, 16) = 64
recvmsg(3, {msg_name={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, msg_namelen=128->16, msg_iov=[{iov_base="E\0\0<\0\33\0\0@\1\367P\300\250\1\1\300\250\1\4\0\0P\361\4^\0\rabcd"..., iov_len=192}], msg_iovlen=1, msg_control=[{cmsg_len=20, cmsg_level=SOL_SOCKET, cmsg_type=SCM_TIMESTAMP, cmsg_data={tv_sec=1580619492, tv_usec=618862}}], msg_controllen=20, msg_flags=0}, 0) = 60
recvmsg(3, {msg_namelen=128}, MSG_DONTWAIT) = -1 EAGAIN (Resource temporarily unavailable)
gettimeofday({tv_sec=1580619492, tv_usec=619427}, NULL) = 0
poll([{fd=3, events=POLLIN}], 1, 999)   = 0 (Timeout)
gettimeofday({tv_sec=1580619493, tv_usec=619783}, NULL) = 0
gettimeofday({tv_sec=1580619493, tv_usec=619935}, NULL) = 0
sendto(3, "\10\0Df\4^\0\16\345V6^\237u\t\0\10\t\n\v\f\r\16\17\20\21\22\23\24\25\26\27"..., 64, 0, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, 16) = 64
recvmsg(3, {msg_name={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, msg_namelen=128->16, msg_iov=[{iov_base="E\0\0<\0\34\0\0@\1\367O\300\250\1\1\300\250\1\4\0\0P\360\4^\0\16abcd"..., iov_len=192}], msg_iovlen=1, msg_control=[{cmsg_len=20, cmsg_level=SOL_SOCKET, cmsg_type=SCM_TIMESTAMP, cmsg_data={tv_sec=1580619493, tv_usec=620285}}], msg_controllen=20, msg_flags=0}, 0) = 60
recvmsg(3, {msg_namelen=128}, MSG_DONTWAIT) = -1 EAGAIN (Resource temporarily unavailable)
gettimeofday({tv_sec=1580619493, tv_usec=620814}, NULL) = 0
poll([{fd=3, events=POLLIN}], 1, 999)   = 0 (Timeout)
gettimeofday({tv_sec=1580619494, tv_usec=621197}, NULL) = 0
gettimeofday({tv_sec=1580619494, tv_usec=621346}, NULL) = 0
sendto(3, "\10\0\300_\4^\0\17\346V6^\"{\t\0\10\t\n\v\f\r\16\17\20\21\22\23\24\25\26\27"..., 64, 0, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, 16) = 64
recvmsg(3, {msg_name={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.1.1")}, msg_namelen=128->16, msg_iov=[{iov_base="E\0\0<\0\35\0\0@\1\367N\300\250\1\1\300\250\1\4\0\0P\357\4^\0\17abcd"..., iov_len=192}], msg_iovlen=1, msg_control=[{cmsg_len=20, cmsg_level=SOL_SOCKET, cmsg_type=SCM_TIMESTAMP, cmsg_data={tv_sec=1580619494, tv_usec=621658}}], msg_controllen=20, msg_flags=0}, 0) = 60
recvmsg(3, {msg_namelen=128}, MSG_DONTWAIT) = -1 EAGAIN (Resource temporarily unavailable)
gettimeofday({tv_sec=1580619494, tv_usec=622216}, NULL) = 0
poll([{fd=3, events=POLLIN}], 1, 999^Cstrace: Process 1118 detached
 <detached ...>

owendelong, 2020-02-24:
tcpdump of PING and response showing problem.
21:07:06.015805 IP 192.168.1.4 > 192.168.1.1: ICMP echo request, id 1118, seq 526, length 64
	0x0000:  4500 0054 f67d 4000 4001 c0d5 c0a8 0104
	0x0010:  c0a8 0101 0800 429c 045e 020e fa58 365e
	0x0020:  933d 0000 0809 0a0b 0c0d 0e0f 1011 1213
	0x0030:  1415 1617 1819 1a1b 1c1d 1e1f 2021 2223
	0x0040:  2425 2627 2829 2a2b 2c2d 2e2f 3031 3233
	0x0050:  3435 3637
21:07:06.015856 IP 192.168.1.1 > 192.168.1.4: ICMP echo reply, id 1118, seq 526, length 40
	0x0000:  4500 003c 0221 0000 4001 f54a c0a8 0101
	0x0010:  c0a8 0104 0000 4ef0 045e 020e 6162 6364
	0x0020:  6566 6768 696a 6b6c 6d6e 6f70 7172 7374
	0x0030:  7576 7761 6263 6465 6667 6869

sarcasticadmin, 2020-02-24:
@owendelong thanks, I think this give us enough info on this for the time being

tweak42, 2020-02-28:
As a work around for the the malformed ping, can you instead use the output of  "arp-scan -l" to verify that the router is at 192.168.1.1 to trigger the tftp push?  arp-scan is not installed by default in raspbian lite but can be installed with apt.

owendelong, 2020-02-28:
For that matter, following up a failed ping with an “arp -an” would do the trick.

Owen

sarcasticadmin, 2022-05-30:
Another workaround is the leverage our nix-shell unixtools.ping pkg added in: https://github.com/socallinuxexpo/scale-network/pull/443

sarcasticadmin, 2022-11-22:
This was still broken in 2022, going to close since we have a workaround.

closed 2022-11-22 by sarcasticadmin