socallinuxexpo/scale-network #309: new flashing script netgear wndr APs

davidelang, 2020-03-04:
old script (written in expect)

arp -d
ping until ap responds
ask for name of ap (we enter name, asset tag)
write name, mac to aplist csv file

new version should

arp -d
ping until ap responds
lookup mac in table of known aps
if known prompt with current name and asset name so if it has the same name, all that's needed is to hit enter
increment MAC address +1 (we flash on LAN interface, but use WAN interface, which is +1 to the MAC)
compare the new data with the saved data, check for dups and save

davidelang, 2020-03-04:
if the name hasn't changed, leave the other columns in place
if the name changes we should put in placeholder values for other columns
columns
IP address (next available IP address depending on building)
network config (default 0)
2.4GHz channel (default 1)
5GHz channel (default 36)
map (default 0)
x coord
y coord (position devices on a 100x100 grid combined with map to display on zabbix)

davidelang, 2020-03-04:
@kylerisse

kylerisse, 2020-03-04:
What do you think about this?

Keep the flashing script pretty simple. It doesn’t need to know about the cab schema. Have it output similar to what it does now with serial, so:
- name
- serial
- MAC address

Then create a separate merge script that pulls in the data, syncs on the serial, verifies the Mac (with 0x02 offset) and updates the existing aplist. This way aplist can be updated without having to touch the flash script and vie versa. It also reduces the complication of having to manage the IP space from the flashing process as well, allowing use of flash script in parallel and doing multiple merged into the master list.

davidelang, 2020-03-05:
in the final merge script we should create multiple dhcp entries

the mac address recorded during the flashing (LAN MAC)
the mac address +1 (the WAN MAC)
the lan mac modified for locally allocated address OR 02:00:00:00:00:00
the wan mac modified for locally allocated address +1 OR 02:00:00:00:00:00

that way any of the common mac addresses that we have seen from the system will map to the same IP address.

closed 2023-03-12 by davidelang