socallinuxexpo/scale-network #343: set pi hostname via lldp

davidelang, 2020-03-06:
## Description
Pis need to ahve a unique hostname that's useful

## Acceptance Criteria
be able to tell where a pi is from it's zabbix registration (i.e. hostname)

install lldp
packages needed: libmariadb3 libsnmp-base libsnmp30 lldpd mariadb-common mysql-common

set hostname via something like:
hostname `echo -n "pi-" ;lldpcli show neighbors |grep -e SysName -e PortID |sed -e s/"ifname.*\/"// -e s/".* "// |tr '\n' - |sed s/"-$"//`

davidelang, 2020-03-06:
this should be done when the network comes up, and if the hostname changes, save and reboot. If lldp returns nothing, do not change the hostname

lldp will remember the hold hostname unless you override it with a different lldp command or restart lldp

since other things also pick up the hostname when they start, it's best to reboot when the hostname changes rather than trying to track down all the services that need to restart

kylerisse, 2020-03-06:
done

closed 2020-03-06 by kylerisse