socallinuxexpo/scale-network #350: ctf-4 switch reports high temps

davidelang, 2020-03-07:
11:34:47 AM				CTF4.scale.lan	FPC: EX4200-48P, 48 POE @ 0/*/*: Temperature is above warning threshold: >50	25m 32s	No	
11:34:47 AM				CTF4.scale.lan	Routing Engine 0: Temperature is above warning threshold: >50

owendelong, 2020-03-07:
Let's plan on opening and vacuuming all of the Juniper switches at the first work party for next year.

owendelong, 2020-03-07:
http://monitoring1.scale.lan/zabbix/latest.php?groupids%5B%5D=16&application=Temperature&select=&filter_set=1

Indicates that the problem is affecting more than one switch.

sarcasticadmin, 2022-11-22:
@owendelong IIRC we checked the switches during last years work party and they were all clean.

Going to close but let me know if we need to reopen for the upcoming work party.

closed 2022-11-22 by sarcasticadmin