socallinuxexpo/scale-network #351: can we use lldp names on APs?

davidelang, 2020-03-07:
can we use LLDP based names for the APs?

pro, no need to label switches per location as the names will say what room switch they are connected to and what port they are plugged in to.

con, to find which AP in the room is which you would have to check which port it is plugged in to (which we can do with the wire labels)

can we make this easy enough to not need the blue tape labels and still know which one is where.

davidelang, 2020-03-08:
owen made the suggestion that we blink LEDs to indicate the port number

we need to test the USB led on all of them

we could blink the power led

we could eliminate the wifi traffic LEDs and have one of them just indicate wifi up (probably the bright blue onee) and the other could be used for other things (we still have combined traffic indication from the wan LED

we could use the LAN LEDs to show the port in binary.

we could make something that plugs into the USB port and gives us an indication.

owendelong, 2020-03-10:
I like the last suggestion…

Yocto sells something along this line, but it’s $43.05 (OUCH!).

OTOH, these look more interesting at $5@ in 100+ units…

There’s a GitHub library for driving them here: https://github.com/jnweiger/led-name-badge-ls32.

We can order 1-10 units for $6+shipping to test with at first. (Depending on the shipping cost, it might make more sense to buy 10 units @$60 to get free shipping than to pay (e.g. $40) to ship one unit at $6.

Owen

davidelang, 2020-03-11:
I'm missing what you are pointing at.

I was thinking something dirt cheap and trivial, something almost yubikey size 
that has a couple rows of LEDs on them that we can just set a pattern on.

David Lang

owendelong, 2020-03-11:
Originally, so was I.

These things I was pointing at are $6@ (cheaper in quantity), so depending on your definition of “dirt cheap”…

The cheapest USB<->GPIO thing I know of for driving a couple of rows of LEDs (I”d probably opt for at least a pair of 7-segment displays, actually) is an FT232 chip.

Those generally run about $3+, so by the time you added the board, the display, the USB connector, etc. you’d be north of $6@ in a heartbeat and we’d have to build them.

(Raw LEDs aren’t significantly cheaper than 7-segment displays these days).

The USB interface is the complicated/expensive part. If you’re building billions of them, it gets dirt cheap, but in our quantities, not so.

If you’ve got a BOM for something cheaper in mind, I’m happy to investigate.

Owen

davidelang, 2020-03-11:
the link you were pointing at didn't come through, can you try re posting it.

owendelong, 2020-03-11:
Somehow, I failed to paste the link in. :(

https://www.alibaba.com//product-detail/Promotional-programmable-led-light-up-flashing_60838008149.html

It’s a 44x11 LED matrix with USB programmability. It’s not ideal to our application, but would do what we need.

It’s designed to be a name tag for a waiter/driver/other service employee.

Has magnet and pin attachments on the back.

Owen

davidelang, 2020-03-11:
I was thinking something like the tomu with a few more LEDs on it. They list the 
tomu as <$10 in parts
https://tomu.im/tomu.html

David Lang

owendelong, 2020-03-11:
Uh, I don’t want to work on that, but i you feel like doing a _LOT_  of surface-mount hand-soldering, go for it.

Since the end result goes INSIDE the serial port rather than sticking out from it, more than a couple of colors is going to be hard to recognize (let alone fit).

Those SMT LEDs are about $1.50 each (or more, depending on color) in small quantities and they’re a bitch to hand-solder.

They don’t provide a link to the processor data sheet, so I can’t tell how many GPIOs are available. From the schematic, looks like 3, so you’d get a total of 5 LEDs.

owendelong, 2020-03-11:
Another option (more fiddling and additional cost in external components, but quite a bit more capability):
https://www.adafruit.com/product/2601?gclid=EAIaIQobChMIxLCb_8yR6AIVTL7ACh0Nng_yEAQYAyABEgIPFPD_BwE

It also occurs to me that we could use a USB->NANO->Display construct and simply program the NANO to take Serial data and feed it to the display.

Then we could just have the AP write to the serial port and the NANO would put whatever we sent on the display.

This could be a lot of fun, but it’s too expensive: https://www.adafruit.com/product/4243

This has a lot of potential, but still a bit on the pricey side: https://www.adafruit.com/product/3130 (similar to this: https://www.adafruit.com/product/1911 which might be better for us)

An alternative: https://www.adafruit.com/product/1907 would require us to drive the LEDs directly or add a controller chip.

Another alternative (non-alpha) would be these: https://www.adafruit.com/product/1001 @$4.95. They’re charlieplexed (sort of).

We could do this, but it’s more than the USB-based name badges just for the board fully of LEDs: https://www.adafruit.com/product/2973

Slightly cheaper in red: https://www.adafruit.com/product/2947

These could be fun: https://www.adafruit.com/product/1815
	Dirt cheap @$1.95/display and we’d still have to add controller/usb interface/etc. However, we could leave the middle 2 bars disconnected (or not) and display BCD for port numbers on the left 4 + right 4 easily enough.
	We could, of course, also get creative with displaying other stuff on the bar graph.

An improved version: https://www.adafruit.com/product/1719 would give us tri-color capabilities and 2 more barograph elements for the same price. These are charlieplexed, however (14 pins instead of 20).

These could be a LOT of fun, but way too expensive: https://www.adafruit.com/product/1399

We could use a pair of these on each unit: https://www.digikey.com/product-detail/en/lite-on-inc/LTP-587HR/160-1108-ND/153560
They’re $2.96@1, $2.623@10, $1.5232@100, $1.35392@500
Datasheet: http://optoelectronics.liteon.com/upload/download/DS-30-97-273/P587hr.pdf

Trick with these is that they take 17 pins per digit + common. We could charlieplex the digits (1 additional GPIO per digit), but we can’t charlieplex the pins within a digit because the anodes are common internally.

These 2-digit units might make more sense ($3.78@1, $3.154@10, $2.20760@100, $1.82918@500): https://www.digikey.com/product-detail/en/lite-on-inc/LTP-3786E/160-1011-ND/121576
Datasheet: http://optoelectronics.liteon.com/upload/download/DS-30-96-086/P3786E.pdf
Electrically, they’re identical to two of the previous unit charlieplexed by common anode (1 anode pin per character).

We’d either need a driver chip or a micro controller with at least 19 GPIOs available.

We might get slightly lower prices on bar graphs from DigiKey than from Adafruit.

Something like this might do the trick as well. (MOQ is 100 units, however, so $80 to get 100 of them + shipping): https://www.alibaba.com/product-detail/1-9mm-pitch-mini-size-8x8_60440485682.html?spm=a2700.details.deiletai6.1.22534b48TQAF2Q

Owen

davidelang, 2020-03-11:
I was thinking of basing it on this, extending the board out far enough to put 
more LEDs (or a 7 segment LED) on it.

it looks like there should be 13 available GPIOs (although it may be 8 due to 
multiple GPIOs being listed for a single pin???)

https://www.mouser.com/Search/Refine?Keyword=EFM32HG309
https://www.mouser.com/datasheet/2/368/efm32hg-datasheet-1500479.pdf

starting at ~1.50 each in Q 100

since we are talking about a run of 100+ of these, is this something that we can 
get made? or is 100 still too small an order?

David Lang

owendelong, 2020-03-11:
That’s 13 total on the chip… There are 10 in use by the existing hardware design.

PC14/PC15 are used by the USB port.

PC0/PC1/PE12/PE13 are used by the capacitors used for triggering DFU mode.

PB13/PB14 are used for Serial Interface (USART0)

PA0/PB7 are used for the existing 2 LEDs.

It looks like PF0/PF1/PF2 are also GPIOs, but they are used for the DEBUG port.

So… After accounting for what’s already in use, looks like you could add 3 LEDs to the existing 2 for a total of 5.

davidelang, 2020-03-11:
I would probably go with the single-color bar graph ($1.95 ea adafruit 8-10 
GPIOs required) with a dirt cheap processor or the 7 segment alphanumeric 
($3.95/pair adafruit 17 GPIOs required) with a few cents more per processor.

the USB ports are vertical on the back and 1" would fit (we could go higher, but 
not lower)

going with anything much bigger starts to require a bigger board

I'm thinking a t-shaped board that plugs into the USB and has the display on the 
back, ~1x1.5" to fit into the USB and hold either of these displays

David Lang

davidelang, 2020-03-11:
> That’s 13 total on the chip… There are 10 in use by the existing hardware design.
>
> PC14/PC15 are used by the USB port.
>
> PC0/PC1/PE12/PE13 are used by the capacitors used for triggering DFU mode.

are these the touch sensors?

> PB13/PB14 are used for Serial Interface (USART0)

would we need this as opposed to just using the USB?

> PA0/PB7 are used for the existing 2 LEDs.

> It looks like PF0/PF1/PF2 are also GPIOs, but they are used for the DEBUG port.

would we need this (after debugging)?

> So… After accounting for what’s already in use, looks like you could add 3 LEDs to the existing 2 for a total of 5.

why can't we re-purpose the others? (and there are 15 GPIOs, I had subtracted 
the USB pins from my count)

if we step up from the next larger processor we go from:
QFN32 to the QFN24 package
~$2 efm32hg310 instead of ~$1.50 efm32hg309
22 GPIOs instead of 15
6mm square instead of 5mm

David Lang

owendelong, 2020-03-11:
T-shaped doesn’t buy us anything. The charge for the board is based on the smallest rectangle the board fits inside of, not the actual area of the board.

I think that a NANO works either way, so $2 for the processor with the USB built in.

I was thinking in terms of something that plugged in and stuck up to be legible from the front.
	Pro: Easy to read.
	Con: We’d have to install/remove for each show, as packing with these attached would likely break things.

OTOH, I’m not convinced that anything we left plugged into the back would have a high survival rate, so I’m not sure that’s really a con.

I think I found a better display alternative:
https://www.amazon.com/Display-Module-SSD1306-3-3V-5V-Arduino/dp/B082S3XQB1/ref=sr_1_17_sspa?keywords=14+segment+lcd+display&qid=1583911161&sr=8-17-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFHU1hESlBPSklKRVYmZW5jcnlwdGVkSWQ9QTA1OTE0NzEyQTdHWDA1NUo1SDdZJmVuY3J5cHRlZEFkSWQ9QTA0NDMzNjAxUEsxRFpCRFE3SlpYJndpZGdldE5hbWU9c3BfYXRmX25leHQmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

This is SPI driven and is a full 128x32 White OLED display for $5/unit (in 2-packs).

I went ahead and did a risk buy of Nano boards (10@$3.34) and a pair of those displays. Both are supposed to arrive Thursday according to Amazon.

(Grand total around $50)

I’ll bread board them together and get software running that allows us to feed serial data over the USB connection to control the display.

Should be pretty straight forward and can be powered from the AP. Designing a case and adding the few components necessary to make it plug into the USB port will take a little more mechanical engineering.

Owen

owendelong, 2020-03-11:
> 
> 
> > That’s 13 total on the chip… There are 10 in use by the existing hardware design.
> >
> > PC14/PC15 are used by the USB port.
> >
> > PC0/PC1/PE12/PE13 are used by the capacitors used for triggering DFU mode.
> 
> are these the touch sensors?

Maybe… Implication in the documentation was that you stick a pair of tweezers in to your USB port to short them out in order to
get back to DFU mode to flash a new program. (SCARY!)

> > PB13/PB14 are used for Serial Interface (USART0)
> 
> would we need this as opposed to just using the USB?

Not aure what’s involved in repurposing it, but that might get you to 7 LEDs if we can.

> > PA0/PB7 are used for the existing 2 LEDs.
> 
> > It looks like PF0/PF1/PF2 are also GPIOs, but they are used for the DEBUG port.
> 
> would we need this (after debugging)?

Again, not clear whether or not re-use otherwise is an option. Also, not sure how you debug those outputs if you can’t
use those outputs while you’re debugging.

> > So… After accounting for what’s already in use, looks like you could add 3 LEDs to the existing 2 for a total of 5.
> 
> why can't we re-purpose the others? (and there are 15 GPIOs, I had subtracted 
> the USB pins from my count)

I suppose we can. I was basing it on the schematic provided in the reference you cited.

Personally, I’m not a fan of the design and I _REALLY hate working with QFNs.

> if we step up from the next larger processor we go from:
> QFN32 to the QFN24 package

That has to be backwards… I can’t imagine that the larger processor has fewer pins.

> ~$2 efm32hg310 instead of ~$1.50 efm32hg309
> 22 GPIOs instead of 15
> 6mm square instead of 5mm

That pretty well confirms that the QFN32 is the efm32HG310 and QFN24 is the efm32hg309.

Unless someone wants to buy me a small P&P machine, I have no interest in putting those together.

Owen

sarcasticadmin, 2021-10-03:
Being worked on as part as: https://github.com/socallinuxexpo/scale-network/tree/od.19.00

owendelong, 2025-02-02:
I think this has languished to the point we should probably close it as "won't fix".

closed 2025-02-04 by sarcasticadmin