socallinuxexpo/scale-network #357: Mandatory Cart handling/safety training for team members

owendelong, 2020-03-10:
## Description
It was clear from the process of loading the truck at the end of SCaLE 18x that there is a need to train team members that will be present during loading/unloading on safe movement and most importantly safe procedures for movement in concert with lift gates.

## Acceptance Criteria
Production of training documents and procedures for training.
Limit participation in loading/unloading to members that have demonstrated an adequate understanding of the safety concerns involved.

tweak42, 2020-03-13:
For visual learners, I highly suggest video or at the minimum pictures / diagrams on procedures to correctly handle the carts.

sbibayoff, 2020-03-16:
I know the "leadership" is not going top like this.

Reduce the heights of the cart. Cut off the top shelf. Reduces the weight and lowers the center of gravity.

Or have the "leadership" demonstrate the proper "procedures" and me or someone else could record a video of it then they maybe could see that there really is no "safe procedures"

davidelang, 2020-03-16:
for what it's worth, I would have been happy to have more, shorter carts, but 
storage would not fit them.

With the exception of the monitor cart, I do believe that there are safe ways to 
handle the carts.

I will be happy to demonstrate the next time we have the carts out.

I have unloaded all the carts (except the monitor cart) by myself without 
putting myself at risk. I have not tried to load them myself, and may not be 
able to due to having to fight gravity rather than taking advantage of it, but 
know what I would need to do to be safe in the process.

The key is to move the carts end-on wherever possible, and when not possible, 
move one end of the cart at a time. moving them directly backwards can only be 
done on completely level serfaces. This is especially critical when the wheels 
may hit an obstruction, such as a doorway transition, or the bottom of a ramp.

Right now, the monitor cart is dangerous, and even if the top half was cut off, 
it would continue to be dangerous becasue the problem is front-to-back weight 
distribution rather than top-to-bottom weight distribution. It's dangerous even 
when completely empty. the problem is that the new monitor cart moved the 
monitors ~4" towards the rear of the cart compared to the old one, and the 
wheels (while a huge benefit in many ways) have their support closer to the 
center of the cart when the cart is moving backwards. The CG is still over the 
area supported by the wheels, even with the lift gate tilted, but just barely, 
and pulling hard or hitting any obstruction is enough to move the CG outside 
the wheels.

All of the other carts have a much better front-to-back weight distribtion, and 
are not the same danger. But they do need to be handled properly. When I gave 
instruction at the work party and we loaded, I thought things went well. But at 
the end of the show, it didn't go nearly as well.

It may be that the right thing is going to be to have the back uphill. I haven't 
tried that.

I have scales that go to 440 pounds, it would be great if someone had some 
vehicle scales that went higher (and were not made of glass like the scales I 
have)

David Lang

On Mon, 16 Mar 2020, sbibayoff wrote:

> I know the "leadership" is not going top like this.
>
> Reduce the heights of the cart. Cut off the top shelf. Reduces the weight and lowers the center of gravity.
>
> Or have the "leadership" demonstrate [...]

sbibayoff, 2020-03-17:
Just from my observation of loading up the carts into the truck on Sun evening. 

When one side was up on the ramp and the other side of the cart needed to be swung around up onto the ramp, if you where pushing to high up on the cart, the cart tended to tip forward (toward the truck) when the wheel hit the ramp. And if where pushing too low, the cart tended tip backwards. It happened on every cart I helped with (4? 5?). It did seem not as bad if instead of stopping when one end got up on the ramp to start swinging the other end around and use the momentum to get the other end onto the ramp. But we where instructed to not do that. I believe trying to use the momentum way was not safe, but marginally safer than coming to a complete stop, then trying to swing around. That way you had to apply more force which would cause more tipping.

As for cutting all of the carts down. If we cut the top part off, and remove the top shelf, that gives us 3 shelves per cart. 5 carts x 3 shelves x 4 totes/shelf - (2 tote area for switches x 5 carts) = 50 toes spaces. We need just under 30 totes (28?) totes to cover all of the rooms/ballrooms/expo/etc . This leaves 20 tote spots for any extra (printer, extra cables, etc) . And we could make a "holder/space" on top of one of the carts to hold the plastic carts that we have. This will lower the cg significantly, at least 8", maybe 12".

I know this wastes the labor and material that went into the carts. And we won't have the capacity to expand greatly (reduced ~20%), but this should make the carts much safer. And I seriously doubt we would ever run into capacity issues.

This problem showed itself when the ramp didn't lay flat on the ground (one side was up almost 3/4"). I don't believe this is something we could control, so we should plan for it. Having the back on the uphill side would help, but it runs the risk of us getting used to it then forgetting about the dangers and precautions when we don't have a slight hill.

For weighing the carts, It looks like house hold scales (metal) w/ 4-500 capacities are around $20. I should be able to round up 4 of them.

davidelang, 2020-03-17:
On Tue, 17 Mar 2020, sbibayoff wrote:

> For weighing the carts, It looks like house hold scales (metal) w/ 4-500 
> capacities are around $20. I should be able to round up 4 of them.

most top out at 330 pounds, there are some that go to 440 (I have a couple), we 
can get by with 1, weighing one wheel at a time (with the adjacent wheel sitting 
on a block the same height as the scale)

but getting a matched pair would be nice.

David Lang

davidelang, 2020-03-17:
On Tue, 17 Mar 2020, sbibayoff wrote:

> When one side was up on the ramp and the other side of the cart needed to be
> swung around up onto the ramp, if you where pushing to high up on the cart,
> the cart tended to tip forward (toward the [...]

if you are running the cart end-on onto the ramp, there is no sideways momentum 
to help you.

the reason I was adament about stopping and then swinging is that people were 
either not stopping the momentum towards the truck (dangerous for the person 
between the cart and the truck) or they were swinging early (causing the cart to 
hit the ramp at an angle)

I didn't realize the ramp was not on the ground at the point we were rotating 
to, if I had, I would have had us rotate the other way.

> As for cutting all of the carts down. If we cut the top part off, and remove
> the top shelf, that gives us 3 shelves per cart. 5 carts x 3 shelves x 4
> totes/shelf - (2 tote area for switches x 5 [...]

Cutting off the top shelf would cost os ~20 bin spaces (4 on each of the 5 
carts), currently we have 2-3 empty bins (and a few that are overstuffed or 
otherwise should be split, but are not using the bottom of the monitor cart).

We need 4 bin spaces to hold the flat carts, ~3-4 to hold the printer and boxes 
of uncut cable. There are 4-5 bins of extension cords/power strips/power cords. 
we have ~14 bins of 'extra' cable (pretty much an entire cart dedicated to this)

There really isn't a lot of extra space on these carts.

At the next work party we will be going through all the bins to inventory them 
and test the cables, that will be a good time to examine the contents of the 
bins and figure out if there is any different packing that we should do.

lowering the height by 15" is not going to lower the CG by much, especially as 
we try to put lighter stuff on the top shelves.

we can test at the next work party by leaving the top shelf empty and see what 
difference that makes (not a perfect test as there is still the plyood there, 
but it should be close). hat's a good way to find the CG along the vertical? (so 
that the test can be an objective measurement)

Im also concerned that if we cut the carts down, we will end up with people 
putting bins on the top of the carts. This was happening with the old carts even 
without a top on them, the top shelf was getting two-high stacks of bins on 
them.

As I say, I think the problem is the front-back CG more than the up/down CG.

davidelang, 2020-03-17:
On Tue, 17 Mar 2020, David Lang wrote:

> On Tue, 17 Mar 2020, sbibayoff wrote:
>
>
> most top out at 330 pounds, there are some that go to 440 (I have a couple),
> we can get by with 1, weighing one wheel at a time (with the adjacent wheel [...]

when we test, we need to test with the casters in three positions, end on, 
moving to the front, and moving to the back. This is so we can figure out where 
the CG is in each wheel position.

would it make sense to measure the weight on each wheel with the cart tilted 
(say a 2x4 under the front or rear??) to see how the CG moves as it's tilted 
(giving us an idea where the CG is vertically)?

David Lang

owendelong, 2020-03-17:
This is perhaps where my piloting skills can come into play.

As you may or may not know, the relationship between the center of gravity of an aircraft and the center of lift is a rather critical one.

So critical in fact, that a pilot is required to compute the center of gravity for each particular loading of an aircraft for each flight.

This includes accounting for pilot, passengers, fuel, cargo, and baggage.

The pilot must also compute the CG for each of the following phases of a flight:

RAMP weight and balance (before starting the aircraft)
Takeoff weight and balance (At the start of the takeoff roll)
Mid-Cruise (50% of planned fuel consumed) weight and balance
Landing weight and balance (The weight and balance after all planned fuel has been consumed)

In an aircraft, this is a little more complicated than with our carts, because as fuel is consumed, the center of gravity moves.

For our carts, absent a change to the loading of the cart, the center of gravity will not change.

The center of gravity will also remain constant regardless of the angle of the cart. In fact, in a weightless environment,
(Zero G, floating, free fall) the easiest way to identify a CG very quickly is to spin the object in question. The rotation
will always be centered on the center of gravity unless an outside force is acting to perturb the rotation (e.g. one end
is resting against an immovable object).

For the cart to remain upright, the CG must remain within the lateral limits of a box defined by the contact patch of the
four casters and extended vertically to the point where a horizontal plane defined by the box would intersect the CG.

The CG does not move when the cart tips other than to the extent that the cart itself moves. That is, if you were to
anchor a marker representing the CG to the cart rigidly, the marker would remain in the correct position throughout
the tipping process.

However, as the cart tilts, the CG does shift relative to the wheels in the direction that the top of the cart is moving.
So long as the CG remains within the box defined above, if everyone lets go and there is no outside force acting
on the cart (e.g. momentum against the lip of the truck ramp), the cart will “fall” back onto its wheels and stop
tipping. (Momentum may cause this to bounce between wheels a few times, but energy lost in each bounce will
eventually result in return to a stable upright position on the wheels.

On a slope, the cart is tilted/tipped by the slope such that the vertical box is no longer parallel to the sides of the
cart. If the slope is sufficient that the vertical box no longer encompasses the CG, the cart will exhibit a tendency
to tip towards the CG. The further outside of the box, the stronger the tendency to tip.

The problem we currently have with the carts and the loading process is that the carts have a relatively high
CG (compared to the wheel location) and the truck ramp is not horizontal. As such, we are applying momentum
(of the cart) against a stopping force (against the bottom of the wheel on one side) which creates a tipping
moment (CG wishes to continue towards the truck, wheel wants to stop, cart tries to rotate vertically about
the pivot point defined by the axel of the now stopped wheel).

Lowering the CG does not cure this problem. It merely reduces the ARM component of the MOMENT
which is attempting to tip the cart. (MOMENT = MASS * ARM). Obviously, by reducing the ARM,
we can significantly reduce the MOMENT (1 unit of ARM reduces MOMENT by 1 MASS), so it does
reduce the problem. Reduce the problem enough and the tipping MOMENT becomes a manageable
or even negligible force.

Therefore, despite conventional wisdom, the solution isn’t using momentum to overcome the tipping
moment during the rotation phase. The solution is to do the following:

1. Bear in mind that the carts are extremely stable end-to-end (the CG is so far inside the box that an attempt
	to tip the cart endwise requires an extreme angle before the CG departs the box).

	The carts are not stable front to back (the carts are relatively narrow and because of the
	high CG, a relatively small angle front to back will take the CG outside of the box).

2. Given 1, to the greatest extent possible, all cart motion should be preserved along the long
	axis of the cart. Front to back movement should be avoided as much as possible. Rotational
	moves should be kept to a minimum.

3. Given (1) above, using momentum to get one end of the cart over the highest lip on the lift gate
	makes sense and is relatively safe. Important not to use too much momentum so that the cart
	can be reliably stopped before crushing someone between the cart and the back of the truck.

4. Once the first end of the cart is stable and stopped on the lift gate, slowly rotate the opposite
	end of the cart towards the lift gate. Stop when the wheel makes contact with the lift gate.

5. Using the (as yet uncut) handholds on the front of the carts, apply both upward and forwards
	pressure to the front of that end of the cart to get the front wheel onto the lift gate.

6. Once the front wheel is on the lift gate, continue rotating the cart until all 4 wheels are stable
	and in position on the lift gate. While stabilizing the cart in this position (two people,
	one on each end of cart), lift gate operator rotate the lift gate to level.

7. When everything is stable with the lift gate level, lift gate operator raises lift gate to truck.

8. When lift is fully raised and lift gate operator calls all clear, desired end of cart is rotated
	into truck first, and then cart is rolled along the long axis into position in the truck.

In no case should anyone be standing in the “danger zones” in front of or behind the carts.

The green semi-circles are the safe handling areas where people should be positioned during cart movement.
The red cross-hatched areas are the “danger zones”.

davidelang, 2020-03-17:
>     training for team members (#357)

it does not look like the diagrams you mention at the end of your post came 
through.

finding the location of the CG reltive to the contact patch when vertical is 
relativly easy (scales and simple math), how can we easily find the vertical 
component of the CG?

I do want to measure where the CG is to find out how close it is to the edge of 
the contact patch. (the monitor cart has it really close to the edge)

David Lang

owendelong, 2020-03-18:
Determining the vertical CG without tipping the cart is difficult (likely impossible).

The best method I know for determining the vertical (Z) component of the CG would be to lay the loaded cart on its back and
use the same procedure as for the X/Y components.

However, that extreme measure may not be all that necessary. We can assume that the carts have a roughly even distribution
of weight from the bottom of the base to the top of the side panels.

I believe we have a known weight for the wheels and if we remove one wheel and determine its vertical CG by weighing (we’d
need to put the wheels on one scale and the top plate on a second scale such that the non-wheel portion of the caster is not
supported on the first scale except by the axel). The weight on the first scale would be attributed to the center of the axel and
the weight on the second scale would be attributed to a point the appropriate distance from the axel.

Weight and balance calculations center around a “datum”. For fixed-wing aircraft, the datum is almost always a point against
the leading edge of the spinner (the nose cap in front of the propeller).

For our purposes, I think if we draw a Z axis line vertically down the middle of a cart placed on level ground, the point where
that axis intersects a plane defined by the contact point of the four wheels should be considered our “datum”.

Each point above that would be “inches above datum”.

For the weight from the first scale, we’d measure the radius of the wheels and consider that to be the distance from datum.
(I’ll give an example of 4 assuming that the wheels are roughly 8” in diameter).

For the weight from the second scale, we’d attribute that to the bottom of the base of the cart. I’ll call this 9” for illustrative
since I remember the caster base being roughly 1” beyond the wheel perimeter.

Further for illustration, I’ll assume 10 pounds per caster distributed with 8 on the first scale and 2 on the second.

The vertical CG of the caster would then be:
Item
Location (Inches above datum)
Quantity
Weight @
Total Weight
Moment (inch pounds)
Caster Axel
4
4
8
32
128
Caster Plate
9
4
2
8
72
CG and Totals
5

40
200

Extending this to encompass the entire cart (with assumptions about weight and approximations of locations):

Item
Location (Inches above datum)
Quantity
Weight @
Total Weight
Moment (inch pounds)
Caster Axel
4
4
8
32
128
Caster Plate
9
4
2
8
72
Cart
45
1
60
60
2700
Bins on Bottom Shelf
10.5
4
20
80
840
Bins on second shelf
24.5
2
20
40
980
Switches
24.5
10
18.2
182
4459
Bins on third shelf
45.5
4
20
80
3640
Bins on top shelf
60
4
20
80
4800
CG and Totals
52.75

160
8440

I’m betting if we measure, the above data won’t be too far off.

I’ll also bet that in terms of X Y CG, we’ll come pretty close to the geometric center of the cart in most cases.

If we remove the top area of the cart as Steve B suggests, the numbers change as follows:

(I”m assuming this takes off 15” of the top of the cart and about 10 pounds of weight, lowering the CG by
roughly half of that 15 inches).

Item
Location (Inches above datum)
Quantity
Weight @
Total Weight
Moment (inch pounds)
Caster Axel
4
4
8
32
128
Caster Plate
9
4
2
8
72
Cart
37.5
1
50
50
1875
Bins on Bottom Shelf
10.5
4
20
80
840
Bins on second shelf
24.5
2
20
40
980
Switches
24.5
10
18.2
182
4459
Bins on third shelf
45.5
4
20
80
3640
Bins on top shelf
0
0
20
0
0
CG and Totals
45.5

80
3640

I don’t think that the 7.25 inch lowering of the CG will make the carts that much less tippy.

My observation during loading and unloading was that other than the monitor cart, the other carts were not all
that tippy. Yes, if you ran them into the lift gate, when the wheel stopped abruptly, other wheels would leave the
ground, but the CG did not generally progress to a point where the cart wanted to continue tipping. (The CG
didn’t seem to get close to or outside of the box).

The monitor cart is different. Virtually all of the weight in the monitor cart is at the back of the cart and the CG is
likely very close to the edge of the box, especially when the casters are swiveled towards the front (as in when
the cart is being moved towards the back or swiveled to bring the back of one side onto the lift gate.

There’s also an issue from this where the monitors want to slide out of position during transport because they
have too much room to move towards the front of the cart.

We’ll need to work on a redesign of the monitor cart to try and find the most efficient way to address this.

Owen

owendelong, 2020-03-18:
Hmmm…

Let’s try it this way:

Image available at http://www.delong.com/Scale_Cart_danger.png

Owen

davidelang, 2020-03-18:
some technical details, 
https://www.northerntool.com/shop/tools/product_200660391_200660391

the plates are inset ~1/4" from the outside edges of the bottoms and are ~6 
7/8". the bottom is 21.75" (plus 1/4" that is inset into the back)

this puts the center of the wheels just under 15" apart.

the holes are 5.5" from each other on the same center, and from the side view of 
the wheel, it looks like the axles are about even (or just a little inside) the 
holes, so the swing of the axle from one extreme to the other is ~5.25-5.5 
inches

this puts the center of support from a wheel up to:

.75" (back) + just shy of 3.5" (edge of place to center pivot) + 2.5" (swing of 
heel)
= 6.75" (probably over by less than 1/4") from the back edge

the shelves on the monitor cart are 13", with 1/4" embedded into the back. That 
would put the CG of the shelves at <7" from the back face. (not accounting 
for their angle, the cuts on the front of the shelves, or the back all of which 
make it worse) No wonder it's so tippy.

The CG of the monitors is a little further from the back (but only by about the 
weight of the stand, so not much)

for the other carts, the shelves are 21", so their CG is 11" from the back of 
the cart.

rather than trying to establish datums for each item, I was thinking in terms of 
weighing at each wheel (with the other wheels on blocks to be at a similar 
height). I think we need to weigh with each wheel in tree positions, back, end, 
front (to account for uneven distribution of mass), although I could be argued 
out of this.

Then I was thinking that we could put a few thicknesses of plywood under the 
heavy side and re-measure the weight on each wheel. Since we know the angle and 
the resulting weight shift, that would let us calculate the height above the 
floor for the CG (and since we are tilting away from the heavy side, this should 
be perfectly safe to do.

since we maxed out a 1800lb rated lift gate (with 4-5 people on it as well), 
there is a lot of weight here. I am thinking of getting a couple of 660lb scales 
(similar to 
ebay.com/itm/660lb-x-0-1LB-Digital-Floor-Platform-Postal-Scale-Shipping-Weight-300kg-10582oz/273789325295 
@~$34 each) to be able to weigh both wheels at one end at the same time.

thoughts?

David Lang

sbibayoff, 2020-03-19:
responding to @davidelang 

I like the idea of having a scale under each wheel then also doing it as a slight angle, it would give a reasonable idea where the vertical cg is. Shit, and all if the cheap scales seem to have glass tops:
https://www.homedepot.com/p/Ozeri-Precision-Bath-Scale-440-lbs-200-kg-with-50-g-Sensor-0-1-lbs-0-05-kg-and-Infant-Pet-and-Luggage-Tare-ZB18-W2/303997824

davidelang, 2020-03-19:
@$15 each, it may be worth trying a couple of those (or a set of 4) rather than 
ordering the 660 pound ones that are about twice that price.

My only concern is that we may max these out, but at this price it's worth the 
risk (and we can probably return them if they do)

David Lang

On Wed, 18 Mar 2020, sbibayoff wrote:

> Date: Wed, 18 Mar 2020 18:15:07 -0700
> From: sbibayoff <notifications@github.com>
> Reply-To: socallinuxexpo/scale-network
>     <reply@reply.github.com>
> To: socallinuxexpo/scale-network <scale-network@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Mention <mention@noreply.github.com>
> Subject: Re: [socallinuxexpo/scale-network] Mandatory Cart handling/safety
>     training for team members (#357)
> 
> responding to @davidelang
>
> I like the idea of having a scale under each wheel then also doing it as a slight angle, it would give a reasonable idea where the vertical cg is. Shit, and all if the cheap scales seem to have glass tops:
> https://www.homedepot.com/p/Ozeri-Precision-Bath-Scale-440-lbs-200-kg-with-50-g-Sensor-0-1-lbs-0-05-kg-and-Infant-Pet-and-Luggage-Tare-ZB18-W2/303997824
>
>

owendelong, 2020-03-19:
> some technical details,
> https://www.northerntool.com/shop/tools/product_200660391_200660391
>
> the plates are inset ~1/4" from the outside edges of the bottoms and are ~6
> 7/8". the bottom is 21.75" [...]

Establishing a datum is pretty easy. You simply pick a point in space and then measure
consistently from it.

In terms of weighing, ideally, we have a scale for each wheel set (or at the very least for each end of the cart)
and we weigh the entire cart at once measuring the weight on each scale.

For aircraft weighing, traditionally, a scale is placed under each of the three wheels.

> Then I was thinking that we could put a few thicknesses of plywood under the
> heavy side and re-measure the weight on each wheel. Since we know the angle and
> the resulting weight shift, that would [...]

I think this is a lot of unnecessary maneuvering to avoid a relatively simple process of measurement, most
of which are already known.

> since we maxed out a 1800lb rated lift gate (with 4-5 people on it as well),
> there is a lot of weight here. I am thinking of getting a couple of 660lb scales
> (similar to [...]

Let’s discuss tonight.

Owen

davidelang, 2020-03-19:
I figured out how to get the weight/balance in onshape and set everything so it 
actually had mass.

I set the material for the wheels (steel frame, all plastic wheel, which isn't 
correct, as they aren't solid, but are partially metal, so close), and 
everything else to birch (it's probably a bit heavier due to the glue.

origin is back right corner (within a tenth of an inch or so)

onshape's estimate for the main carts is:

Mass:
358.345 lb
Volume:
13402.188 in3
Surface area:
38738.104 in2
Center of mass:
X: -36.470 in
Y: -9.011 in
Z: 35.698 in
Moments of inertia: lb in2
Lxx: 2.321e+5	Lxy: -1.707e-1	Lxz: -2.432e-1
Lyx: -1.707e-1	Lyy: 4.373e+5	Lyz: -7.690e+3
Lzx: -2.432e-1	Lzy: -7.690e+3	Lzz: 2.445e+5

This has the CG a fair bit closer to the back than I expected.

monitor cart is about 100 lb heavier:

Mass:
458.895 lb
Volume:
17378.216 in3
Surface area:
50828.702 in2
Center of mass:
X: -36.531 in
Y: -7.659 in
Z: 37.649 in
Moments of inertia: lb in2
Lxx: 2.554e+5	Lxy: 1.129e+0	Lxz: 1.730e+2
Lyx: 1.129e+0	Lyy: 5.084e+5	Lyz: -1.587e+4
Lzx: 1.730e+2	Lzy: -1.587e+4	Lzz: 2.871e+5

my back-of-the-envelope calculations didn't include the fact that the sides 
stick out 1/2" from the back, so that would put the axles about 7-7.25" from the 
origin. Not including the castors in the weight calculation puts the CG at 
7.211". It's not much more than the weight of the wheels on the front that's 
keeping it upright.

we'll have to see what the loaded balance is, but this may mean that we need to 
load them facing the other way and think about doing something to keep the bins 
on the shelves when facing downhill.

David Lang

davidelang, 2020-03-19:
I ordered 4 of these, they will arrive in about a week.

I think it's a reasonable guess that the loaded carts are between 800-1000lb, so 
one of these under each corner should work.

David Lang

On Wed, 18 Mar 2020, sbibayoff wrote:

> https://www.homedepot.com/p/Ozeri-Precision-Bath-Scale-440-lbs-200-kg-with-50-g-Sensor-0-1-lbs-0-05-kg-and-Infant-Pet-and-Luggage-Tare-ZB18-W2/303997824

owendelong, 2020-03-19:
How can the volume be so radically different between the two carts?

Is onshape calculating the volume of the solids and not counting free space?

> On Mar 18, 2020, at 22:41 , David Lang <notifications@github.com> wrote:
> 
> 
> I figured out how to get the weight/balance in onshape and set everything so it 
> actually had mass.
> 
> I set the material for the wheels (steel frame, all plastic wheel, which isn't 
> correct, as they aren't solid, but are partially metal, so close), and 
> everything else to birch (it's probably a bit heavier due to the glue.
> 
> origin is back right corner (within a tenth of an inch or so)
> 
> onshape's estimate for the main carts is:
> 
> Mass:
> 358.345 lb
> Volume:
> 13402.188 in3
> Surface area:
> 38738.104 in2
> Center of mass:
> X: -36.470 in
> Y: -9.011 in
> Z: 35.698 in
> Moments of inertia: lb in2
> Lxx: 2.321e+5	Lxy: -1.707e-1	Lxz: -2.432e-1
> Lyx: -1.707e-1	Lyy: 4.373e+5	Lyz: -7.690e+3
> Lzx: -2.432e-1	Lzy: -7.690e+3	Lzz: 2.445e+5
> 
> This has the CG a fair bit closer to the back than I expected.

For an empty cart, this doesn’t surprise me. Everything we put on the cart
will tend to move that somewhat forward.

> monitor cart is about 100 lb heavier:
> 

Sounds about right.

> Mass:
> 458.895 lb
> Volume:
> 17378.216 in3
> Surface area:
> 50828.702 in2
> Center of mass:
> X: -36.531 in
> Y: -7.659 in
> Z: 37.649 in
> Moments of inertia: lb in2
> Lxx: 2.554e+5	Lxy: 1.129e+0	Lxz: 1.730e+2
> Lyx: [...]

That also sounds about right. The monitors on the lower shelves probably help somewhat as well.

> we'll have to see what the loaded balance is, but this may mean that we need to 
> load them facing the other way and think about doing something to keep the bins 
> on the shelves when facing downhill.

I think we’ll be OK on all but the monitor cart. For the monitor cart, I think a redesign is in order
regardless. Hopefully we can save most(all) of the shelves and the base in that process and only
replace the sides.

IIRC, we can make new sides out of a single sheet of plywood.

Owen

davidelang, 2020-03-19:
On Thu, 19 Mar 2020, owendelong wrote:

> Is onshape calculating the volume of the solids and not counting free space?

I believe so.

>> On Mar 18, 2020, at 22:41 , David Lang <notifications@github.com> wrote:
>>
>> onshape's estimate for the main carts is:
>>
>> This has the CG a fair bit closer to the back than I expected.
>
> For an empty cart, this doesn’t surprise me. Everything we put on the cart
> will tend to move that somewhat forward.

unless things fall to the back of the bins (which should only happen on fairly 
empty bins, so I wouldn't expect it to matter)

> I think we’ll be OK on all but the monitor cart. For the monitor cart, I think a redesign is in order
> regardless. Hopefully we can save most(all) of the shelves and the base in that process and only [...]

Yes we could.

background:

the right side (as you look at the front of the cart) is not glued, 
it will come off easily, but getting it on is hard.

the left side had glue applied, but much of it was not screwed down for a couple 
hours (so the glue dried a fair bit before the joint was squeezed tight), so the 
strength of the glue is questionable.

the back-to-bottom joint is very solidly glued together.

we can move the back forward 4-6" to have the monitors in the same position 
relative to the front as the old monitor cart (I need to measure exactly)

There is a matrix of possibilities here with three options on what to do with 
the sides, and three on what to do with the back.

1. for the sides
   a. get them free from the base, but leave them attached to the shelves, trim 
4" from the front of the sides, move forward and add filler strips to the back.
   b. detach them entirely and cut new slots in the exiting plywood and 
re-attach. This is a bit ugly as the slots would overlap (the new ones being 
.35" lower than the old ones) but the uglyness starts ~4" back from the front of 
the shelves (and is visible from the back)
   c. get a new sheet of plywood and cut completely new slots. Since the shelf 
spacing on the back is not quite even, fitting the sides is hard.

2. for the back
   a. cut the back just below the bottom shelf, leaving the bottom part in place 
so we have the full depth of the cart for storage. The drawback is this reduces 
the strength a little bit as the bottom 8" would rely only on the strenth of the 
sides.
   b. cut the back flush with the top of the bottom, moving it forward to rest on 
the bottom. this would reduce the depth available for storage by the amount we 
move the monitors forward.
   c. cut the back flush with the bottom, cut a slot on the bottom and reduce the 
height of the cart by ~1/4-3/8" to have the back rest in the slot. This would 
reduce the depth available for storage as in option b and the height a smidge, 
but would be the strongest.

currently  I am leaning towards 1a, 2a falling back to 1c if I damage the sides 
too much.

David Lang

owendelong, 2020-03-19:
> On Mar 19, 2020, at 10:57 , David Lang <notifications@github.com> wrote:
> 
> 
> On Thu, 19 Mar 2020, owendelong wrote:
> 
> > Is onshape calculating the volume of the solids and not counting free space?
> 
> I believe so.

OK… I guess that’s useful for something, but seems largely useless in this context.

> unless things fall to the back of the bins (which should only happen on fairly
> empty bins, so I wouldn't expect it to matter)

Even if things fall to the back of the bins, the combination of the inward slant of the
back of the bin due to injection molding plus the angle of the shelves results in some
distance inboard for the bottom rear corner of the bin. ;-)

> Yes we could.
>
> background:
>
> the right side (as you look at the front of the cart) is not glued,
> it will come off easily, but getting it on is hard.

The advantage to having shelves recessed into slots in the back panel is strength.
The disadvantage was great complexity and difficulty of assembly.

> the left side had glue applied, but much of it was not screwed down for a couple 
> hours (so the glue dried a fair bit before the joint was squeezed tight), so the 
> strength of the glue is questionable.

It’s probably strong enough to be annoying, but not strong enough to hold up to actual
use. Murphy’s law dictates this.

> the back-to-bottom joint is very solidly glued together.

That’ll be fun, but IIRC if the back ends up being 1/2” short because we took it off of
the bottom with a sawzall, that won’t be the end of the world.

> we can move the back forward 4-6" to have the monitors in the same position 
> relative to the front as the old monitor cart (I need to measure exactly)

IMHO, we want to come as close as possible to having the monitor bases on the
innermost position of the front line of the contact-patch-box. (i.e. wheel position
with cart moving towards the front of the cart).

This should provide us a CG well inside of that line and nowhere near the back
line as well. I think that using some additional plywood to make a small pocket
at the back for things like plastic carts, whiteboards, easels, etc. which are light
light weight in nature would probably be OK and might prove useful.

> There is a matrix of possibilities here with three options on what to do with
> the sides, and three on what to do with the back.
>
> 1. for the sides
> a. get them free from the base, but leave them [...]

Personally not a fan of this idea.

> b. detach them entirely and cut new slots in the exiting plywood and 
> re-attach. This is a bit ugly as the slots would overlap (the new ones being 
> .35" lower than the old ones) but the uglyness starts ~4" back from the front of 
> the shelves (and is visible from the back)

For $70, I think we can avoid the ugliness.

> c. get a new sheet of plywood and cut completely new slots. Since the shelf 
> spacing on the back is not quite even, fitting the sides is hard.

Could we possibly re-route the back to even up the spacing with firing strips
as necessary to fill in?

> 2. for the back
> a. cut the back just below the bottom shelf, leaving the bottom part in place 
> so we have the full depth of the cart for storage. The drawback is this reduces 
> the strength a little bit as the bottom 8" would rely only on the strenth of the 
> sides.

If I’m envisioning this correctly, you’re actually now supporting the entire back
only from the sides. Would it be possible to build a box with a center support
that would support the cut back? All of the storage containers I’ve found that
are <8” tall turn out to be <7” tall anyway, so invading this inch to provide
structural integrity isn’t a major loss.

> b. cut the back flush with the top of the bottom, moving it forward to rest on 
> the bottom. this would reduce the depth available for storage by the amount we 
> move the monitors forward.

This isn’t my first choice as I’d like to be able to stick keyboards down there
with the long axis of the keyboard oriented front to back.

> c. cut the back flush with the bottom, cut a slot on the bottom and reduce the 
> height of the cart by ~1/4-3/8" to have the back rest in the slot. This would 
> reduce the depth available for storage as in option b and the height a smidge, 
> but would be the strongest.

I’m not of the impression that we have a significant front-to-back load being applied
against the back of the monitor cart. As such, I don’t think the structure gained
from this is necessarily significant in our use case.

> currently I am leaning towards 1a, 2a falling back to 1c if I damage the sides 
> too much.

My vote is for 1c, 2a (modified as proposed above).I realize that 1a is both
cheaper and easier, but I’m not wild about what I see when I picture the end
result. With the amount of money we spent on these carts, tossing in another
$70 to fix our mistakes is OK by me. I’ll pitch in $70 if SCaLE is hurting and
nobody else wants to pay for this.

Owen

davidelang, 2020-03-19:
On Thu, 19 Mar 2020, owendelong wrote:

>> On Mar 19, 2020, at 10:57 , David Lang <notifications@github.com> wrote:
>>
>>
>> On Thu, 19 Mar 2020, owendelong wrote:
>>
>>> Is onshape calculating the volume of the solids and not counting free space?
>>
>> I believe so.
>
> OK… I guess that’s useful for something, but seems largely useless in this context.

I agree, I was just pasting in the entire data block.

>>>> On Mar 18, 2020, at 22:41 , David Lang <notifications@github.com> wrote:
>
>> the left side had glue applied, but much of it was not screwed down for a couple
>> hours (so the glue dried a fair bit before the joint was squeezed tight), so the
>> strength of the glue is questionable.
>
> It’s probably strong enough to be annoying, but not strong enough to hold up to actual
> use. Murphy’s law dictates this.

yep

>> the back-to-bottom joint is very solidly glued together.
>
> That’ll be fun, but IIRC if the back ends up being 1/2” short because we took it off of
> the bottom with a sawzall, that won’t be the end of the world.

I am expecting that we would need to cut it off.

> IMHO, we want to come as close as possible to having the monitor bases on the
> innermost position of the front line of the contact-patch-box. (i.e. wheel position
> with cart moving towards the front [...]

I expect to have the monitor bases beyond that contact-patch-box, as-is the 
bottoms of the base is about at the front of that box.

> This should provide us a CG well inside of that line and nowhere near the back
> line as well. I think that using some additional plywood to make a small pocket
> at the back for things like plastic [...]

most of the weight is in the shelves and the moitor displays, not in the monitor 
bases.

> Personally not a fan of this idea.

I can understand that, I am thinking along these lines as it would be the 
easiest to do.

> For $70, I think we can avoid the ugliness.

that's fine. We will need to get the wood and get it cut to size.

>> c. get a new sheet of plywood and cut completely new slots. Since the shelf
>> spacing on the back is not quite even, fitting the sides is hard.
>
> Could we possibly re-route the back to even up the spacing with firing strips
> as necessary to fill in?

no, we would need to re-make the back (which would require 2 sheets of plywood, 
assuming we can get the shelves out of the existing back intact, shelves would 
add another 3-4 sheets, basically only re-using the base)

This has been a been a very big lesson to me about the weakness of repeatability 
via measurements. If our jig to cut the back had worked I think we would have 
very little trouble with the assembly, but the slight variation in manually 
marking and lining up with the marks resulted in a very hard-to-fit structure.

> If I’m envisioning this correctly, you’re actually now supporting the entire back
> only from the sides. Would it be possible to build a box with a center support
> that would support the cut back? All [...]

that is correct, I don't have any concerns about supporting the weight. the base 
provides side-to-side support. We could tie the upper and lower back together 
with a horizontal panel to add more strength back.

while the monitor cart is much heavier empty, I expect that it is going to be 
much lighter

> This isn’t my first choice as I’d like to be able to stick keyboards down there
> with the long axis of the keyboard oriented front to back.
>
>
> I’m not of the impression that we have a significant [...]

I agree with both

> My vote is for 1c, 2a (modified as proposed above).I realize that 1a is both
> cheaper and easier, but I’m not wild about what I see when I picture the end
> result. With the amount of money we spent on [...]

my biggest concern is the glue on the left side, 1a avoids taking that joint 
apart, while 1c requires that it come apart pretty cleanly (or that we can cut 
the remains off cleanly). I'm concerned that we may end up with an unusable 
monitor cart if we start down 1c.

David Lang

owendelong, 2020-03-19:
[snip]
> I expect to have the monitor bases beyond that contact-patch-box, as-is the
> bottoms of the base is about at the front of that box.

Hmmm…

> most of the weight is in the shelves and the moitor displays, not in the monitor
> bases.

OK… Hard for me to judge, but it seemed to me that most of the weight of those
HP crap monitors (which are most of our monitors) was in the base, not the
monitor itself.

> I can understand that, I am thinking along these lines as it would be the
> easiest to do.
>
>
> that's fine. We will need to get the wood and get it cut to size.

If we can find a way to transport it from Mountain View, I have a source up there
that keeps it in stock and has a nice panel saw. Not sure if they charge per cut,
but since it’s only 2 cuts…

Come to think of it, they’ll sell us half-sheets. As such, I’m betting we can get them
to consider it 1 paid cut to lop off the top before they turn it into half-sheets.

> no, we would need to re-make the back (which would require 2 sheets of plywood,
> assuming we can get the shelves out of the existing back intact, shelves would
> add another 3-4 sheets, basically [...]

Yeah, that’s probably not feasible.

> This has been a been a very big lesson to me about the weakness of repeatability
> via measurements. If our jig to cut the back had worked I think we would have
> very little trouble with the [...]

Yeah… I was really counting on this all being CNC’d. Unfortunately, when that failed,
we were all put in a position of improvising on short notice and things went south.

> that is correct, I don't have any concerns about supporting the weight. the base
> provides side-to-side support. We could tie the upper and lower back together
> with a horizontal panel to add more [...]

Agreed… The weight of the monitors is negligible compared to the mass
density of things like buckets full of cables, switches, etc.

My thinking rather than tie the backs together with a horizontal panel was to trim
down the bottom of the top section sufficiently to essentially enclose the bottom
area (cover the top of it leaving only the front open with a stiffener holding the top
up in the middle) and then place the shortened top section on top of that.

> I agree with both
>
>
> my biggest concern is the glue on the left side, 1a avoids taking that joint
> apart, while 1c requires that it come apart pretty cleanly (or that we can cut
> the remains off [...]

I think if we cut it off the shelves and back with a flush-trim router bit, we should
be able to get a pretty clean cutout. Admittedly, that doesn’t get the last bits at the
end of the shelves off, but once we’ve trimmed it down to that point with the router,

The trick after that removes most of the side is removing the bit at the end. I think
judicious application of a sawzall can do that with minimal risk.

I admit it’s not ideal, but I think it’s achievable.

Owen

davidelang, 2020-03-20:
On Thu, 19 Mar 2020, owendelong wrote:

> [snip]
>
> Hmmm…
>
>
>
> OK… Hard for me to judge, but it seemed to me that most of the weight of those
> HP crap monitors (which are most of our monitors) was in the base, not the
> monitor itself.

this is why we are going to get numbers from the actual carts before doing 
anything.

> If we can find a way to transport it from Mountain View, I have a source up there
> that keeps it in stock and has a nice panel saw. Not sure if they charge per cut,
> but since it’s only 2 cuts…
>
> Come [...]

it's 3 cuts (still minimal cost), can you fit a 23x70" board in the prius?

1. cut it down to length (~70" IIRC)
2,3. cut it to width, two panels @ 23" each (leaving an extra 2" wide piece)

> Yeah, that’s probably not feasible.

note, the back + sides is 2 sheets of plywood, not +2 sheets for the back. The 
question would be if the shelves come free from the back or not.

> Yeah… I was really counting on this all being CNC’d. Unfortunately, when that failed,
> we were all put in a position of improvising on short notice and things went south.

yeah, although the CNC would not have been able to do the angled cut, so we 
would have needed to do a wider cut and/or do beveled cuts on the backs of the 
shelves.

> My thinking rather than tie the backs together with a horizontal panel was to trim
> down the bottom of the top section sufficiently to essentially enclose the bottom
> area (cover the top of it leaving [...]

We may be talking about a similar thing. the back ending about flush with the 
bottom shelf and everything below that being an enclosed box (including a top on 
the box).

the back panel will result in more than enough strength that there won't be a 
need for a center support (a 1/2" sheet would be overkill for that)

> I think if we cut it off the shelves and back with a flush-trim router bit, we should
> be able to get a pretty clean cutout. Admittedly, that doesn’t get the last bits at the
> end of the shelves off, [...]

using a flush trim bit on the edge of a board is non-trivial, but doable (the 
problem is the router base on the 3/4" edge will have a tendency to tip and not 
cut flush)

> The trick after that removes most of the side is removing the bit at the end. I think
> judicious application of a sawzall can do that with minimal risk.

a sawzall isn't good for trying to do straight cuts, but a circular saw with a 
good fence can do it.

> I admit it’s not ideal, but I think it’s achievable.

It's possible.

David Lang

owendelong, 2020-03-20:
> this is why we are going to get numbers from the actual carts before doing
> anything.
>
>
> it's 3 cuts (still minimal cost), can you fit a 23x70" board in the prius?

Probably not, but once the lockdown ends, I can probably find a friend that can get
the lumber to my house.

If we get it down to 2@24x70 (2 cuts), I can take the remaining inch off of each of
those with my table saw using the rollers I have easily enough.

I realize that’s a bit more wasteful than having a single 2” wide scrap piece, but I
doubt we would make much use of the 2” piece anyway.

I’ll see what they’ll charge once I can actually go get the lumber and proceed accordingly.

> 1. cut it down to length (~70" IIRC)
> 2,3. cut it to width, two panels @ 23" each (leaving an extra 2" wide piece)
>
>
> note, the back + sides is 2 sheets of plywood, not +2 sheets for the back. The [...]

Right.

Getting the shelves off the back would be relatively easy…

1.	Cut the back (between the shelves) off with a flush-trim router bit.
2.	Use a table saw to remove the remaining back glued to the shelf.

> yeah, although the CNC would not have been able to do the angled cut, so we
> would have needed to do a wider cut and/or do beveled cuts on the backs of the
> shelves.

I’m doing beveled cuts on the backs of shelves for the library cart I’m building. In
hind sight, that’s probably what we should have done here, IMHO.

> We may be talking about a similar thing. the back ending about flush with the
> bottom shelf and everything below that being an enclosed box (including a top on
> the box).

I suppose that would work. I was thinking in terms of the back extending under the
bottom shelf all the way to the front.

> the back panel will result in more than enough strength that there won't be a 
> need for a center support (a 1/2" sheet would be overkill for that)

Well, I was thinking in terms of this supporting the back panel since it bears a lot
of the weight from all the shelves. If it’s free floating, then the full span of each shelf
is essentially only supported by the sides.

> using a flush trim bit on the edge of a board is non-trivial, but doable (the
> problem is the router base on the 3/4" edge will have a tendency to tip and not
> cut flush)

So my thinking with the flush trim bit was to have the router base resting on the board
being cut. For the sides, we’d follow the edges of the shelves and back. For the back
(if we decided to replace the back), we’d follow the edges of the shelves.

Once we’ve cut off most of the side using the router, we could (if we can detach from the back)
then trim the remainder off the end of the shelves with a table saw.

I wasn’t thinking in terms of running the router along the narrow edge of any of the boards.
(Down that path lies madness, destruction, and probably injury).

> > The trick after that removes most of the side is removing the bit at the end. I think
> > judicious application of a sawzall can do that with minimal risk.
> 
> a sawzall isn't good for trying to do straight cuts, but a circular saw with a 
> good fence can do it.

Getting a circular saw in between shelves still attached to the back with a good fence would
be interesting IMHO, but go for it if you think you can do it.

A sawzall isn’t good for doing straight cuts most of the time, but for trimming off the remainder
of the old side from a board that’s glued into it, we have two advantages:

1.	It doesn’t have to be super straight as long as it stays close to the glue line.
2.	We should have a visible glue-line to follow.

Another alternative given the small amount of material to be removed would be an aggressive
application of a belt sander with some 40 or 60 grit. ;-)

Owen

davidelang, 2020-03-20:
On Thu, 19 Mar 2020, owendelong wrote:

> Probably not, but once the lockdown ends, I can probably find a friend that can get
> the lumber to my house.
>
> If we get it down to 2@24x70 (2 cuts), I can take the remaining inch off of each of
> those [...]

normally it's something like $1/cut

> I’m doing beveled cuts on the backs of shelves for the library cart I’m building. In
> hind sight, that’s probably what we should have done here, IMHO.

I agree

> I suppose that would work. I was thinking in terms of the back extending under the
> bottom shelf all the way to the front.
>
>
> Well, I was thinking in terms of this supporting the back panel since it [...]

the back prevents the shelves from sagging and helps transfer the weight to the 
sides (where there is the long joint to hold the weight, not just the board 
ends).

the back also prevents the sides/shelves from racking side to side

when we cut off the back for the bottom 8", we loose the resistance to the 
racking force. but adding a horizontal brace (either on just the back 4" or all 
the way to the front) adds this back in.

We can always add in a center brace if it seems like we need to. I would try 
leaving it out first.

> So my thinking with the flush trim bit was to have the router base resting on the board
> being cut. For the sides, we’d follow the edges of the shelves and back. For the back
> (if we decided to [...]

we need to cut the sides flush with the top/bottom/front/back of the shelves 
this puts the shaft of the router parallel to the long edge of the shelves. so 
the base of the router ends up resting on the narrow end of the board.

for the first side, you have good support, but as you cut around the end, you 
loose your support and end up resting on the 3/4" think shelf (or the part of 
the side that extends it).

if we re-do the back, and have to cut the back free of the shelves, this will be 
even worse as the back is angled compared to the shelves.

This can be worked around, but you need to build a jig to hold the router to let 
you do this safely.

> Once we’ve cut off most of the side using the router, we could (if we can detach from the back)
> then trim the remainder off the end of the shelves with a table saw.
>
> I wasn’t thinking in terms of [...]

I my be missing some trick here, but I don't think so.

> Getting a circular saw in between shelves still attached to the back with a good fence would
> be interesting IMHO, but go for it if you think you can do it.
>
> A sawzall isn’t good for doing straight [...]

if the cut line isn't stright, it will be hard to get a good glue joint on the 
new sides

something like 
https://www.kregtool.com/store/c48/saw-attachments/p424/rip-cuttrade/ is what I 
was thinking for cutting the sides/back away from the bottom, and it could be 
used to trim the 1/2" of the ends.

cutting along the bottom of the shelf is a bit more difficult, but if we aren't 
trying to save the sides, we can screw a board to the sides across the back to 
guide the saw. Which is what I'm thinking to cut flush with the bottom shelf 
(I'd drill a couple holes from the front to act as guide points and then align 
the guide board with those holes)

> Another alternative given the small amount of material to be removed would be an aggressive
> application of a belt sander with some 40 or 60 grit. ;-)

it's about a half inch we need to cut away, that's a lot of sanding.

David Lang

owendelong, 2020-03-20:
[snip]

> the back prevents the shelves from sagging and helps transfer the weight to the
> sides (where there is the long joint to hold the weight, not just the board
> ends).
>
> the back also prevents the [...]

Yeah, but in that constrained space, it’s a whole lot easier to put it in before we put everything
on top of it.

> we need to cut the sides flush with the top/bottom/front/back of the shelves
> this puts the shaft of the router parallel to the long edge of the shelves. so
> the base of the router ends up resting [...]

This probably won’t come through on GitHub so I sent it to david. I was picturing a cut path
like this for the router on the side panel… Rather than attempt another 1,000 word explanation,
here’s a picture:

(conceptual, cuts might not be in the right places or to scale)

I’m having trouble seeing where the router is supported on the long edge of
a board during that cut. That will free the bulk of the side from the shelves, back,
and base. What will be left will be a ~1/2” wide piece glued to the end of each
shelf, the back, etc.

That’s the part I was figuring could be trimmed off with a sawzall.

owendelong, 2023-09-13:
This wandered far afield from the training intended to be the subject matter.

Most of the discussion is rendered moot by the new monitor cart.

I still believe that we should have a required cart safety training for individuals involved in cart movement at the start of each cart movement process. Individuals which have not been trained or which David, Rob, or myself identify as not sufficiently competent in the safe handling of the carts should be excluded from cart handling, especially onto/off of the lift gate until one of us can address the issue. The intent here is to ensure everyone's safety, including the individuals in question.

I'm going to go ahead and close out this issue, but let's treat cart safety training as a mandatory item at the start of each major cart movement and any time it appears to be needed.

closed 2023-09-13 by owendelong

davidelang, 2023-09-13:
note that if we use the trailer instead of a truck, this becomes much easier.

David Lang

On Tue, 12 Sep 2023, Owen DeLong wrote:

> This wandered far afield from the training intended to be the subject matter.
>
> Most of the discussion is rendered moot by the new monitor cart.
>
> I still believe that we should have a required cart [...]