socallinuxexpo/scale-network #358: Create permission level for Configuration-Only changes without Review

owendelong, 2020-03-11:
## Description
Sometimes a minor configuration change needs to be made to the network which really doesn't warrant a full review prior to committing it to master.

Ideally, I'd like to have an access level defined where we can add a few select users that are able to merge their own commits if the change doesn't involve any code and only affects 3 or fewer configuration files.

## Acceptance Criteria
Creation of an "authorization group" in GitHub and a list of files that represent configuration rather than code and a process where an automated "review" (e.g. passing circlci checks) is considered adequate for members of that group who are committing changes to ≤3 configuration files can merge the commit once automated testing is passed without requiring a second person to review.

owendelong, 2020-03-11:
This issue is created for discussion of the topic. Do not act on it yet.

sarcasticadmin, 2020-03-11:
@owendelong I dont think this is possible (out of the box) due to the permissions model used by github. It does not make a distinction about which files have changed then allowing you to skip to peer review process. Instead its based on the branch protections for the branch where the changes are targeted, `master` in our case.

Regardless, I dont think there was much lag on configuration change approval to the repo this year and things seemed to work pretty well from what I could tell.

owendelong, 2020-03-11:
Yeah, if it’s too hard (or impossible) than no worries. Lag wasn’t the issue, but it seemed like it was an unnecessary interruption for trivial changes so I figured I’d put it out as a possibility to discuss.

Owen

closed 2020-03-19 by owendelong