socallinuxexpo/scale-network #359: One sourcew of truth for the ssh keys?

sbibayoff, 2020-03-19:
## Description
We currently have at least 2 file[s] locations where public ssh keys are stored on the githib repo, is it possible to store this info in one area and have that pulled into whatever coinfigs need it?

## Acceptance Criteria
This was people who need to up[load, date] keys don't have to do in multi locations.

owendelong, 2020-03-19:
Yes, this should be doable. Can you link the two locations here so that it's easier to find them?

sarcasticadmin, 2020-03-25:
@owendelong :heavy_check_mark: Updated the description on the issue to include the two locations.

kylerisse, 2021-03-19:
Round tuit:

I think the permanent version is the one in switch configuration. there is a file per user which is how the junipers and ansible consume them. The openwrt version can be generated.

Given that, are you looking for something like this?

1. remove authorized key file from openwrt directory
2. create a script to generate the authorized_key file by looping through the keys in the switch configuration directory
3. modify the makefile to run this script when building openwrt and putting the file in the current location
4. modify the gold image testing to generate another copy of this file using the script and putting it in the gold image directory

The downside to this approach is that it doesn't capture key updates during the show, which is the same thing happening today. It does dedup the data though, which is the primary concern. Perhaps live updates of the file is another topic.

sarcasticadmin, 2021-03-24:
@kylerisse I would like to make a small tweak to your proposal

> 2.  create a script to generate the authorized_key file by looping through the keys in the switch configuration directory

I think we should put the public keys that would be compiled into `authorized_keys` in the `facts/` dir. I think that makes the most sense as many of the source of truth come from this part of the repo already. Downside being this would require modifying both switch-configuration scripts and our openwrt build process.

> The downside to this approach is that it doesn't capture key updates during the show, which is the same thing happening today. It does dedup the data though, which is the primary concern. Perhaps live updates of the file is another topic.

As for your downside about this, we dont really support live updates as it is today. We should be setting the expectation that we gather all relevant tech team members pub keys prior to image finalization. Its not a big deal to update it after that but this should alleviate the problem for the most part.

kylerisse, 2021-03-24:
Great. Agreed on the file system location, more true to the intended hierarchy. I can tackle this.

PR #379 (2021-10-02): [READY] Deduplicate SSH Public Keys
Fixes #359 

## Description of PR
Moves SSH Public Keys into the facts directory and replaces the static openwrt root authorized_keys file with one dynamically generated through gomplate.

## Previous Behavior
Stored SSH Keys in switch configuration directory and had to manually update openwrt root's authorized_keys file manually.

## New Behavior
Update SSH Key in /facts and you're done. OpenWRT golden files have been updated.

## Tests
- verified gomplate template procedure mostly using the test.sh workflow
- kicked off a quick build of openwrt using make  [... cut]
closed 2021-10-02 by sarcasticadmin