socallinuxexpo/scale-network #360: Improve CG and/or Stability of Monitor cart

owendelong, 2020-03-19:
## Description
Monitor cart is too unstable front-to-back. CG either needs to be brought more inside the
box defined by the contact patch and/or lowered closer to the ground or contact patch box needs to be expanded (outriggers?).

## Acceptance Criteria
Monitor cart is stable and does not exhibit strong tipping tendency.

davidelang, 2020-03-19:
My initial plan  is to move the shelving/back ~4" forward.

I am currently thinking to cut the back just below the bottom shelf and am debating between moving the sides forward (and trimming the front of the sides 4", then put filler pieces behind them) or re-cutting the sides

one side is only screwed on, not glued, the other side is glued, but the side was not clamped in place in a timely manner, so the glue may not have a good hold.

sarcasticadmin, 2021-10-04:
@owendelong @davidelang Can you summarize whats transpired at the work party? That way I think we can close this out

davidelang, 2021-10-04:
The monitor cart was cut and the top part that had the monitor shelves was moved 
forward 3" and the cart spliced back together to improve the stability.

There is a little bit of work to do, but it's minor tweaks (rounding edges and 
that sort of thing), the major work was completed.

On Sun, 3 Oct 2021, Robert James Hernandez wrote:

> Date: Sun, 03 Oct 2021 18:17:41 -0700
> From: Robert James Hernandez ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: David Lang ***@***.***>, Mention ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] Improve CG and/or Stability of
>     Monitor cart (#360)
> 
> @owendelong @davidelang Can you summarize whats transpired at the work party? That way I think we can close this out
>
>

closed 2021-10-07 by sarcasticadmin