socallinuxexpo/scale-network #364: change network test script from pinging 8.8.8.8 to pinging the gateway

davidelang, 2020-03-21:
## Description
when there is a problem connecting to the Internet, we don't want to loose wifi connectivity to the local equipment (which we may need to get to to fix the problem)

we are no better off having wifi not advertising than having it up with no Internet connectivity, but full internal connectivity.

We may also want to consider having the network test script trigger a network profile change, or program one of the buttons on the AP to trigger a change if held down long enough (so that if we get into a really strange/broken network config with a new profile, we have a way to revert to the .0 profile to recover)

sarcasticadmin, 2020-03-25:
There are limations to `apinger` about how its able to trigger an outage. It can only monitor a single endpoint as far as I know. So we might have to template out the config file and generate it on the fly so it can support either buildings gateway.

Regarding the network profile change: Id prefer to have the button trigger the AP to go back to config `0`.

davidelang, 2020-03-25:
On Tue, 24 Mar 2020, Robert James Hernandez wrote:

> There are limitations to `apinger` about how its able to trigger an outage. It can only monitor a single endpoint as far as I know. So we might have to template out the config file and generate it on the fly so it can support either buildings gateway.

there should only be one default route, so can the script just do a netstat -nr 
and extract the gateway from that?

> Regarding the network profile change: Id prefer to have the button trigger the AP to go back to config `0`.

that works for me.

David Lang

owendelong, 2020-03-25:
> On Mar 25, 2020, at 01:08 , David Lang <notifications@github.com> wrote:
> 
> 
> On Tue, 24 Mar 2020, Robert James Hernandez wrote:
> 
> > There are limitations to `apinger` about how its able to trigger an outage. It can only monitor a single endpoint as far as I know. So we might have to template out the config file and generate it on the fly so it can support either buildings gateway.
> 
> there should only be one default route, so can the script just do a netstat -nr 
> and extract the gateway from that?

Looking at:

https://github.com/Jajcus/apinger/blob/master/src/apinger.conf

Looks like multiple “targets” can be defined and we have good flexibility on what parameters and how often are evaluated.

Unfortunately, it doesn’t look like we have the option of allowing “either target” good = good vs. “either target” bad = bad
which appears to be the default.

However, I suspect we could have the script that the alarm runs provide the required intelligence there.

(It looks like the OpenWRT package apinger is a port of this code).

> > Regarding the network profile change: Id prefer to have the button trigger the AP to go back to config `0`.
> 
> that works for me.

Agreed. Having the button trigger a well known “safe” state is desirable.

Owen

sarcasticadmin, 2022-02-11:
Adding some notes here.

Seems like it would be easiest to do the following:
  - Omit `target` in apinger config: https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/files/etc/apinger.conf#L56-L59 also removing the gomplate var since its only referenced here
  - Update `udhcpc.user` https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/files/etc/udhcpc.user to in place add the results of default route (`netstat -rn`) to `apinger.conf` and hup/restart apinger on `bound`

Seems like the most straight forward way to get what we want out of the configuration

PR #437 (2022-04-19): [READY] - Enable dynamic apinger gateway check
## Description of PR

Fixes: #364 

Updating `apinger` to leverage local gateway instead of google

## Previous Behavior
- Leveraged google for apinger, was simplest path at the time to get this up and working at the time
- udhcpc.user had duplicate logic in the case
- set apinger inputs via `openwrt*.yaml`
- apinger status was logging to `/tmp/apinger.status`

## New Behavior
- Removed apinger inputs in `openwrt*.yaml`
- `/etc/apinger.conf` moved to `/etc/apinger.tmpl` so that gateway can be added during dhcp bound and renew
- Default gateway ($r [... cut]
closed 2022-04-19 by kylerisse