socallinuxexpo/scale-network #368: Investigate the relevance for DAWN for scale wifi

sarcasticadmin, 2020-06-24:
## Description

Upon discussion from Make-wifi-fast mailing list, @davidelang noticed an interesting project called DAWN: https://github.com/berlin-open-wireless-lab/DAWN

Lets investigate to see what new capabilities (if any) would be possible with the introduction of DAWN into the scale wifi deployment.

## Acceptance Criteria
* What are the pros and cons of DAWN
* Is there a sample configuration that we can prove out against our existing builds

sarcasticadmin, 2025-03-22:
Theres also https://git.openwrt.org/project/usteer.git

MrHamel, 2026-02-26:
Using `usteer-ng` would be the safest bet unless that fix made it in to `usteer`.

Having troubleshot Wi-Fi with Ruckus and Meraki enterprise gear in co-working office environments, handing off and roaming with Apple products is a complete and total **nightmare**.

From the `usteer` link:

> **Issues with Intel Wifi cards and potentially other vendors**
> 
> According to the OpenWrt development mailing list, some devices (like some Intel Wifi cards, Apple devices) tends to ignore roaming requests as the current implementation of Usteer sends no disassociation timer in the transition frame. The transition requests is handled like a tip not like a request. If you experience your computer sticks to access points and/or frequencies you can think of using usteer-ng until the fix is merged into the main repository.

davidelang, 2026-09-24:
I have learned a lot about these protocols and tools, first the underlying protocols
802.11k is a protocol for Radio Resource Management
method 1 send a query to a client and ask it to report all APs that it hears.
method 2 provide a method for a client to ask the AP for a directory of nearby APs
method 3 add data to the beacon that has minimal data about nearby APs

802.11v is a protocol to request that a client move off of this AP/ssid (optionally provide a list of suggested new destinations, can be different ssids). The client does not have to move, and does not have to go to one of the suggested destinations, but it is better than depending on the client to decide if the signal is weak enough that it should move

802.11r is a protocol to speed up the transition from one AP to another
This requires setting up shared communications and clusters of APs
There are two methods

method 1, the client tells the AP, I plan to move to <BSSID> the AP looks up what AP has that BSSID and send data about the client to it (a few hundred bytes), the AP tells the client it has done so, the client talks to the new AP and the connection is established with far fewer communication turns and less encryption work

method 2. the client tells the AP I plan to move to <BSSID>, the AP send a message to a central server/db that server looks up what AP has that AP and sends it a notification, the client talks to the new AP and the new AP fetches data from the central db (I am a little fuzzy about exactly which data gets in the db and which data is forwarded)

method 3. over the air, similar to method 1 but the initial AP sends the message to the new AP over the air (requires that they be on the same channel)

Dawn and usteer are management layers that use these protocols to shift clients around with different criteria. Neither one scales well and would not be able to handle the scale environment

Additional problems:
when many of the 802.11k items are enabled, the client goes off channel to check on the other APs (always for method 1, but sometimes even if you just enable method 2, I am still pinning down details)

the BSSID numbers are not stable by default (mac addresses with offsets)

Proposed implementation

802.11k we only setup the data broadcasting _ send signal strength to central server

802.11v enable, but be very cautious about using. have the central server that is getting client signal strength information decide using criteria like:
1. how many other APs are close to the one it's connected to
2. is the signal strength stable or dropping
3. when did this last roam, if so from where (avoid flapping between two APs)
4. did this device refuse a prior kick
5. how busy are the aps in that area
This is potentially useful and potentially dangerous

802.11r
create a dynamic dns zone that the APs update with their BSSID information when the radios come up. stale info from dead APs is not a problem as it will never be looked up. have a caching dns server on the AP to cache this information. It will quickly get populated with the data of near enough APs  that clients use (I am unsure if the zone would just be bssid.foo or bssid.ssid.foo)

use method 1 (officially FT over DS) with encryption keys distributed so any AP can connect to any other AP (can do by building, but can also do across everything)

do not try to implement usteer or dawn, but instead implement a much more conservative/quiet process to watch the 802.11k data about client and see what can be done and what thresholds are useful.

Use the feature that lets us send a request to the client for it to scan for what APs it can hear to do a continual ping of a phone that has some sort of mapping capability (like the camera based map creation of WIFIman) and have someone walk around with it getting reports of the signal strength from each AP every few seconds. mapping out the footprint of each AP on each ssid collected and mapped on a central system rather than the all-combined map that wifiman provides. I am researching if this can be done on the existing SSIDs or if we would need to create new ssids for these tests.