socallinuxexpo/scale-network #385: Mass flash of openwrt aps

sarcasticadmin, 2021-07-23:
## Description
Considering that by and large we have the aps flashed with openwrt from a previous conference: We should be able to do a mass update via ssh: https://openwrt.org/docs/guide-user/installation/sysupgrade.cli#upgrading_openwrt_firmware_using_cli

This would save a significant amount of time turning the pre conference prep if were able to our flashing in this manner so I think its worth investigating at the very least.

Would need to leverage either existing ssh keys on the aps or the default conference password we set on those APs.

## Acceptance Criteria
- Is a mass update viable for the APs?
- What consideration would be have to make for supporting multiple versions

owendelong, 2021-07-23:
I like this idea. I suggest we prepare to test it out at our first working session (which I need to try and schedule soon).

I would propose that going forward, we create an administrative SSH key that goes into the flash images for the APs by default so that we aren't depending on the SSH key of any particular team member for this purpose.

In general, I would prefer to be able to configure things not to allow remote login as root.

owendelong, 2021-07-23:
Since we control the DHCP client on them, we should theoretically be able to set any DHCP Option field we want, including "vendor-specific" option fields.

Since we control the DHCP Server they'd be talking to, we should be able to easily process those DHCP options and send back a DHCP response that will trigger any desired behavior we like.

In an IPv6-only environment, so long as the SSH server and DHCP client on the APs can handle it and any other underlying software for flashing works with IPv6, it should be a non-issue. We can run an IPv6 DHCP server easily if it is beneficial to do so. So far, we just haven't because it wasn't needed. In short, unless there's an IPv6 incapable dependency in the chain, it should work generally the same.

>>  In general, I would prefer to be able to configure things not to allow remote login as root.
> I have mixed feelings, it's a LOT easier to automate things if you ssh in and
> can execute commands directly than if you have to sudo first, and if the account
> is able to sudo (unlimited) are you really more secure?

You responded to my statement prior to my last edit... The full statement is:
>>  In general, I would prefer to be able to configure things not to allow remote login as root with a simple password.

I have no problem with root logins authenticated by SSH key, but I prefer not to support root logins authenticated by password.

sarcasticadmin, 2021-07-24:
@owendelong @davidelang thanks for the support and ideas!

> Since we control the DHCP Server they'd be talking to, we should be able to easily process those DHCP options and send back a DHCP response that will trigger any desired behavior we like.

Totally agree and we already do this on the APs for the network & wifi configs: https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/files/etc/udhcpc.user we just dont have custom responses but that would be pretty cool too 🙂 

> In an IPv6-only environment, so long as the SSH server and DHCP client on the APs can handle it and any other underlying software for flashing works with IPv6, it should be a non-issue. We can run [...]

Glad itll work more or less the same way

>  I would propose that going forward, we create an administrative SSH key that
 goes into the flash images for the APs by default so that we aren't depending
 on the SSH key of any particular team member for this purpose.
 
 Agreed. I think we should do  this for the switches as well

> I have no problem with root logins authenticated by SSH key, but I prefer not to support root logins authenticated by password.

Yep I agree we should nuke the password from the image we use during the conference. I think id still like to keep it in the image outside the conference since it allows for easier use and testing

sarcasticadmin, 2021-07-25:
> I like this idea. I suggest we prepare to test it out at our first working session (which I need to try and schedule soon).

@owendelong add https://github.com/socallinuxexpo/scale-network/issues/386 so we can track the agenda for the 1st work party for 19x

Additionally I spun out a few more issues around the SSH public key login stuff:
- https://github.com/socallinuxexpo/scale-network/issues/387
- https://github.com/socallinuxexpo/scale-network/issues/388
- https://github.com/socallinuxexpo/scale-network/issues/389

owendelong, 2021-07-27:
We should preserve at least one password login with SUDO capability so that we can do a local login via serial if absolutely necessary.

owendelong, 2021-07-27:
> @owendelong @davidelang thanks for the support and ideas!
>
>
> Totally agree and we already do this on the APs for the network & wifi configs: [...]

What would be really cool would be if we could have the AP bring up LLDP and report it's upstream switch name and port number in a custom DHCP option so that we could assign update the MAC->IP mapping dynamically to automatically assign the AP's identity on the fly.

> Glad itll work more or less the same way

For maximum flexibility, I'd like to advocate that we look into converting from DHCPd to KEA either this year or next at the latest.

> Agreed. I think we should do this for the switches as well
>
>
> Yep I agree we should nuke the password from the image we use during the conference. I think id still like to keep it in the image [...]

I vote for keeping a defined user with SUDO for local login even at the conference, but nuke the ability to log in as root using a password.

sarcasticadmin, 2021-08-16:
@owendelong 

> What would be really cool would be if we could have the AP bring up LLDP and report it's upstream switch name and port number in a custom DHCP option so that we could assign update the MAC->IP mapping dynamically to automatically assign the AP's identity on the fly.

That sounds pretty cool! We already have lldp working from work done to prep for scale 18x: https://github.com/socallinuxexpo/scale-network/issues/33 so we could already do this with minimal effort.

> For maximum flexibility, I'd like to advocate that we look into converting from DHCPd to KEA either this year or next at the latest.

Spun out for KEA: https://github.com/socallinuxexpo/scale-network/issues/392

> I vote for keeping a defined user with SUDO for local login even at the conference, but nuke the ability to log in as root using a password.

We could do that, but there isnt a way to serial/console into these APs without opening them and connecting to the serial pins. I do like the idea of just getting rid of the passwords across the board, if we do that I dont think there would be any need to keep the images private and even our live builds could be public 🙂

PR #445 (2022-06-18): [READY] - init massflash for APs via kea dhcpd
## Description of PR
Fixes: #385

At the past scale work party we were able to `massflash` the entire set of 100+ APs in about 2 hours. The result of the teams efforts is the following script and configuration that made this possible

## Previous Behavior
- Used to have to update each AP via the existing flash (soloflash) script

## New Behavior
- `massflash` process exists
- `kea.tmpl` for setting up a `massflash` dhcpd server

## Tests
- Flashed 100+ aps at the last workparty for scale 19x
closed 2022-06-18 by owendelong