socallinuxexpo/scale-network #386: Scale 19x Work Party #1 Checklist

sarcasticadmin, 2021-07-24:
## Description
Check list for the 1st work party. Feel free to add/comment about things in prep for this meeting.

Date: TBD

Agenda:
- POC AP mass flashing. See: https://github.com/socallinuxexpo/scale-network/issues/385

davidelang, 2021-07-24:
On Sat, 24 Jul 2021, Robert James Hernandez wrote:

> ## Description
> Check list for the 1st work party. Feel free to add/comment about things in prep for this meeting.
>
> Date: TBD
>
> Agenda:
> - POC AP mass flashing. See: https://github.com/socallinuxexpo/scale-network/issues/385

modify monitor cart to move COG forward
finish cutting  shelf dividers/supports, and mount with pocket screws.
test cables and get cable inventory of every box

David Lang

owendelong, 2021-07-27:
- Verify good stable boot images on all switches
- Test and stage all cables, update cable database as needed
- Determine if a cable order is necessary
- Inventory tools/tape/cable repair supplies/etc. Determine anything that needs to be ordered
- Keep an eye towards minimizing expenses given the increased costs and reduced revenue resulting from COVID

kylerisse, 2021-10-02:
adding #394

norsehorse89, 2021-10-03:
Cable Testing:

Cart 1 Good: 1-1,1-2,1-4,1-5,1-7,1-12
Cart 1 Bad (loose tabs):
1-3: 5'(0316),20'(0382),45'(0604)
1-6: 150'(0499, NOT replaced)
1-9: 65'(0044) *(replacement cable (0627) also had loose tab)
1-10: 70'(0601)
Cart 1 Bad cables:
1-11: 110'(0567, NOT replaced)

Cart 2 Good: 2-1,2-5,2-7,2-8,2-9,2-10
Cart 2 Bad (loose tabs):
2-2:6.5'(0039)

All cables with loose tabs replaced unless otherwise specified

sarcasticadmin, 2021-10-10:
We got a lot done thanks to everyone who was able to make it to the work party the past weekend! I went ahead and update the related stories that were completed and moved for anything to https://github.com/socallinuxexpo/scale-network/issues/402 which will be the next time we all get together to work on these items.