socallinuxexpo/scale-network #387: Generate SSH key for administrative SSH access

sarcasticadmin, 2021-07-24:
## Description

Relates to: https://github.com/socallinuxexpo/scale-network/issues/359

Make the that the key we generate is compatible with the SSH servers we are trying to connect to. Consider things like key type and size. Well need to make sure that the following SSH servers:
- Juniper switches
- Openwrt APs
- Servers

## Acceptance Criteria
- Add the public key to the source of truth for our SSH keys, if https://github.com/socallinuxexpo/scale-network/issues/359 is completed first then this will be a central location
- Store the SSH private key in a save place and share it with the maintainers of the tech team

owendelong, 2022-01-28:
I believe that everything should support ED25519 at this point.

If that's the case, I propose using the ED25519 standard.

I would also suggest that this key should rotate at code freeze for each conference.

sarcasticadmin, 2022-02-11:
> I believe that everything should support ED25519 at this point.

@owendelong I agree, I think we pretty much on a current version of OpenSSH for the APs and the linux boxes will be fairly up to date as well.

Dont we already have a key for this? I thought you generated it as the 3rd work party. If not I can generate one for this use case.

closed 2022-04-13 by kylerisse