socallinuxexpo/scale-network #389: Disable password SSH login on switch production configurations

sarcasticadmin, 2021-07-25:
## Description

Relates to #387

We should disable password ssh login for our juniper switch production configurations. I think we would still have to have a password for the login via serial for physically connecting to the switch.

## Acceptance Criteria
- Disable password ssh login in our switch config for production images
- Ensure that console connectivity still works as expected

davidelang, 2021-07-25:
I agree.

David Lang

On Sat, 24 Jul 2021, Robert James Hernandez wrote:

> Date: Sat, 24 Jul 2021 17:14:07 -0700
> From: Robert James Hernandez ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: Subscribed ***@***.***>
> Subject: [socallinuxexpo/scale-network] Disable password SSH login on switch
>     production configurations (#389)
> 
> ## Description
>
> Relates to #387
>
> We should disable password ssh login for our juniper switch production configurations. I think we would still have to have a password for the login via serial for physically connecting to the switch.
>
> ## Acceptance Criteria
> - Disable password ssh login in our switch config for production images
> - Ensure that console connectivity still works as expected
>
>

owendelong, 2021-07-27:
It’s not clear to me that this is possible. Will require investigation.

To the best of my knowledge, user configuration on the Juniper switch does not allow for distinction between authentication methods allowed by access methodology.

Owen

owendelong, 2024-01-28:
This turns out to be quite simple... set system services ssh no-passwords, so I will add it to the protoconfig template.

owendelong, 2024-01-28:
Resolved via PR#662

PR #662 (2024-01-28): [READY] Issue #389 disable password based SSH logins on switches
## Description of PR
Fixes: #389 

Disable password based SSH logins on switches

## Previous Behavior
Switches allowed password or ssh-key based logins via SSH

## New Behavior
Switches only accept public-key authentication over SSH

## Tests
No testing required, small text change in proto config files.
closed 2024-01-28 by owendelong