socallinuxexpo/scale-network #390: Output switch maps in additional media format

sarcasticadmin, 2021-07-25:
## Description

Relates to: https://github.com/socallinuxexpo/scale-network/issues/286

As a result of creating the postscript switch maps, we should also create a more use friendly media format for the cable team to consume if we put together an onboarding packet. I think a PNG format would be good.

Currently messing around with `imagemagick`:

```
convert -density 150 -geometry 100%  AVSwitch-map.eps test.png
```

Results in a cutoff image so it just seems like we would need to tweak the size.

Other refs: http://www.imagemagick.org/script/command-line-options.php

## Acceptance Criteria
- Media format (PNG, etc.) that would be nice to include on a webpage or onboarding packet for cabling team
- Potentially integrate this into https://github.com/socallinuxexpo/scale-network/issues/286

davidelang, 2021-07-25:
ghostscript can also output in multiple formats. There is also the ps2* set of 
utilities that will convert postscript to many other formats

on my ubuntu latop I have:
ps2ascii    ps2epsi     ps2lexmark  ps2pdf12    ps2pdf14    ps2pk       ps2ps2
ps2eps      ps2frag     ps2pdf      ps2pdf13    ps2pdfwr    ps2ps       ps2txt

it's been a few years, but I've done a lot of raw postscript coding, what is it 
you are looking to produce?

On Sat, 24 Jul 2021, Robert James Hernandez wrote:

> Date: Sat, 24 Jul 2021 21:34:05 -0700
> From: Robert James Hernandez ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: Subscribed ***@***.***>
> Subject: [socallinuxexpo/scale-network] Output switch maps in additional media
>      format (#390)
> 
> ## Description
>
> Relates to: https://github.com/socallinuxexpo/scale-network/issues/286
>
> As a result of creating the postscript switch maps, we should also create a more use friendly media format for the cable team to consume if we put together an onboarding packet. I think a PNG format would be good.
>
> Currently messing around with `imagemagick`:
>
> ```
> convert -density 150 -geometry 100%  AVSwitch-map.eps test.png
> ```
>
> Results in a cutoff image so it just seems like we would need to tweak the size.
>
> Other refs: http://www.imagemagick.org/script/command-line-options.php
>
> ## Acceptance Criteria
> - Media format (PNG, etc.) that would be nice to include on a webpage or onboarding packet for cabling team
> - Potentially integrate this into https://github.com/socallinuxexpo/scale-network/issues/286
>
>

sarcasticadmin, 2021-07-27:
> ghostscript can also output in multiple formats.

@davidelang ya thats another viable option.

>  what is it you are looking to produce?

The main thing that came up in my discussion with some of the other tech team members is that we should be able to make these diagrams easy to consume outside the repo and tooling. Hence `pngs`, etc. that we could easily point members to outside our repo. I attached the output of the `png` with magicmagick in the original description so you get an idea.

owendelong, 2021-07-27:
We already produce 3-up PDF files automatically intended for printing on 11x17 paper (or adhesive) so that the map can be placed on top of the switch to provide 1:1 scale port map indications.

Would suggest that for any sort of cable team onboarding packet, we probably want to shrink to 8.5x11 (50% scale) and print 4 or possibly even 6 per page.

I suspect that a slight modification to the current PDF generator would easily achieve the goal if this is considered an acceptable result.

owendelong, 2021-07-27:
Rob, can you comment more on what it is that the team members are hoping to glean from these diagrams outside of the context of a map pasted to the top of the switch (which was the original intent)?

I realize that this vision wasn't properly realized at the last conference (first iteration of this process) because of communication difficulties between the person producing the maps and myself, but it's unclear to me how these diagrams are particularly useful outside of that context.

davidelang, 2021-07-27:
one thing that I can think of is ease of browsing the images during planning and 
prep. If the images can be trivially visible (without loading a pdf and reader) 
it can be easier to understand things.

David Lang

  On Mon, 26 Jul 2021, Owen DeLong wrote:

> Date: Mon, 26 Jul 2021 23:25:34 -0700
> From: Owen DeLong ***@***.***>
> Reply-To: socallinuxexpo/scale-network
>     ***@***.***>
> To: socallinuxexpo/scale-network ***@***.***>
> Cc: David Lang ***@***.***>, Mention ***@***.***>
> Subject: Re: [socallinuxexpo/scale-network] Output switch maps in additional
>     media format (#390)
> 
> Rob, can you comment more on what it is that the team members are hoping to glean from these diagrams outside of the context of a map pasted to the top of the switch (which was the original intent)?
>
> I realize that this vision wasn't properly realized at the last conference (first iteration of this process) because of communication difficulties between the person producing the maps and myself, but it's unclear to me how these diagrams are particularly useful outside of that context.
>
>
>

owendelong, 2021-07-27:
Are there really still browsers that don’t display PDFs in the browser natively?

Owen

sarcasticadmin, 2021-08-16:
> Rob, can you comment more on what it is that the team members are hoping to glean from these diagrams outside of the context of a map pasted to the top of the switch (which was the original [...]

Sure @owendelong this came up will talking with Bill the other week. He wanted to make it easier for volunteers working on the cabling team to be able to pull up these port maps via a browser. And I agreed that expecting each team member to be able to run the perl scripts and render the output themselves is unrealistic (especially if the individual doesnt have any other interactions with this repo). I can ask him for a bit more insight as to his specific needs here but it did seem like having these port maps were helpful for team members beyond just having them on the top of the switches.

A .pdf should be more than sufficient for this usecase. But I missed how we were already generating the printable ".pdf" that combines the .eps files when I originally spun out this issue.  After looking at the script: [generate_ps_maps.pl](https://github.com/socallinuxexpo/scale-network/blob/master/switch-configuration/config/scripts/generate_ps_maps.pl) It does seem like its what we need except its only to stdout and the comments at the top of `generate_ps_maps.pl` calls this out too:

```
# Currently output is to STDOUT. Might convert to sending to a file later.
```

Do you think we could generate an actual .pdf as output here? Either instead of the stdout or have it be an flag? That way it could be the artifact that we can host via a webpage for the cable team (assuming theres a genuine need for these diagrams outside of being just on top of the switch).

I'll update the description in this issue if that sounds good to you and I'll see if we can get Bill to chime in here.

sarcasticadmin, 2021-08-16:
> Would suggest that for any sort of cable team onboarding packet, we probably want to shrink to 8.5x11 (50% scale) and print 4 or possibly even 6 per page.

@owendelong One last thing I wanted to comment on. That yes, it was relates to this onboarding packet. And I think thatd be a good way to help get this port maps incorporated there.

owendelong, 2021-08-16:
Well… > filename still works on most modern systems. Heck, it even works on Windows to the best of my understanding.

Owen

sarcasticadmin, 2021-08-18:
Sure and rendering it via `ghostscript` this looks good from what I can tell (see below). But like you said @owendelong we might just need to shrink it to fit within the paper format for the packet:

```
cd scale-network/switch-configuration/config
./scripts/generate_ps_maps.pl | ps2pdf - test.pdf
```
> using ghostscript

Result: [test.pdf](https://github.com/socallinuxexpo/scale-network/files/7003679/test.pdf)

owendelong, 2021-08-18:
There are Ghostscript options to easily do that, or, for packet generation, we could add a parameter to generate_ps_maps that would generate 
the 8.5x11 style in a different directory simultaneously.

PR #393 (2021-10-02): [READY] - Adding reference pages for switches
## Description of PR
Fixes: #390 

Generating pdfs as output for the switches. This is @owendelong work from our sync up in SD. Thanks @owendelong!

## Previous Behavior
- No output for pdfs for the cable team

## New Behavior
- PS artifacts for cable team to generate .pdf
- Start of the `makefile`

## Tests
-  Build the output .ps for pdf

```
$ make switch-maps-bundle
cd config && \
perl scripts/generate_ps_refs.pl > switch-maps/bundle.ps
$ make switch-maps-bundle
make: Nothing to be done for 'switch-maps-bundle'.
```
> Caching working
closed 2021-10-02 by kylerisse