socallinuxexpo/scale-network #397: Ensure ntpd is running for APs

sarcasticadmin, 2021-10-03:
## Description

Had to run the following to setup time on the APs:

```
ntpd -q -p 1.openwrt.pool.ntp.org
```

## Acceptance Criteria
- Make sure ntpd is working
- Make sure we clean up https://github.com/socallinuxexpo/scale-network/blob/master/openwrt/scripts/local/update-resolv.conf.sh

sarcasticadmin, 2021-10-07:
> DHCP shouldn't be pushing NTP config to APs.

Perfect, I suspected that was the case but while trying to prune the branches I came across https://github.com/socallinuxexpo/scale-network/commit/49c9bad0590f29db6c2ebb422e224864d915e806#diff-2b8e9f89adaa32060d40e5f0f957126bb967983798fec88fdeae33b2f35277e3L7 . It has since been removed but I wasnt quiet sure why I had that line there when we were relinking `resolv.conf`. It was most likely due to the inputs to the build either missing or just kickstarting the ntp client to update the time once relinked.

Going to keep this open just in case as a reminder for me to check

owendelong, 2021-10-10:
> On Oct 6, 2021, at 18:26 , kylerisse ***@***.***> wrote:
> 
> 
> DHCP shouldn't be pushing NTP config to APs.
> 
> 
Why on earth not?

DHCP should be pushing NTP config to everything, unless you’ve got a really good reason not to.

> We send 2 options per subnet, domain-name-servers and routers via the subnet template, with the inventory script ensuring your primary dns is the one in the same building: https://github.com/socallinuxexpo/scale-network/blob/master/ansible/roles/dhcpserver/templates/dhcpd.conf.subnets.j2 <https://github.com/socallinuxexpo/scale-network/blob/master/ansible/roles/dhcpserver/templates/dhcpd.conf.subnets.j2>I'm surprised we aren’t sending NTP as well, and I think we should.
> We send another 4 options per static lease (APs), host-name, radio0-channel, radio1-channel, and ap-network-type via the host template https://github.com/socallinuxexpo/scale-network/blob/master/ansible/roles/dhcpserver/templates/dhcpd.conf.hosts.j2 <https://github.com/socallinuxexpo/scale-network/blob/master/ansible/roles/dhcpserver/templates/dhcpd.conf.hosts.j2>Makes sense.
> Looked at the bug. Since we're using every option we send, it doesn't seem to apply.
> 
Is there a reason we don’t want to run NTP on APs and/or provide NTP server information to the various networks?

kylerisse, 2023-03-13:
ha @owendelong! I didn't mean `DHCP shouldn't be pushing NTP config to APs` in a "that's best practice sense". More of "in this case it shouldn't be based on current config". I agree that it can, and if possible should, but in this case I think the behavior @sarcasticadmin wanted was for it not to. I believe that is still the case given the Nix config.

```
Additionally it might be a good idea to disable ntp server config being pushed to APs via the dhcp server (dont want them to conflict with what we are setting in our config): https://forum.openwrt.org/t/bug-udhcpc-makes-use-of-dhcp-options-which-werent-specifically-requested/45210/2 Confirmed to be not an issue by kyle in https://github.com/socallinuxexpo/scale-network/issues/397#issuecomment-937371132
```

nixinator, 2023-03-15:
we can run a deploy a  local ntp on something like ntp.scale.lan ..

Obviously installing a atomic clock in the noc is the solution, however this might be costly ;-).

sarcasticadmin, 2023-03-15:
> More of "in this case it shouldn't be based on current config". I agree that it can, and if possible should, but in this case I think the behavior @sarcasticadmin wanted was for it not to. 

I think that was just the case due to it not being served up via dhcpd. Id like to use our server via options if its defined, but I think we probably still need to set something like the `-g` as @davidelang mentions here: **https://github.com/socallinuxexpo/scale-network/issues/397#issuecomment-937865874** in the case that time is very far out of sync.

> I believe that is still the case given the Nix config

Correct any kind of `ntp` service is currently absent from the `core` role or any other role

sarcasticadmin, 2023-03-15:
Spun out the following for the server and kea config:

- https://github.com/socallinuxexpo/scale-network/issues/592
- https://github.com/socallinuxexpo/scale-network/issues/593

owendelong, 2024-01-28:
Given that this is now covered in #592 and #593, I'm going to close this.

closed 2024-01-28 by owendelong