socallinuxexpo/scale-network #404: Enable WPA3 wifi encryption on APs

sarcasticadmin, 2021-10-03:
## Description
WPA3 has landed in openwrt and it has a mode to support backward compatibility with WPA2. We should support this new encryption mode going forward.

## Acceptance Criteria
- WPA3 enabled with WPA2 backward compat
- WPA3 and WPA2 clients can successfully connect

sarcasticadmin, 2022-12-24:
Limitations with pis (macbooks too but only for recovery mode): https://rachelbythebay.com/w/2022/12/22/wpa3/

davidelang, 2022-12-24:
encryption is a nice-to-have, so I don't see us disabling the fallbacks any time 
soon.

Thanks for calling this out, but I don't think it will be a limitation for us.

David Lang

On Sat, 24 Dec 2022, Robert James Hernandez wrote:

> Limitations with pis (macbooks too but only for recovery mode): https://rachelbythebay.com/w/2022/12/22/wpa3/
>
>

sarcasticadmin, 2022-12-25:
Also considering the password is on our shirts during the conference :laughing: