socallinuxexpo/scale-network #409: ansible-lint failing against newer version

sarcasticadmin, 2021-10-04:
## Description

Newer version of ansible-lint from nixpkgs seems to find some new linting errors.

See results when running the following:

```
$ ansible-lint --version
ansible-lint 5.0.8 using ansible 2.11.4:
$ ansible-lint -x 403 -x 503 scale.yml
yaml: truthy value should be one of [false, true] (truthy)
roles/chrony/handlers/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/chrony/handlers/main.yml:8

yaml: truthy value should be one of [false, true] (truthy)
roles/chrony/tasks/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/chrony/tasks/main.yml:23

yaml: truthy value should be one of [false, true] (truthy)
roles/chrony/tasks/main.yml:24

yaml: truthy value should be one of [false, true] (truthy)
roles/chrony/tasks/main.yml:25

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:8

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:14

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:15

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:16

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:22

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:23

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/handlers/main.yml:24

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/tasks/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/tasks/main.yml:117

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/tasks/main.yml:118

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/tasks/main.yml:119

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/tasks/main.yml:127

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/tasks/main.yml:128

yaml: truthy value should be one of [false, true] (truthy)
roles/dhcpserver/tasks/main.yml:129

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsclient/handlers/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsclient/handlers/main.yml:8

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsclient/handlers/main.yml:9

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsclient/tasks/main.yml:16

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsclient/tasks/main.yml:18

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsclient/tasks/main.yml:19

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsserver/handlers/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsserver/handlers/main.yml:8

yaml: no new line character at the end of file (new-line-at-end-of-file)
roles/dnsserver/handlers/main.yml:9

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsserver/handlers/main.yml:9

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsserver/tasks/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsserver/tasks/main.yml:152

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsserver/tasks/main.yml:153

yaml: truthy value should be one of [false, true] (truthy)
roles/dnsserver/tasks/main.yml:154

yaml: truthy value should be one of [false, true] (truthy)
roles/noapparmor/tasks/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/noapparmor/tasks/main.yml:8

yaml: truthy value should be one of [false, true] (truthy)
roles/noapparmor/tasks/main.yml:9

yaml: truthy value should be one of [false, true] (truthy)
roles/noapparmor/tasks/main.yml:16

yaml: truthy value should be one of [false, true] (truthy)
roles/signs/tasks/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/signs/tasks/main.yml:26

yaml: truthy value should be one of [false, true] (truthy)
roles/signs/tasks/main.yml:64

yaml: truthy value should be one of [false, true] (truthy)
roles/signs/tasks/main.yml:65

yaml: truthy value should be one of [false, true] (truthy)
roles/techteam/tasks/users.yml:16

yaml: truthy value should be one of [false, true] (truthy)
roles/zabbixagent/tasks/main.yml:7

yaml: truthy value should be one of [false, true] (truthy)
roles/zabbixagent/tasks/main.yml:21

yaml: truthy value should be one of [false, true] (truthy)
scale.yml:4

yaml: truthy value should be one of [false, true] (truthy)
scale.yml:12

yaml: truthy value should be one of [false, true] (truthy)
scale.yml:20
```

## Acceptance Criteria
- Newer version of `ansible-lint` passes for our ansible code

kylerisse, 2021-10-08:
Interesting find @sarcasticadmin . This isn't a versioning issue. I looked at the Nix package and noticed they include yamllint in their build. I tested with it and get those errors now in ansible-lint. Technically the yaml linter suggestions don't match the Ansible docs but I suppose we should err on the side of Nix compatibility. I'll update the build container and make all the suggested changes.

kylerisse, 2023-03-13:
unnecessary now based on #590

closed 2023-03-13 by kylerisse