socallinuxexpo/scale-network #412: Work Party #3 Checklist

sarcasticadmin, 2021-10-10:
## Description

Two Day Work Party

Date: Jan 15th-16th
Location: Pending Confirmation

Agenda:
  - Monitoring:
    - [ ] @sbibayoff to demo his monitoring solution
    - [ ] Game plan for automation around monitoring automation
- Cabling:
  - [ ] Finish testing the remaining cable (mostly minimal)
- APs:
  - [ ] POC AP mass flashing. See: Mass flash of openwrt aps #385
  - [ ] Generate official AP image for 19x

Previous work pary work that was completed: 
- https://github.com/socallinuxexpo/scale-network/issues/386
- https://github.com/socallinuxexpo/scale-network/issues/402

## Acceptance Criteria
What is the expected outcome that resolves this issue?

owendelong, 2022-07-30:
Changed subject because monitoring is being held over as a task for SCaLE 20x (or later, sigh).

sarcasticadmin, 2023-03-23:
Closing as this was completed for 20x

closed 2023-03-23 by sarcasticadmin