socallinuxexpo/scale-network #417: Allow multiple approvals to be sufficient to override failed checks for merge

owendelong, 2022-01-11:
## Description
Currently, we can't merge the issue to fix circle-ci failures because of circle-ci failures.

I propose we eliminate this risk of circular dependency by allowing 2 (or perhaps 3) reviewers approving the merge to be sufficient even if the tests have failed.

Admittedly, this does carry the risk of a conspiracy to merge broken code, but I think both the likelihood of such an event and the long-term consequences are minimal.

## Acceptance Criteria
If multiple reviewers are made sufficient to override failed tests, merging can be completed if multiple reviewers agree, despite the failed tests.

kylerisse, 2022-01-11:
That's probably a good idea generally, but for this specific issue I believe an admin just needs to disable circle as noted here: https://github.com/socallinuxexpo/scale-network/pull/413#issuecomment-1010200826

sarcasticadmin, 2022-01-12:
> I propose we eliminate this risk of circular dependency by allowing 2 (or perhaps 3) reviewers approving the merge to be sufficient even if the tests have failed.

This makes sense, unfortunately I dont think githubs permission logic is that sophisticated :disappointed: Approvals can just be N number of contributors as far as I know. But we do have some options that we can enable. Ill call them out below.

> Admittedly, this does carry the risk of a conspiracy to merge broken code

Its reasonable that 2 reviewers who are tech team members would only exercise this only when absolutely necessary.

- Option 1) We can identify individuals who are able to bypass the failed checks. Its the bottom checkbox: `allow specified actors to bypass all pr requirements` in the repo options here.
[image]
- Options 2) Elevate specific individuals from `write` to `admin` since an admin can override the PR and merge regardless. Currently this only applies to me and I think individuals like Ilan. Which is why in the case of #413 I can override the test failures
[image]

sarcasticadmin, 2022-11-22:
I believe we have sufficient permissions at the moment to override this when necessary. Opting for option 2 in the above call outs with @owendelong and myself being admins of this repo.