socallinuxexpo/scale-network #425: Update network config for LAX Hilton location change

kylerisse, 2022-03-20:
## Description
Currently the switches and vlans have names and configuration for PCC. Since we're moving back to the LAX Hilton for 2022, we'll need to rename them and change the topology a bit.

## Acceptance Criteria

- Data files in [switch-configuration directory](https://github.com/socallinuxexpo/scale-network/tree/master/switch-configuration/config) have new names, IPs, etc associated with the LAX Hilton instead of PCC

kylerisse, 2022-03-20:
One question around this is whether or not we'll store both hypervisors in the NOC or if we wanted to run one in another area of the building. Since we're not routing they can probably both be on the same VLAN, but the physical location will be a consideration when defining the network ports.
There's also a tickets for APs and Pis which will probably need similar consideration.

davidelang, 2022-03-20:
IIRC there is no second location at the Hilton that would be a somewhat central 
place to run the second server, so I would plan for them to both be in the NOC

David Lang

owendelong, 2022-03-21:
I haven’t seen the “fiber served” wiring closets, so I don’t know if one of them would be an
appropriate location or not.

If one of them has the appropriate space/power/temperature characteristics, that might be
a preferable option.

Owen

sarcasticadmin, 2022-11-22:
Changed initially for LAX 19x via: 
 - https://github.com/socallinuxexpo/scale-network/pull/452
 
Since moved back for PCC 20x via:
 - https://github.com/socallinuxexpo/scale-network/pull/506

closed 2022-11-22 by sarcasticadmin