socallinuxexpo/scale-network #426: Update server config and automation for LAX Hilton location change

kylerisse, 2022-03-20:
## Description
Currently the servers and related automation are setup for PCC and make assumptions about there being 2 buildings, specifically called `Expo` and `Conference`. Since we're moving back to the LAX Hilton for 2022 we will have a different network topology. The IPs, scripts, and Ansible roles will need to be adjusted accordingly. This work relies on #425 .

## Acceptance Criteria
- Update the [server list](https://github.com/socallinuxexpo/scale-network/blob/master/facts/servers/serverlist.csv) with the new IPs
- Update the [Vagrantfile](https://github.com/socallinuxexpo/scale-network/blob/master/ansible/Vagrantfile) to match the new server list.
- Determine if any changes need to be made to the [inventory script](https://github.com/socallinuxexpo/scale-network/blob/ef507b6130c33c0f1ea45c2f9bfe2fe3e4249e2d/ansible/inventory.py#L71) based on new topology
- Determine if any changes need to be made to the [DHCP server](https://github.com/socallinuxexpo/scale-network/blob/ef507b6130c33c0f1ea45c2f9bfe2fe3e4249e2d/ansible/roles/dhcpserver/templates/dhcpd6.conf.subnets.j2#L5) role based on new toplogy
- Determine if any changes need to be made to the [DNS server](https://github.com/socallinuxexpo/scale-network/blob/master/ansible/roles/dnsserver/templates/named.conf.scale-acls.j2) role based on new toplogy (specifically DNS views)

sarcasticadmin, 2022-07-26:
Updates from monday specifically for `core` role: https://github.com/socallinuxexpo/scale-network/pull/461

sarcasticadmin, 2022-11-22:
Closing as majority of this was dealt with in #461 and 19x has passed

closed 2022-11-22 by sarcasticadmin