socallinuxexpo/scale-network #456: Collect Management port MAC addresses of all switches and add to switch types

owendelong, 2022-07-19:
## Description
To facilitate bulk loading switch configurations at SCaLE 19x, we will need to be able to identify switches by their management MAC address to know which config file to install.

## Acceptance Criteria
Collect the MAC address for the management port from each switch and add it to the switchtypes file.

Unfortunately, I think this will need to be done by hand, and it will need to be done early in the setup process.

owendelong, 2022-07-20:
Added temporary stand-ins to switchtypes in branch owen_19x_pre.04 of format 02:00:00:00:00:nn where nn is BCD of the switch number (i.e. 01 - 53).

sarcasticadmin, 2022-07-26:
Fixed in https://github.com/socallinuxexpo/scale-network/pull/458

closed 2022-07-26 by sarcasticadmin